ghdl/ghdl #1243: synth: SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:762

daveshah1, 2020-04-18:
I haven't been able to minimise this one as I can't easily trace this back to a line. Please see the attached file and run:
```
ghdl --synth --std=08 DSPn.vhd -e DSPn
```

An assertion failure occurs:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-360-gea367247) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
/home/david/SNES_mister/src/chip/DSP/
Command line:
ghdl --synth --std=08 DSPn.vhd -e DSPn
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:762
Call stack traceback locations:
0x55d89e7f1832 0x55d89e545f1a 0x55d89e6d6097 0x55d89e6d2b01 0x55d89e6ea79b 0x55d89e6bfbf8 0x55d89e6c2b7d 0x55d89e6d009b 0x55d89e6d02e5 0x55d89e6bf1af 0x55d89e6bf323 0x55d89e6b7cd2 0x55d89e723e96 0x55d89e6495f5 0x55d89e7d718b 0x55d89e407dfc 0x7f40ca8eb021 0x55d89e40659c 0xfffffffffffffffe
******************************************************************
```

File: [DSPn.vhd.txt](https://github.com/ghdl/ghdl/files/4496871/DSPn.vhd.txt)

anhangaba, 2020-04-18:
Repro code:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity issue is
end issue;

architecture beh of issue is
    signal foo  : std_logic_vector (10 downto 0) := (others=>'0');
    signal bar1 : std_logic_vector (10 downto 0) := (others=>'0');
    signal bar2 : std_logic_vector (10 downto 0) := (others=>'0');
begin
    bar1 <= foo or x"40";
    bar2 <= foo or "1";
end architecture beh;
```

tgingold, 2020-04-18:
The problem is at line 483:

DATA_RAM_ADDR_B <= DP_ADDR(11 downto 1) when DP_SEL = '1' and (WR_N = '0' or RD_N = '0') else DATA_RAM_ADDR_A or x"40";

But the last "or" expressions `DATA_RAM_ADDR_A or x"40"` is not correct as the length doesn't match.

Of course, I will fix so that the error is reported but I don't understand why it doesn't fail on simulation or during synthesis with other tools.

tgingold, 2020-04-18:
Now fixed, but the whole file looks interesting...

closed 2020-04-18 by tgingold

daveshah1, 2020-04-18:
Thanks, I understand the MiSter project is intended for Quartus, and it's unsurprising that is is more lenient than it should be. Is there anything else that looks obviously problematic in that file?

tgingold, 2020-04-18:
If I replace x"40" by 11x"040", the synthesis passes.  So it should be OK.

anhangaba, 2020-04-18:
According to the std, under some specific conditionsm the bit string literal can be truncated:

From the std:
> If the length is less than the number of characters in the expanded bit value and the base specifier is B, UB, O, UO, X, UX, or D, the bit string value is obtained by deleting sufficient characters from the left of the expanded bit value to yield a string whose length is the length of the bit string literal. It is an error if any of the characters so deleted is other than the digit 0.

> If the length is less than the number of characters in the expanded bit value and the base specifier is SB, SO, or SX, the bit string value is obtained by deleting sufficient characters from the left of the expanded bit value to yield a string whose length is the length of the bit string literal. It is an error if any of the characters so deleted differs from the leftmost remaining character.

Enforcing them to have the same width throws an error on valid VHDL code.

Should I open a new issue for this?

tgingold, 2020-04-18:
Yes please!

tmeissner, 2020-04-18:
> According to the std, under some specific conditions, the bit string literal can be truncated

Wow, when started VHDL to behave like a second Verilog? This is something I wouldn't expect to work and (until now) I thought the VHDL tool would prevent me from bugs (for  me, this is clearly a bug) to work :(

anhangaba, 2020-04-19:
Actually the behavior is reasonable and safe. It truncates the leading zeros on unsigned bases and repetition of leftmost bit on signed number. Also, it only does so if length is specified.

Therefore:
`std_logic_vector'(8sx"W-")` is the equivalent as `std_logic_vector'(8sx"WWWWW-")`.

I think I my test was actually wrong, on further tests everything seems to be working fine with GHDL.

commit acc1a9f1 (2020-04-18): synth-oper: check that dyadic logical operands have same length.  Fix #1243