ghdl/ghdl #1244: synth_type_definition: cannot handle IIR_KIND_PROTECTED_TYPE_DECLARATION

rodrigomelo9, 2020-04-18:
Another example from Xilinx (ISE): [ram_protected_sharedvar.vhd.zip](https://github.com/ghdl/ghdl/files/4497287/ram_protected_sharedvar.vhd.zip)

It is a Dual Port BRAM (probably, it will fail for that when solved), which uses a protected shared variable (VHDL 2002, but I used `--std=08`). It is not a very important thing to solve, but I am assuming that it is easy.

```
$ docker run --rm -t -v $(pwd):/src -w /src ghdl/synth:beta ghdl --synth -fsynopsys --std=08 examples/ise/HDL_Coding_Techniques/rams/ram_protected_sharedvar.vhd -e 
ghdl:note: top entity is "ram_protected_sharedvar"
synth_type_definition: cannot handle IIR_KIND_PROTECTED_TYPE_DECLARATION (examples/ise/HDL_Coding_Techniques/rams/ram_protected_sharedvar.vhd:17:8)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-337-g8346c712) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/src/
Command line:
ghdl --synth -fsynopsys --std=08 examples/ise/HDL_Coding_Techniques/rams/ram_protected_sharedvar.vhd -e
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x55773a06e96e 0x55773a192563 0x55773a194097 0x55773a194200 0x55773a19348d 0x55773a164401 0x55773a16453e 0x55773a15e3df 0x55773a1c5241 0x55773a0f3ff5 0x55773a26fdff 0x557739ec39d1 0x7fa6b6136099 0x557739ec21a8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-04-18:
To be honest, I am surprising that protected types are considered synthesizable.  I think it is the first time I saw an example of it.
Ok I haven't looked a lot.

rodrigomelo9, 2020-04-18:
Is true, I have never used them. I think that is enough to avoid the ```GHDL BUG report```.

rodrigomelo9, 2020-04-18:
I checked, and this example fails with ISE and Vivado. A message such as *protected is not synthesizable* instead of the ```GHDL Bug Report``` is ok.

eine, 2020-04-18:
@rodrigomelo9, are you aware of how is this example shown to the user in ISE/Vivado? I thought that you were taking the [Language Templates](http://labs.domipheus.com/blog/wp-content/uploads/2015/10/lang_templates.png). However, since this is not sinthesizable, I wonder if there is any other source of code examples.

rodrigomelo9, 2020-04-18:
Hi @eine. No, I am not using the Language Templates. I am using `ug901-vivado-synthesis-examples.zip` for Vivado (as you can see, they are presented as synthesizable) and `xstug_examples.zip` for ISE (XST UG examples). Both of them are presented as synthesizable code. In fact, most of the provided examples are synthesizable by ISE and Vivado, but there are some exceptions, such as the presented in this issue.

RocketRoss, 2020-04-19:
Oh. I get this exact same error when a protected type has a foreign VHPIDIRECT subprogram in it. My work around is to have the foreign subprog outside of the protected type. Thought I would add my 2 cents, even though this seems to be a pretty umbrella-error... Can post a MWE of my source if anyone wants it, just shout.

tgingold, 2020-04-19:
"this exact same error" ?  I thought you were working with simulation (and not synthesis).

Please, post an mwe (on a different issue).  I am not sure we want to support foreign subprograms within a protected type, but we certainly want a better error message.

commit 80f764d9 (2020-04-19): synth: reject protected types.  Fix #1244
closed 2020-04-19 by tgingold