ghdl/ghdl #1245: Unsupported TDP RAMs

rodrigomelo9, 2020-04-18:
Hi @tgingold, I know that you are still working with the inference of BRAMs. There are several examples from Xilinx, but only four of them fails with ```ghdl --synth``` (and probably also #1244 when solved). Here you are them: [tdprams.zip](https://github.com/ghdl/ghdl/files/4497317/tdprams.zip) (the idea of this issue was to provide more test cases to you).

All of them fail with the same messages:
```
examples/vivado/asym_ram_tdp_read_first.vhd:78:9:error: multiple assignments for offsets 0:4095

examples/ise/HDL_Coding_Techniques/rams/rams_16a.vhd:29:12:error: multiple assignments for offsets 0:2047

examples/vivado/rams_tdp_record.vhd:39:8:error: multiple assignments for offsets 0:32767

examples/vivado/asym_ram_tdp_write_first.vhd:78:9:error: multiple assignments for offsets 0:4095
```

tgingold, 2020-05-28:
I am not sure the examples are correct.  The dual port ram is modelized using a signal.  But this is not a correct model.  Two drivers are created because there are two ports.  So, the value of each cell is the resolution of the two drivers, which is different from the last writing value.

I have the feeling this comes directly from verilog, without understanding the differences.

I really prefer not to handle such rams. 

But feel free to argue!

closed 2020-05-28 by tgingold

aspdigital, 2020-05-28:
Shared variables are usually used for true dual-port RAMs.

rodrigomelo9, 2020-05-28:
I understand what you say @tgingold (also, I know what @aspdigital comment). However, they are provided by Xilinx as examples (so legacy code could be using this inference them) and it works with ISE and Vivado. I will re-check (exactly what is inferred), I will try with Quartus, and I will comment (or provide a link, because I think that could be useful to have a special section on the [repo](https://github.com/rodrigomelo9/verifying-foss-hdl-synthesizers) where I testing this kind of things).

Regards

PS: I am not particularly interested in to support this kind of inference, I am only concerned about legacy code and the interoperability of tools.

tmeissner, 2020-05-28:
I agree with @tgingold the code clearly models multiple drivers as the RAM signal is written by two separate processes. 

But for legacy designs, maybe adding a switch like `relaxed` for synthesis could be used to allow such styles of RAM inference.

tgingold, 2020-05-29:
Let's wait until someone reports a problem in an actual use of such a model.