ghdl/ghdl #1254: synth: crashes when no file or unit is provided when std=08

anhangaba, 2020-04-21:
GHDL crashes when no unit is provided with `--std=08`:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-386-g5b9fb819) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl --synth --std=08
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : files_map.adb:83
Call stack traceback locations:
0x5582d5b8fa1c 0x5582d5b9210a 0x5582d5b9caa0 0x5582d5c465c1 0x5582d5cd7b2b 0x5582d5cd8506 0x5582d5d7ff0f 0x5582d5c6706b 0x5582d5c66ff6 0x5582d5d80277 0x5582d5d83007 0x5582d5da96cb 0x5582d5daa334 0x5582d5ccec97 0x5582d5e5d6cb 0x5582d5a8cdfc 0x7f1b04174151 0x5582d5a8b59c 0xfffffffffffffffe
******************************************************************
```

It works fine without the `--std=08`:
```
$ ghdl --synth
ghdl:error: no top unit found
```

tgingold, 2020-04-22:
I suppose you have a existing `work-obj08.cf` file in your directory.

anhangaba, 2020-04-22:
Yes I have. Without the `work-obj08.cf` it throws the an error.

commit b7992d86 (2020-04-23): find_top_entity: add location to report errors.  Fix #1254
closed 2020-04-23 by tgingold