ghdl/ghdl #1255: Allow running the test suite in tree (without install)

FFY00, 2020-04-22:
**Is your feature request related to a problem? Please describe.**
Yes, when creating a package for GHDL I should also run the tests, this happens before the install step. It seems like the current testsuite runs with the system GHDL instead of the tree GHDL, requiring us to install before running it.

**Describe the solution you'd like**
I would like to have a new make target, `test`. When invoked it would run the testsuite with the GHDL binary, and the other required files if needed, from the build tree.

```
make test
```

**Describe alternatives you've considered**
I could install GHDL in a temporary destination and run the test suite from there, but this is hacky and not the great solution. I could also try to pass the correct environment to the testsuite but this is not trivial and backend dependent (eg. it would be different between a gcc and a llvm build).

**Additional context**
It is common practice to have the build system provide a test target, instead of running the tests from the system.

tgingold, 2020-04-22:
Just do:
 GHDL=my_ghdl_binary ./testsuite.sh

FFY00, 2020-04-22:
As I said, this differs between the builds. And I want to test everything, not just the GHDL binary, this includes the libs. Since the Makefile knows where everything is located it should be fairly easy to make add a target to run the testsuite.

eine, 2020-04-24:
I've been quite busy this week and I yet have to check several of @FFY00's comments and commits regarding packaging for Arch Linux. However, if this issue is to replace `./testsuite.sh` with `make test` in the PKGBUILD file, I'm afraid that there should be no difference. This is because the testsuite expects both passing and failing tests. When I tried to run the testsuite from the PKGBUILD file, I found that such behaviour would make the build fail on the first test which was indeed expected to fail. I'd be glad to know whether this can be solved regardless of which command is used in the testing task.

FFY00, 2020-04-24:
@tgingold that is great, thanks! Are the in-tree libraries used by the testsuite?

> I've been quite busy this week and I yet have to check several of @FFY00's comments and commits regarding packaging for Arch Linux. However, if this issue is to replace `./testsuite.sh` with `make [...]

Then the testsuite is handling it incorrectly. It should be asserting the exit code of the tests expected to fail is correct instead of erroring itself, which is waht is happening here. Asserting the exit code also makes sure the test fail due to the expected reason, and not an unrelated one :smile:.

tgingold, 2020-04-24:
> @tgingold that is great, thanks! Are the in-tree libraries used by the testsuite?

Most of them are.

> Then the testsuite is handling it incorrectly. It should be asserting the exit code of the tests expected to fail is correct instead of erroring itself, which is waht is happening here. Asserting the exit code also makes sure the test fail due to the expected reason, and not an unrelated one 😄.

That's the same testsuite that is also used for CI, so expected failures are handled.

tgingold, 2020-04-24:
I am not sure I understand your comment.  We also use this testsuite for CI.

eine, 2020-04-24:
> It should be asserting the exit code of the tests expected to fail is correct instead of erroring itself, which is waht is happening here. Asserting the exit code also makes sure the test fail due to the expected reason, and not an unrelated one 😄.

Actually, that's how it works. When you run `GHDL=/path/to/ghdl ./testsuite.sh`, some tests are expected to fail in some specific step and others are expected to pass. All of them are properly handled and when an expected failure does fail the test result is success (which is correct).

However, when an Arch package is being built, as soon as the failure is produced, the build is aborted, without letting the testsuite catch it and continue. I.e., it seems that `makepkg` forces `set -e`.

FFY00, 2020-04-24:
Yes, makepkg exits on failures, but that has nothing to do with the failures inside `testsuite.sh`. makepkg will only exit if `testsuite.sh` exits with an error code.

This is the output when I try to run the tests. I can reproduce outside makepkg.
```
Running tests for ghdl-mcode...
tests: sanity gna vests synth
[GHDL - test] sanity
sanity 000hello: ok
sanity 001hello87: ok
sanity 002hello2008: ok
sanity 004all08: ok
sanity 005examples: ok
sanity 006upf: ok
[GHDL - test] gna
gna bug01: ok
gna bug010: ok
gna bug0100: ok
gna bug0101: ok
gna bug0103: ok
gna bug0104: ok
gna bug0105: ok
gna bug0106: ok
gna bug0108: ok
gna bug0109: failed
analyze repro.vhdl
elaborate and simulate repro
repro.vhdl:33:9:@0ms:(report note): z = image-ent
analyze err.vhdl
err.vhdl:3:10: unit "fixed_pkg" not found in library "ieee"
err.vhdl:9:28: entity 'phz_calc' was not analysed
```

FFY00, 2020-04-24:
This is on the `v0.37` tag. I will try -git.

eine, 2020-04-24:
> Yes, makepkg exits on failures, but that has nothing to do with the failures inside `testsuite.sh`. makepkg will only exit if `testsuite.sh` exits with an error code.

That's good to know! Then, it was probably an error on my side. I'll try it again ASAP.

> This is the output when I try to run the tests. I can reproduce outside makepkg.
> analyze repro.vhdl
> elaborate and simulate repro
> repro.vhdl:33:9:@0ms:(report note): z = image-ent
> analyze err.vhdl
> err.vhdl:3:10: unit "fixed_pkg" not found in library "ieee"
> err.vhdl:9:28: entity 'phz_calc' was not analysed
> ```

This might be some misconfiguration/misinstallation. This is what I get:

```sh
~/ghdl/testsuite/gna/bug0109# GHDL=ghdl ./testsuite.sh
analyze repro.vhdl
elaborate and simulate repro
repro.vhdl:33:9:@0ms:(report note): z = image-ent
analyze err.vhdl
elaborate and simulate phz_calc
err.vhdl:30:9:@0ms:(report note): z = 0011.010
analyze ok.vhdl
elaborate and simulate foo
ok.vhdl:57:9:@0ms:(report note): sfixed_val = 1101.1010
Remove work library
Test successful
```

Did you build/install GHDL with option `--gpl`?

FFY00, 2020-04-24:
Do you mean `--enable-openieee`? I don't see a reference to `--gpl` anywhere.

FFY00, 2020-04-24:
These are the parameters I am passing to `configure`: https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/ghdl#n53

FFY00, 2020-04-24:
After removing `--enable-openieee`:

```
tests: sanity gna vests synth
[GHDL - test] sanity
sanity 000hello: ok
sanity 001hello87: ok
sanity 002hello2008: ok
sanity 004all08: ok
sanity 005examples: ok
sanity 006upf: ok
[GHDL - test] gna
gna bug01: ok
gna bug010: ok
gna bug0100: ok
gna bug0101: ok
gna bug0103: ok
gna bug0104: ok
gna bug0105: ok
gna bug0106: ok
gna bug0108: ok
gna bug0109: ok
gna bug011: ok
gna bug0110: ok
gna bug012: ok
gna bug014: ok
gna bug015: ok
gna bug016: ok
gna bug017: ok
gna bug018: ok
gna bug019: ok
gna bug02: ok
gna bug021: ok
gna bug022: ok
gna bug023: ok
gna bug024: ok
gna bug026: ok
gna bug027: ok
gna bug028: ok
gna bug029: ok
gna bug03: ok
gna bug030: ok
gna bug031: ok
gna bug032: ok
gna bug033: ok
gna bug034: ok
gna bug035: ok
gna bug036: ok
gna bug037: ok
gna bug039: ok
gna bug04: ok
gna bug040: ok
gna bug041: ok
gna bug042: ok
gna bug043: ok
gna bug044: ok
gna bug045: ok
gna bug046: ok
gna bug047: ok
gna bug048: ok
gna bug049: ok
gna bug05: ok
gna bug050: ok
gna bug051: ok
gna bug052: ok
gna bug053: ok
gna bug054: ok
gna bug055: ok
gna bug056: ok
gna bug057: ok
gna bug058: ok
gna bug059: ok
gna bug06: ok
gna bug060: ok
gna bug061: ok
gna bug062: ok
gna bug063: ok
gna bug064: ok
gna bug065: ok
gna bug066: ok
gna bug067: ok
gna bug069: ok
gna bug07: ok
gna bug071: ok
gna bug072: ok
gna bug073: ok
gna bug074: ok
gna bug075: ok
gna bug077: ok
gna bug078: ok
gna bug079: ok
gna bug08: ok
gna bug080: ok
gna bug081: ok
gna bug082: ok
gna bug083: ok
gna bug084: ok
gna bug085: ok
gna bug086: ok
gna bug087: ok
gna bug088: ok
gna bug09: ok
gna bug090: ok
gna bug092: ok
gna bug093: ok
gna bug094: ok
gna bug096: ok
gna bug097: ok
gna bug098: ok
gna bug099: ok
gna bug14953: ok
gna bug15638: ok
gna bug15702: ok
gna bug15933: ok
gna bug15966: ok
gna bug15993: ok
gna bug16095: ok
gna bug16096: ok
gna bug16144: ok
gna bug16287: ok
gna bug16695: ok
gna bug16782: ok
gna bug17127: ok
gna bug17202: ok
gna bug17203: ok
gna bug17309: ok
gna bug17545: ok
gna bug17759: ok
gna bug18280: ok
gna bug18351: ok
gna bug18353: ok
gna bug18359: ok
gna bug18360: ok
gna bug18361: ok
gna bug18659: ok
gna bug18810: ok
gna bug19195: ok
gna bug20255: ok
gna bug20312: ok
gna bug20549: ok
gna bug20597: ok
gna bug20703: ok
gna bug20767: ok
gna bug20771: ok
gna bug21052: ok
gna bug21078: ok
gna bug21274: ok
gna bug21332: ok
gna bug21487: ok
gna bug21497: ok
gna bug21500: ok
gna bug21513: ok
gna bug22868: ok
gna bug23013: ok
gna bug23165: ok
gna bug23482: ok
gna bug24064: ok
gna bug24065: ok
gna bug24324: ok
gna bug24326: ok
gna bug7186: ok
gna bug7751: ok
gna deb573721: ok
gna deb585748: ok
gna deb585750: ok
gna gls7: ok
gna issue10: ok
gna issue1015: ok
gna issue103: ok
gna issue1038: ok
gna issue1051: ok
gna issue1057: ok
gna issue106: ok
gna issue1063: ok
gna issue107: ok
gna issue1120: ok
gna issue1123: ok
gna issue1125: ok
gna issue1137: ok
gna issue1145: ok
gna issue123: ok
gna issue125: ok
gna issue126: ok
gna issue132: ok
gna issue136: ok
gna issue14: ok
gna issue140: ok
gna issue141: ok
gna issue143: ok
gna issue144: ok
gna issue147: ok
gna issue148: ok
gna issue15: ok
gna issue150: ok
gna issue151: ok
gna issue152: ok
gna issue155: ok
gna issue156: ok
gna issue158: ok
gna issue160: ok
gna issue163: ok
gna issue167: ok
gna issue17: ok
gna issue176: ok
gna issue18: ok
gna issue190: ok
gna issue191: ok
gna issue199: ok
gna issue2: ok
gna issue20: ok
gna issue201: ok
gna issue202: ok
gna issue204: ok
gna issue205: ok
gna issue207: ok
gna issue209: ok
gna issue212: ok
gna issue216: ok
gna issue223: ok
gna issue225: ok
gna issue227: ok
gna issue228: ok
gna issue233: ok
gna issue235: ok
gna issue238: ok
gna issue241: ok
gna issue242: ok
gna issue243: ok
gna issue244: ok
gna issue25: ok
gna issue253: ok
gna issue254: ok
gna issue256: ok
gna issue257: ok
gna issue258: ok
gna issue259: ok
gna issue26: ok
gna issue261: ok
gna issue262: ok
gna issue263: ok
gna issue264: ok
gna issue265: ok
gna issue268: ok
gna issue273: ok
gna issue278: ok
gna issue283: ok
gna issue290: ok
gna issue293: ok
gna issue295: ok
gna issue30: ok
gna issue300: ok
gna issue301: ok
gna issue309: ok
gna issue310: ok
gna issue312: ok
gna issue314: ok
gna issue316: ok
gna issue317: ok
gna issue320: ok
gna issue321: ok
gna issue326: ok
gna issue328: ok
gna issue332: ok
gna issue333: ok
gna issue339: ok
gna issue349: ok
gna issue351: ok
gna issue356: ok
gna issue361: ok
gna issue367: ok
gna issue368: ok
gna issue369: ok
gna issue370: ok
gna issue371: ok
gna issue372: ok
gna issue373: ok
gna issue374: ok
gna issue375: ok
gna issue376: ok
gna issue379: ok
gna issue38: ok
gna issue381: ok
gna issue382: ok
gna issue384: ok
gna issue387: ok
gna issue388: ok
gna issue389: ok
gna issue392: ok
gna issue394: ok
gna issue395: ok
gna issue397: ok
gna issue40: ok
gna issue400: ok
gna issue406: ok
gna issue407: ok
gna issue413: ok
gna issue418: ok
gna issue419: ok
gna issue42: ok
gna issue424: ok
gna issue43: ok
gna issue439: ok
gna issue44: ok
gna issue440: ok
gna issue441: ok
gna issue447: ok
gna issue449: ok
gna issue45: ok
gna issue450: failed
analyze disptree.vhdl
elaborate disptree
/usr/bin/gcc -c vpi2.c -I/build/ghdl/src/ghdl-mcode/include -fPIC
vpi2.c:9:10: fatal error: vpi_user.h: No such file or directory
    9 | #include <vpi_user.h>
      |          ^~~~~~~~~~~~
compilation terminated.
```

FFY00, 2020-04-24:
After removing `--enable-openieee`:

```
tests: sanity gna vests synth
[GHDL - test] sanity
sanity 000hello: ok
sanity 001hello87: ok
sanity 002hello2008: ok
sanity 004all08: ok
sanity 005examples: ok
sanity 006upf: ok
[GHDL - test] gna
gna bug01: ok
gna bug010: ok
gna bug0100: ok
gna bug0101: ok
gna bug0103: ok
gna bug0104: ok
gna bug0105: ok
gna bug0106: ok
gna bug0108: ok
gna bug0109: ok
gna bug011: ok
gna bug0110: ok
gna bug012: ok
gna bug014: ok
gna bug015: ok
gna bug016: ok
gna bug017: ok
gna bug018: ok
gna bug019: ok
gna bug02: ok
gna bug021: ok
gna bug022: ok
gna bug023: ok
gna bug024: ok
gna bug026: ok
gna bug027: ok
gna bug028: ok
gna bug029: ok
gna bug03: ok
gna bug030: ok
gna bug031: ok
gna bug032: ok
gna bug033: ok
gna bug034: ok
gna bug035: ok
gna bug036: ok
gna bug037: ok
gna bug039: ok
gna bug04: ok
gna bug040: ok
gna bug041: ok
gna bug042: ok
gna bug043: ok
gna bug044: ok
gna bug045: ok
gna bug046: ok
gna bug047: ok
gna bug048: ok
gna bug049: ok
gna bug05: ok
gna bug050: ok
gna bug051: ok
gna bug052: ok
gna bug053: ok
gna bug054: ok
gna bug055: ok
gna bug056: ok
gna bug057: ok
gna bug058: ok
gna bug059: ok
gna bug06: ok
gna bug060: ok
gna bug061: ok
gna bug062: ok
gna bug063: ok
gna bug064: ok
gna bug065: ok
gna bug066: ok
gna bug067: ok
gna bug069: ok
gna bug07: ok
gna bug071: ok
gna bug072: ok
gna bug073: ok
gna bug074: ok
gna bug075: ok
gna bug077: ok
gna bug078: ok
gna bug079: ok
gna bug08: ok
gna bug080: ok
gna bug081: ok
gna bug082: ok
gna bug083: ok
gna bug084: ok
gna bug085: ok
gna bug086: ok
gna bug087: ok
gna bug088: ok
gna bug09: ok
gna bug090: ok
gna bug092: ok
gna bug093: ok
gna bug094: ok
gna bug096: ok
gna bug097: ok
gna bug098: ok
gna bug099: ok
gna bug14953: ok
gna bug15638: ok
gna bug15702: ok
gna bug15933: ok
gna bug15966: ok
gna bug15993: ok
gna bug16095: ok
gna bug16096: ok
gna bug16144: ok
gna bug16287: ok
gna bug16695: ok
gna bug16782: ok
gna bug17127: ok
gna bug17202: ok
gna bug17203: ok
gna bug17309: ok
gna bug17545: ok
gna bug17759: ok
gna bug18280: ok
gna bug18351: ok
gna bug18353: ok
gna bug18359: ok
gna bug18360: ok
gna bug18361: ok
gna bug18659: ok
gna bug18810: ok
gna bug19195: ok
gna bug20255: ok
gna bug20312: ok
gna bug20549: ok
gna bug20597: ok
gna bug20703: ok
gna bug20767: ok
gna bug20771: ok
gna bug21052: ok
gna bug21078: ok
gna bug21274: ok
gna bug21332: ok
gna bug21487: ok
gna bug21497: ok
gna bug21500: ok
gna bug21513: ok
gna bug22868: ok
gna bug23013: ok
gna bug23165: ok
gna bug23482: ok
gna bug24064: ok
gna bug24065: ok
gna bug24324: ok
gna bug24326: ok
gna bug7186: ok
gna bug7751: ok
gna deb573721: ok
gna deb585748: ok
gna deb585750: ok
gna gls7: ok
gna issue10: ok
gna issue1015: ok
gna issue103: ok
gna issue1038: ok
gna issue1051: ok
gna issue1057: ok
gna issue106: ok
gna issue1063: ok
gna issue107: ok
gna issue1120: ok
gna issue1123: ok
gna issue1125: ok
gna issue1137: ok
gna issue1145: ok
gna issue123: ok
gna issue125: ok
gna issue126: ok
gna issue132: ok
gna issue136: ok
gna issue14: ok
gna issue140: ok
gna issue141: ok
gna issue143: ok
gna issue144: ok
gna issue147: ok
gna issue148: ok
gna issue15: ok
gna issue150: ok
gna issue151: ok
gna issue152: ok
gna issue155: ok
gna issue156: ok
gna issue158: ok
gna issue160: ok
gna issue163: ok
gna issue167: ok
gna issue17: ok
gna issue176: ok
gna issue18: ok
gna issue190: ok
gna issue191: ok
gna issue199: ok
gna issue2: ok
gna issue20: ok
gna issue201: ok
gna issue202: ok
gna issue204: ok
gna issue205: ok
gna issue207: ok
gna issue209: ok
gna issue212: ok
gna issue216: ok
gna issue223: ok
gna issue225: ok
gna issue227: ok
gna issue228: ok
gna issue233: ok
gna issue235: ok
gna issue238: ok
gna issue241: ok
gna issue242: ok
gna issue243: ok
gna issue244: ok
gna issue25: ok
gna issue253: ok
gna issue254: ok
gna issue256: ok
gna issue257: ok
gna issue258: ok
gna issue259: ok
gna issue26: ok
gna issue261: ok
gna issue262: ok
gna issue263: ok
gna issue264: ok
gna issue265: ok
gna issue268: ok
gna issue273: ok
gna issue278: ok
gna issue283: ok
gna issue290: ok
gna issue293: ok
gna issue295: ok
gna issue30: ok
gna issue300: ok
gna issue301: ok
gna issue309: ok
gna issue310: ok
gna issue312: ok
gna issue314: ok
gna issue316: ok
gna issue317: ok
gna issue320: ok
gna issue321: ok
gna issue326: ok
gna issue328: ok
gna issue332: ok
gna issue333: ok
gna issue339: ok
gna issue349: ok
gna issue351: ok
gna issue356: ok
gna issue361: ok
gna issue367: ok
gna issue368: ok
gna issue369: ok
gna issue370: ok
gna issue371: ok
gna issue372: ok
gna issue373: ok
gna issue374: ok
gna issue375: ok
gna issue376: ok
gna issue379: ok
gna issue38: ok
gna issue381: ok
gna issue382: ok
gna issue384: ok
gna issue387: ok
gna issue388: ok
gna issue389: ok
gna issue392: ok
gna issue394: ok
gna issue395: ok
gna issue397: ok
gna issue40: ok
gna issue400: ok
gna issue406: ok
gna issue407: ok
gna issue413: ok
gna issue418: ok
gna issue419: ok
gna issue42: ok
gna issue424: ok
gna issue43: ok
gna issue439: ok
gna issue44: ok
gna issue440: ok
gna issue441: ok
gna issue447: ok
gna issue449: ok
gna issue45: ok
gna issue450: failed
analyze disptree.vhdl
elaborate disptree
/usr/bin/gcc -c vpi2.c -I/build/ghdl/src/ghdl-mcode/include -fPIC
vpi2.c:9:10: fatal error: vpi_user.h: No such file or directory
    9 | #include <vpi_user.h>
      |          ^~~~~~~~~~~~
compilation terminated.
```

Seem like the include directory is wrong.

eine, 2020-04-24:
> Do you mean `--enable-openieee`? I don't see a reference to `--gpl` anywhere.

Yes. `--gpl` is the option used in https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh, while `--enable-openieee` is the configure option. Surprisingly, the former does not seem to set up the later explicitly.

Anyway, regarding your issue, openieee is a subset/incomplete implementation, compared to IEEE's libraries. Ref [April 23, 2020 6:07 PM](https://gitter.im/VUnit/vunit?at=5ea1bd3f94ac2758e302d1a6). You need the non-openieee build in order to run the full test suite. I believe that the subset of tests for openieee only is not defined.

> Seems like the include directory is wrong.

This might be an issue with `master`. Did you try v0.37 without `--enable-openieee`?

FFY00, 2020-04-24:
> Yes. `--gpl` is the option used in [/dist/ci-run.sh@`master`](https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh) [](/ghdl/ghdl/blob/HEAD@{2020-04-24T19:09:01Z}/dist/ci-run.sh), while `--enable-openieee` is the configure option. Surprisingly, the former does not seem to set up the later explicitly.
> 
> Anyway, regarding your issue, openieee is a subset/incomplete implementation, compared to IEEE's libraries. Ref [April 23, 2020 6:07 PM](https://gitter.im/VUnit/vunit?at=5ea1bd3f94ac2758e302d1a6). You need the non-openieee build in order to run the full test suite. I believe that the subset of tests for openieee only is not defined.

How are those libraries licensed?

> This might be an issue with `master`. Did you try v0.37 without `--enable-openieee`?

That output is from `v0.37`.

eine, 2020-04-24:
> How are those libraries licensed?

Versions published before VHDL 2019 are copyrighted by IEEE and there is no open source or copyleft license. However, GHDL has been distributing them verbatim for years now, hence we can assume that distribution is allowed.

Since december, libraries corresponding to VHDL 2019 are open source (under an Apache license). Those are hosted in a GitLab repo.

Currently, most of open source VHDL 2019 IEEE libs are, in fact, compatible with VHDL 2008. Hence, it has been proposed to have a tag for backward compatibility (i.e. to have open source libs for VHDL 2008), before enhancements are pushed which do require VHDL 2019 features. However, becase VHDL 2008 was published a decade ago and conditions were/are already set, it seems that such a movement might not be straightforward. This has been discussed in https://gitter.im/vhdl/General and in the VASG mailing list (http://grouper.ieee.org/groups/1076/email/thread.html).

> That output is from `v0.37`.

Does `vpi_user.h` exist in a different subdir? Or is it just not copied during the `make install`?

tgingold, 2020-04-24:
> Does vpi_user.h exist in a different subdir? Or is it just not copied during the make install?

It is possible that vpi_user.h is always looked from the install dir.

FFY00, 2020-04-24:
> > How are those libraries licensed?
> 
> Versions published before VHDL 2019 are copyrighted by IEEE and there is no open source or copyleft license. However, GHDL has been distributing them verbatim for years now, hence we can assume that distribution is allowed.
> 
> Since december, libraries corresponding to VHDL 2019 are open source (under an Apache license). Those are hosted in a GitLab repo.
> 
> Currently, most of open source VHDL 2019 IEEE libs are, in fact, compatible with VHDL 2008. Hence, it has been proposed to have a tag for backward compatibility (i.e. to have open source libs for VHDL 2008), before enhancements are pushed which do require VHDL 2019 features. However, because VHDL 2008 was published a decade ago and conditions were/are already set, it seems that such a movement might not be straightforward. This has been discussed in https://gitter.im/vhdl/General and in the VASG mailing list (http://grouper.ieee.org/groups/1076/email/thread.html).

Sorry, but that does not work for me. I cannot distribute them in the Arch Linux repositories without clear permission from them :/

> > That output is from `v0.37`.
> 
> Does `vpi_user.h` exist in a different subdir? Or is it just not copied during the `make install`?

This is the output running the testsuite in-tree, without `make install`.

eine, 2020-04-24:
> Sorry, but that does not work for me. I cannot distribute them in the Arch Linux repositories without clear permission from them :/

I know. That's why openieee exists, because the same requirement applies to Debian since several years ago (https://gitter.im/VUnit/vunit?at=5ea1bd3f94ac2758e302d1a6). On the one hand, you might want to ask for permission. On the other hand, you can avoid running the testsuite. I believe that contributions to split the subset of the testsuite that passes with openieee are also welcome.

> > Does `vpi_user.h` exist in a different subdir? Or is it just not copied during the `make install`?
> 
> This is the output running the testsuite in-tree, without `make install`.

Then, you might need to `make DESTDIR=/path/to/temporal/dir install` before running the testsuite.

FFY00, 2020-04-24:
Well, that was the whole point of this issue :smile:. Since the makefile knows where to find the missing file we should be able to pass it to the testsuite and resolve the issue.

eine, 2020-04-24:
> Well, that was the whole point of this issue 😄. Since the makefile knows where to find the missing file we should be able to pass it to the testsuite and resolve the issue.

The point is that GHDL adds `-I` to GCC, as a relative location to the GHDL binary/installation that is being used. It is already possible to add other custom/unrelated locations, but it needs to be done explicitly for each `testsuite.sh`. For example, adding `-Wl,-I"$(pwd)"/../../../src/grt/` to https://github.com/ghdl/ghdl/blob/master/testsuite/gna/issue450/testsuite.sh#L11 might solve this issue. Similar fixes need to be applied to all the tests that require resources which are expected to exists in the install dir but not in the build dir.

tgingold, 2020-04-25:
The vhdl 2008 libraries in ghdl comes from the open source vhdl-2019 ilbraries, so they can be distributed.
So they could be distributed with openieee.  That's a change to do.

tgingold, 2020-04-25:
> Well, that was the whole point of this issue 😄. Since the makefile knows where to find the missing file we should be able to pass it to the testsuite and resolve the issue.

That makes sense.  Currently you can do 'make install.vpi.local' before 'make tests'

eine, 2020-04-26:
> The vhdl 2008 libraries in ghdl comes from the open source vhdl-2019 ilbraries, so they can be distributed. So they could be distributed with openieee. That's a change to do.

What about the fact that `--gpl` seems not to use `--enable-openieee` explicitly. Should we change that too?

FFY00, 2020-04-26:
Should this be reopened?

tgingold, 2020-04-28:
Yes, that would make sense.

eine, 2021-01-17:
I think that this is fixed now. In the current master, all installed libraries are based on the Apache 2.0 sources from IEEE. When option `--enable-gplcompat` is used in configure, VITAL libraries are removed, which are the only non-free sources now. Configure option `--enable-openieee` was removed.

@tgingold, @FFY00 can we close this?

/cc @sharkcz @marzoul, pinging here for you to be aware of the change from `--enable-openieee` to `--enable-gplcompat`, in case that's relevant in your packaging projects.

commit a4e7fd3e (2020-04-24): Makefile: add test target.  Fix #1255
closed 2021-06-05 by tgingold