ghdl/ghdl #1256: Unable to Read/Write enum types using VPI

bellaz89, 2020-04-23:
**Description**
When using vpi_set_value on a VHDL enum (without encoding hints), apparently no effects are produced (no writes), reading from a VHDL enum type returns a null.  

**Expected behaviour**
That writing to an enum changes the state of the simulation and that reading an enum returns its bit representation.

**How to reproduce?**

```c :file: vpi_plugin.c
#include<vpi_user.h>
#include<inttypes.h>
#include<stdio.h>

#define STOP_ITERATION 5

uint32_t iteration = 0;

const char* input_values[5] = {"000",
                               "001",
                               "010",
                               "011",
                               "100"};

vpiHandle enum_in_handle;
vpiHandle enum_out_handle;
vpiHandle enum_decoded_handle;

PLI_INT32 start_cb(p_cb_data);
PLI_INT32 end_cb(p_cb_data);
PLI_INT32 rw_cb(p_cb_data);
PLI_INT32 ro_cb(p_cb_data);
PLI_INT32 delay_rw_cb(p_cb_data);
PLI_INT32 delay_ro_cb(p_cb_data);

void register_cb(PLI_INT32(*f)(p_cb_data),
                 PLI_INT32 reason,
                 int64_t cycles){

    s_cb_data cbData;
    s_vpi_time simuTime;
    if (cycles < 0){
        cbData.time = NULL;
    } else {
        cbData.time = &simuTime;
        simuTime.type = vpiSimTime;
        simuTime.high = (PLI_INT32) (cycles >> 32);
        simuTime.low = (PLI_INT32) (cycles & 0xFFFFFFFF);
    }

    cbData.reason = reason;
    cbData.cb_rtn = f;
    cbData.user_data = 0;
    cbData.value = 0;

    vpi_register_cb(&cbData);
}

void entry_point_cb() {
    register_cb(start_cb, cbStartOfSimulation, -1);
    register_cb(end_cb, cbEndOfSimulation, -1);
    register_cb(delay_ro_cb, cbAfterDelay, 0);
}

PLI_INT32 start_cb(p_cb_data data){
    (void) data;
    printf("Start of simulation \n");

    enum_in_handle = vpi_handle_by_name("enum_test.enum_in", NULL);
    if(!enum_in_handle) printf("enum in not found\n");
    enum_out_handle = vpi_handle_by_name("enum_test.enum_out", NULL);
    if(!enum_out_handle) printf("enum out not found\n");
    enum_decoded_handle = vpi_handle_by_name("enum_test.enum_decoded", NULL);
    if(!enum_decoded_handle) printf("enum decoded not found\n");
    
    return 0;
}

PLI_INT32 end_cb(p_cb_data data){
    (void) data;
    printf("End of simulation %u \n", iteration);
    return 0;
}

PLI_INT32 rw_cb(p_cb_data data){
    (void) data;
    s_vpi_value val;
    val.format = vpiBinStrVal;
    
    if(iteration > 0) {
        vpi_get_value(enum_decoded_handle, &val);
        val.format = vpiBinStrVal;
        printf("enum decoded = %s iteration %d \n", val.value.str, iteration);    
        vpi_get_value(enum_out_handle, &val);
        printf("enum out = %s iteration %d \n", val.value.str, iteration);    

        val.format = vpiBinStrVal;
    }

    if(iteration < STOP_ITERATION) {

        val.value.str = (char *) input_values[iteration];
        printf("enum in = %s iteration %d \n", val.value.str, iteration);    
        vpi_put_value(enum_in_handle, &val, NULL, vpiNoDelay);    
        register_cb(delay_ro_cb, cbAfterDelay, 1);
    } else {
        vpi_control(vpiFinish, 0);
    }

    iteration++;
    return 0;
}

PLI_INT32 ro_cb(p_cb_data data){
    (void) data;
    register_cb(delay_rw_cb, cbAfterDelay, 0);
    return 0;
}

PLI_INT32 delay_rw_cb(p_cb_data data){
    (void) data;
    register_cb(rw_cb, cbReadWriteSynch, 0);
    return 0;
}

PLI_INT32 delay_ro_cb(p_cb_data data){
    (void) data;
    register_cb(ro_cb, cbReadOnlySynch, 0);
    return 0;
}

void (*vlog_startup_routines[]) () = {
    entry_point_cb,
    0
};
```

```vhd :file: enum_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pkg_enum is
  type State is (A,B,C,D,E);
end pkg_enum;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.all;
use work.pkg_enum.all;

entity enum_test is
    port(enum_in : in State;
         enum_out : out State;
         enum_decoded: out unsigned(2 downto 0)
     );
end enum_test;

architecture arch of enum_test is
begin
    with enum_in select
        enum_decoded <= "000" when A,
                        "001" when B,
                        "010" when C,
                        "011" when D,
                        "100" when others;
    
    enum_out <= enum_in;
end arch;
```
```sh :file: execute.sh
$GHDL -a enum_test.vhd
$GHDL --vpi-compile gcc -c vpi_plugin.c
$GHDL --vpi-link gcc -o vpi_plugin.vpi vpi_plugin.o -lc
$GHDL --elab-run enum_test --vpi=./vpi_plugin.vpi
```

to run:
```sh :image: ghdl/ghdl:buster-mcode
GHDL=ghdl sh execute.sh
```

**Context**
Please, provide the following information:

- OS: Debian 11 (Bullseye)
- Origin:
  - [ x ] Package manager: ghdl 0.37+dfsg-1
  - [ ] Released binaries: 
  - [ x ] Built from sources: commit d8cd7bc8a47e283028c7f90ece2e768f7bfb4fa6

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

**Additional context**
Since the VPI standard doesn't explicitly support the use of enum a behavior for GHDL has to be defined.

This bug was discovered working for https://github.com/SpinalHDL/SpinalHDL/issues/146

tgingold, 2020-04-23:
I will have a look, but I am not sure how well enumerations were supported by VPI.

For sure, the encoding is sequential, starting from 0.

tgingold, 2020-04-23:
For the enums, vpi_get_value returns NULL because it is not supported.

commit 520e8f8e (2020-04-24): grt-vpi: handle read/write of enums.  Fix #1256
closed 2020-04-24 by tgingold