ghdl/ghdl #1257: Make bound check failures provide more information

GlenNicholls, 2020-04-23:
[Feature Request]

When getting bound check failures, it can sometimes be ambiguous to figure out exactly what is wrong. Take the following example:

```
    signal left : std_logic_vector(1 downto 0);
    signal right : std_logic_vector(2 downto 0);
...
    left <= right;
```

Now, as expected, GHDL fails with this error:

```
D:\<path>\tb_test-tb:error: bound check failure at D:\<path>\test.vhd:40
```

In this scenario, it would be nice if GHDL reported the bound check failure similar to ModelSim such that it provides extra information about the range/size of the left and right of the assignment:

```
** Fatal: (vsim-3420) Array lengths do not match. Left is 2 (1 downto 0). Right is 3 (2 downto 0).
<also provides file and line number where error occurs like GHDL>
```

In the above case, this is easy to spot, but when doing comlicated assignments, it gets annoying to track down where the problem is.

The other problem with the current error is when `downto/to` are not used properly:

```
    signal left : std_logic_vector(1 downto 0);
    signal right : std_logic_vector(0 to 1);
...
    left <= right;
```

ModelSim provides the following error that makes this easy to spot:

```
# ** Fatal: (vsim-3607) Slice range direction (downto) does not match slice prefix direction (to).
#    Time: 50945 ns  Iteration: 2  Process: /tb_serialrx/Runner_p File: D:/<path>/test.vhd
```

In GHDL, however, the error is the same as before which makes it even more difficult to debug when code is complicated:

```
D:\<path>\tb_test-tb:error: bound check failure at D:\<path>\test.vhd:40
```

I think that GHDL should handle the above cases more gracefully by providing the vector indexing direction if that is the problem, or the sizes. The above two cases highlight why it can be difficult to figure out which is the culprit case. 

The code I used to find this problem is the following:

```
    -- function that can flip words around the center axis of an SLV
    -- ex. input=0x002_001, width=12, output=0x001_002
    -- ex. input=0x03_02_01, width=8, output=0x01_02_03
    function wordrevorder (
        arg   : std_logic_vector;
        width : positive
    )
    return std_logic_vector is
        constant c_ratio : integer := arg'length/width;
        variable v_ret : std_logic_vector(arg'range);
    begin

        for i in 0 to c_ratio-1 loop
            report "wordrevorder:" &LF&
                "i=" & integer'image(i) &LF&
                "c_ratio=" & integer'image(c_ratio) &LF&
                "width=" & integer'image(width) &LF&
                "(i*width)+width-1=" & integer'image((i*width)+width-1) &LF&
                "(i*width)=" & integer'image(i*width) &LF&
                "((c_ratio-i-1)*width)+width-1=" & integer'image(((c_ratio-i-1)*width)+width-1) &LF&
                "((c_ratio-i-1)*width)=" & integer'image(((c_ratio-i-1)*width));
error ->            v_ret((i*width)+width-1 downto (i*width)) :=
                arg(((c_ratio-i-1)*width)+width-1 downto ((c_ratio-i-1)*width));
        end loop;
        return v_ret;
    end;
```

The problem was that I made the assumption that `arg`'s range was `downto` when in reality it was `to` After providing the above debugging to see all of the ranges, it still didn't work. I had to open ModelSim to get more insight into why GHDL was barfing.

anhangaba, 2020-04-24:
Also, the `error: index not within bounds` could provide more information.

Repro code fragment:
```
signal foo : std_logic_vector(7 downto 0);
---
for v_i in 0 to 8 loop
    foo (v_i) <= '1';
end loop;
```

tgingold, 2020-04-24:
Please, provide mwe.  There are many cases to be handled, and I'd like to know which ones are the most important.

For the initial case, you get a diagnostic at analysis time, like:
```
repro1.vhdl:11:13:warning: value constraints don't match target ones [-Wruntime-error]
```

So I am not sure if I need to improve this diagnostic or to add a better one a runtime (for *this* example).

GlenNicholls, 2020-04-27:
@tgingold Alright, here is an MWE for both cases 
[range_direction_mismatch.zip](https://github.com/ghdl/ghdl/files/4541041/range_direction_mismatch.zip)

Sorry about the above not working, it was because what I was describing was inside a function.

Anyways, for the case when ranges are mismatched (e.g. same direction but different sizes), GHDL IMO should report that this is a width difference and provide the width of the left and right vectors. When running tb_range_mismatch, I only get a bound check error:

```
Starting mwe.tb_range_mismatch.test_range
Output file: vunit_out\test_output\mwe.tb_range_mismatch.test_range_45fdc5b383bfcdcae59023089e517b54832c6be4\output.txt
D:\ghdl\mwe\range_mismatch\vunit_out\test_output\mwe.tb_range_mismatch.test_range_45fdc5b383bfcdcae59023089e517b54832c6be4\ghdl\tb_range_mismatch-tb:error: bound check failure at D:\ghdl\mwe\range_mismatch\range_mismatch.vhd:14
D:\ghdl\mwe\range_mismatch\vunit_out\test_output\mwe.tb_range_mismatch.test_range_45fdc5b383bfcdcae59023089e517b54832c6be4\ghdl\tb_range_mismatch-tb:error: simulation failed
```

For the case of range mismatches, the code is not valid and is immediately discovered with signal assignments. However, because it is masked inside a function, The root cause is said to be a bound check failure like the previous case:

```
Starting mwe.tb_direction_mismatch.test_direction
Output file: vunit_out\test_output\mwe.tb_direction_mismatch.test_direction_a5b1177fd10bca29c22d78c426a567db23efa40f\output.txt
D:\ghdl\mwe\range_mismatch\vunit_out\test_output\mwe.tb_direction_mismatch.test_direction_a5b1177fd10bca29c22d78c426a567db23efa40f\ghdl\tb_direction_mismatch-tb:error: bound check failure at D:\ghdl\mwe\range_mismatch\direction_mismatch.vhd:31
D:\ghdl\mwe\range_mismatch\vunit_out\test_output\mwe.tb_direction_mismatch.test_direction_a5b1177fd10bca29c22d78c426a567db23efa40f\ghdl\tb_direction_mismatch-tb:error: simulation failed
```

In both the above cases, ModelSim does a backtrace to figure out exactly what the problem was whereas GHDL just gives a bound check failure.

GlenNicholls, 2020-09-30:
I've been messing with ModelSim and tracing these types of problems is nice because it provides a call sequence:

```
# ** Fatal: (vsim-3607) Slice range direction (downto) does not match slice prefix direction (to).
#    Time: 1925 ns  Iteration: 3  Process: /tb/scoreboard_p File: D:/dev/<path>/tb.vhd
# Fatal error in Process scoreboard_p at D:/dev/<path>/tb.vhd line 701
#
# HDL call sequence:
# Stopped at D:/dev/<path>/tb.vhd 701 Subprogram get_bbheader
# called from  D:/dev/<path>/tb.vhd 717 Subprogram get_data_field_len
# called from  D:/dev/<path>/tb.vhd 777 ForLoop loop
# called from  D:/dev/<path>/tb.vhd 1334 Subprogram check_data_field
# called from  D:/dev/<path>/tb.vhd 1551 Process scoreboard_p
```