ghdl/ghdl #1259: Race condition in `make ghdllib -j8`

forrestv, 2020-04-24:
Here's a cleaned-up build log I had where `make ghdllib -j8` failed (compiling the v0.36 branch):

```
forrest@forrest-tool:~/ghdl/work/ghdl$ make ghdllib -j8
...
mkdir grt
...
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -u -c -g -gnatec/home/forrest/ghdl/work/ghdl/./src/grt/grt.adc -gnatdY -o main.o /home/forrest/ghdl/work/ghdl/./src/grt/main.adb -Igrt
/bin/sh: 1: cd: can't cd to grt
...
ar rcv lib/ghdl/libgrt.a `sed -e "/^-/d" -e "s!^!grt/!" < grt/grt-files` \
 jumps.o times.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o grt/run-bind.o grt/main.o
...
ar: grt/main.o: No such file or directory
src/grt/Makefile.inc:109: recipe for target 'lib/ghdl/libgrt.a' failed
make: *** [lib/ghdl/libgrt.a] Error 1
```

It seems that `mkdir grt` hadn't completed by the time that `cd grt` was attempted. Because `cd grt` is followed by a semicolon and not `&&`, the failure was ignored, and gnatmake dropped `main.o` into the wrong directory. Later on, `ar` couldn't find `main.o`.

The race condition exists because the `grt/main.o` target doesn't have `grt` (which runs `mkdir grt`) as a prerequisite, so `mkdir grt` runs in parallel with the `cd grt` command.

closed 2020-04-24 by tgingold

eine, 2020-04-24:
Does this have any relation with https://github.com/ghdl/ghdl/blob/master/Makefile.in#L607-L613?

tgingold, 2020-04-24:
I don't think so.  It was only for grt.

PR #1260 (2020-04-24): Makefiles: fixed race condition in creation of "grt" directory before main.o is put in it
This closes #1259.