ghdl/ghdl #1262: VHDL 2008: Use of Generic Package Type in Entity Port

RocketRoss, 2020-04-24:
**Description**
In order to use a type that comes from a generic package as an entity's port, the package has to be instantiated in the entity's generic statement. This causes a bug printout in GHDL.

**Expected behaviour**
I expect to be able to instantiate the generic package and declare the port as the type declared in the generic package.

**How to reproduce?**
Commented out is the offensive code, and in between long dashed lines are the poor workarounds. Swap them out to have the MWE provide the printout.

**ent.vhd**
```vhd :file: ent.vhd
-----------------------------------------------------
package slv_pkg is new work.slv generic map (N=>3);
-----------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;

entity ent is
	generic(
		-----------------------------------------------------
		WIDTH: integer
		-----------------------------------------------------
		-- WIDTH: integer;
		-- package slv_pkg is new work.slv
		-- 	generic map(N => WIDTH)
		-----------------------------------------------------
	);
	port (
		o_slv: out work.slv_pkg.slv_t
	);
end ent;

architecture beh of ent is
	constant ones : std_logic_vector(WIDTH-1 downto 0) := (others => '1');
begin
	o_slv <= ones;
end architecture beh;
```

**pkg_slv.vhd**
```vhd :file: pkg_slv.vhd
library ieee;
use ieee.std_logic_1164.all;

package slv is
	generic(
		N: integer
	);
	subtype slv_t is std_logic_vector(N-1 downto 0);
end package;
```
**tb.vhd**
```vhd :file: tb.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
end entity;

architecture test of tb is
	constant w : integer := 3;

	package slv_tb_pkg is new work.slv generic map(N => w);

	component ent
		generic(
			WIDTH: integer
		);
		port (
			o_slv: out slv_tb_pkg.slv_t
		);
	end component ent;

	signal s_out : std_logic_vector(w-1 downto 0);
begin
	e : ent
		generic map(
			WIDTH => w
		)
		port map(
			o_slv => s_out
		);

	process
	begin
		wait for 1 ns;
		report integer'image(to_integer(unsigned(s_out)));
		wait;
	end process;
end architecture test;
```

```sh :image: ghdl/ghdl:buster-mcode
echo "Analyse pkg" &&
ghdl -a --std=08 pkg.vhd &&
echo "Analyse ent" &&
ghdl -a --std=08 ent.vhd && #crashes here
echo "Analyse tb" &&
ghdl -a --std=08 tb.vhd &&
echo "Elaborate" &&
ghdl -e --std=08 tb &&
echo "Execute" &&
./tb
```

**Context**
Please, provide the following information:

- OS: WSL Ubuntu 18.04
- Origin: LLVM
  - [x] Built from sources: `43263a95`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-416-g43263a95) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/d/Development/Code/HDL/bug_genpkg_type/
Command line:
/etc/ghdl-llvm-PIC/bin/ghdl1-llvm --std=08 -P/etc/ghdl-llvm-PIC/lib/ghdl/ieee/v08/ -P/etc/ghdl-llvm-PIC/lib/ghdl/ -c -fpic -o ent.o ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-sem_decls.adb:508
Call stack traceback locations:
0x56738c 0x5676c8 0x573e48 0x57ae31 0x533160 0x64d9e9 0x64f61c 0x650f15 0x4086c9 0x7f7e3d2a1b95 0x4079b8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
I indirectly got this declaration trick from `VHDL 2008: Just the new stuff` ([commented here](https://forums.xilinx.com/t5/Synthesis/How-to-use-generic-packages-as-introduced-in-VHDL-2008/m-p/705566/highlight/true#M18651)).

tgingold, 2020-04-27:
Now fixed.  Thank you for the issue.

closed 2020-04-27 by tgingold

RocketRoss, 2020-04-28:
Love it, Thanks Monsieur @tgingold.
I have a follow up!

I double checked that the tb was able to declare its signal not as a std_logic_vector of the (coincidentally) same width, but rather as the package type itself. There were no problems there!

So I forged ahead to expand the pkg to define a generically constrained array of logic_vectors. Here comes the snag: I cannot assign signals of the slv_arr type to the entity's port:
```
ent.vhd:43:37:error: can't associate 's_out_arr' with port "o_slv_arr"
ent.vhd:43:37:error: (type of 's_out_arr' is slv_arr_t)
ent.vhd:11:16:error: (type of port "o_slv_arr" is slv_arr_t)
```

This is the expanded pkg.vhd

```
library ieee;
use ieee.std_logic_1164.all;

package slv is
	generic(
		N: integer;
		M: integer
	);
	subtype slv_t is std_logic_vector(N-1 downto 0);
	type slv_arr_t is array(M-1 downto 0) of slv_t;
end package;
```

And the expanded ent-tb.vhd file:
```
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	generic(
		WIDTH: integer;
		package slv_pkg is new work.slv
			generic map(N => WIDTH, M => WIDTH)
	);
	port (
		-- o_slv_arr: out slv_pkg.slv_arr_t;
		o_slv: out slv_pkg.slv_t
	);
end ent;

architecture beh of ent is
	constant ones : std_logic_vector(WIDTH-1 downto 0) := (others => '1');
begin
	o_slv <= ones;
	o_slv_arr(0) <= ones;
end architecture beh;
-----------------------------------
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
end entity;

architecture test of tb is
	constant w : integer := 3;

	package slv_tb_pkg is new work.slv generic map(N => w, M => w);

	signal s_out : slv_tb_pkg.slv_t;
	-- signal s_out_arr : slv_tb_pkg.slv_arr_t;
begin
	e : entity work.ent
		generic map(
			WIDTH => w
		)
		port map(
			-- o_slv_arr => s_out_arr,
			o_slv => s_out
		);

	process
	begin
		wait for 1 ns;
		report integer'image(to_integer(unsigned(s_out)));
		report integer'image(to_integer(unsigned(s_out_arr(0))));
		report integer'image(to_integer(unsigned(s_out_arr(1))));
		wait;
	end process;
end architecture test;
```

Expanding the declaration of the generic map to include the internal generic package didn't change anything. This was only valid with the more verbose component declaration and instantiation it seems...

tgingold, 2020-04-28:
You're welcome!

> So I forged ahead to expand the pkg to define a generically constrained array of logic_vectors. Here comes the snag: I cannot assign signals of the slv_arr type to the entity's port:

I fear you are reaching the limit of this feature.

The type of the port in the entity ent is defined in the package slv_pkg of its generic.
The type of the signal in s_out_arr is defined in package slv_tb_pkg.  So that's not the same type.

RocketRoss, 2020-04-28:
I thought that would be the case. Thanks for the consideration! 👍