ghdl/ghdl #1263: Executable is not generated in linux environment

mgorzkowski, 2020-04-25:
**Description**
While working over the documentation (https://ghdl.readthedocs.io/en/latest/examples/quick_start/README.html#quick-start-guide), after calling `ghdl -e hello_world` **any** executable file (*.out) is not generated (in Linux), although after `ghdl -r hello_world` the program is executed properly. The rest of examples also not generate proper executable binary in the directory. The biggest problem is in the last example of the documentation where while `ghdl -r dlx` can't find any *.out file. 
Ghdl with -e, or -m flag always returns 0 code (success expected)

**Expected behaviour**
`ghdl -e hello_world` returns 0 and generate executable binary (the *.out file).
`ghdl -m dlx` returns 0 and generate executable binary (the *.out file).

**How to reproduce?**
- Install Xubuntu 20.04 LTS
- `sudo apt install ghdl`   // there is the latest 0.37 version in ubuntu repository
- work over the https://ghdl.readthedocs.io/en/latest/examples/quick_start/DLXModelSuite.html:
    - mkdir work
    - ghdl -i --workdir=work *.vhdl
    - ghdl -m --workdir=work dlx_test_behaviour // after that there is no any *.out file both in dxl and work directory
    - ghdl -r --workdir=work dlx_test_behaviour // produces

```
ghdl:error cannot open file dlx.out"
in process .dlx_test(bench).mem@memory(behavior).mem
  from work.memory(behavior).mem.load at memory-behavior.vhdl:63
ghdl:error simulation failed
```

**Context**
- OS: Xubuntu - virtual machine in VirtualBox (not VM installation was not checked, but highly likely). 
- ghdl version is 0.37
- Origin - Package manager - apt (0.37v)

eine, 2020-04-25:
@mgorzkowski, please do read the full quick start guide carefully. Pay special attention to the hints, tips and notes.

> after calling `ghdl -e hello_world` **any** executable file (*.out) is not generated (in Linux), although after `ghdl -r hello_world` the program is executed properly.

Executable files are only generated with LLVM or GCC backends. You are likely using GHDL with mcode backend. See the two last hints in https://ghdl.readthedocs.io/en/latest/examples/quick_start/hello/README.html.

> The biggest problem is in the last example of the documentation where while `ghdl -r dlx` can't find any *.out file.

This is a different issue. Read the second hint in https://ghdl.readthedocs.io/en/latest/examples/quick_start/DLXModelSuite.html: *"Just take one sample: `cp test_loop.out dlx.out`*".

mgorzkowski, 2020-04-25:
@eine Thanks a lot, After build ghdl with gcc as a backend it works pretty well.
What about to wirte that mcode doesn't support creating executable files in prop-con table? I can prepare propper PR. Thanks once again!

The issue seems to be resolved, you can close it.

eine, 2020-04-25:
> What about to wirte that mcode doesn't support creating executable files in pros-cons table? I can prepare propper PR. Thanks once again!

I believe it is already explained in the hint just below the table: https://ghdl.readthedocs.io/en/latest/getting/index.html. Nevertheless, enhancements are always welcome!

closed 2020-04-25 by eine