ghdl/ghdl #1264: synth: IIR_PREDEFINED_IEEE_NUMERIC_STD_S{R,L}L_SGN_INT

anhangaba, 2020-04-25:
Unhandled shift operation

```vhd :file: issue.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    port
        (srl_usn : out unsigned (8-1 downto 0);
         sll_usn : out unsigned (8-1 downto 0);
         srl_sgn : out   signed (8-1 downto 0);
         sll_sgn : out   signed (8-1 downto 0));
end issue;

architecture beh of issue is
begin
    srl_usn <= unsigned'(b"0000_0000") srl 1; -- work
    sll_usn <= unsigned'(b"0000_0000") sll 1; -- fail
    srl_sgn <=   signed'(b"0000_0000") srl 1; -- fail
    sll_sgn <=   signed'(b"0000_0000") sll 1; -- fail
end architecture beh;
```

```
issue.vhdl:16:40:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_SLL_UNS_INT
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-416-g43263a95) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl --synth --std=08 issue.vhdl -e issue
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-expr.adb:776 access check failed
Call stack traceback locations:
0x5564ffb689ab 0x5564ffb455d6 0x5564ffb463ce 0x5564ffb53b0c 0x5564ffb53d85 0x5564ffb428ac 0x5564ffb42a39 0x5564ffb3b306 0x5564ffba8312 0x5564ffacc60d 0x5564ffc5b5f1 0x5564ff889e0e 0x7f13f89ac021 0x5564ff88859c 0xfffffffffffffffe
******************************************************************
```

closed 2020-04-26 by tgingold

anhangaba, 2020-04-26:
With resize, I still getting the same error:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    port (uns_inp : in  unsigned (4-1 downto 0);
          uns_sll : out unsigned (8-1 downto 0);
          uns_srl : out unsigned (8-1 downto 0);
          sgn_inp : in    signed (4-1 downto 0);
          sgn_sll : out   signed (8-1 downto 0);
          sgn_srl : out   signed (8-1 downto 0));
end issue;

architecture beh of issue is
begin
    uns_sll <= resize (uns_inp, sgn_srl'length) sll 1; -- work
    uns_srl <= resize (uns_inp, sgn_srl'length) srl 1; -- work
    sgn_sll <= resize (sgn_inp, sgn_srl'length) sll 1; -- error
    sgn_srl <= resize (sgn_inp, sgn_srl'length) srl 1; -- error
end architecture beh;
```

```
issue.vhdl:18:49:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_SLL_SGN_INT
issue.vhdl:19:49:error: synth_dyadic_operation: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_SRL_SGN_INT
```

tgingold, 2020-04-26:
This one is now OK.  This wasn't exactly the same error: the former was for static values, the latter for dynamic values.

anhangaba, 2020-04-26:
Didn't realized the the error msgs were different but makes sense.
Thanks for the fix!

tgingold, 2020-04-26:
You;'re welcome!

commit 5d26e8e0 (2020-04-26): synth-static_oper: handle sll/srl for signed/unsigned.  Fix #1264