ghdl/ghdl #1268: weird bug when using OSVVM inside protected type (VHDL2008)

GlenNicholls, 2020-04-27:
When using protected types, I am seeing a weird bug when using `RandomPType` variables within a procedure. I have attached the MWE which uses vunit to expose the bug.

I would expect that performing `rnd.InitSeed(rnd'instance_name);` would properly initialize the OSVVM seed after declaring `variable rnd : RandomPType` inside a procedure within a protected type.

To reproduce, run the VUnit `run.py` file within the zipped file
[RandomPType_in_protected.zip](https://github.com/ghdl/ghdl/files/4540823/RandomPType_in_protected.zip)

```
library ieee;
context ieee.ieee_std_context;

library osvvm;
use osvvm.RandomPkg.all;

package mwe_pkg is
    type mwe_t is protected
        procedure send_random_data (
            constant num_bits : integer range 1 to 32 := 1
        );

    end protected;

end package;

package body mwe_pkg is
    type mwe_t is protected body
        procedure send_random_data (
            constant num_bits : integer range 1 to 32 := 1
        ) is
            variable rnd : RandomPType;
        begin
            -- init random seed
            rnd.InitSeed(rnd'instance_name);
        end procedure;
    end protected body;

end package body;
```

GHDL output:

```
=== Command used: ===
D:\ghdl\ghdl-0.37-llvmx64\bin\ghdl -a --workdir=D:\ghdl\mwe\vunit_out\ghdl\libraries\mwe --work=mwe --std=08 -PD:\ghdl\mwe\vunit_out\ghdl\libraries\vunit_lib -PD:\ghdl\mwe\vunit_out\ghdl\libraries\osvvm -PD:\ghdl\mwe\vunit_out\ghdl\libraries\mwe --ieee=synopsys -frelaxed-rules D:\ghdl\mwe\mwe_pkg.vhd

=== Command output: ===

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1084-g5a57a31b) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-w64-mingw32
D:\ghdl\mwe\
Command line:
D:\ghdl\ghdl-0.37-llvmx64\bin\ghdl1-llvm.exe --workdir=D:\ghdl\mwe\vunit_out\ghdl\libraries\mwe --work=mwe --std=08 -frelaxed-rules -PD:\ghdl\mwe\vunit_out\ghdl\libraries\vunit_lib\ -PD:\ghdl\mwe\vunit_out\ghdl\libraries\osvvm\ -PD:\ghdl\mwe\vunit_out\ghdl\libraries\mwe\ -PD:\ghdl\ghdl-0.37-llvmx64\lib/ghdl\synopsys\v08\ -PD:\ghdl\ghdl-0.37-llvmx64\lib/ghdl\ -c -o D:\ghdl\mwe\vunit_out\ghdl\libraries\mwe\mwe_pkg.o D:\ghdl\mwe\mwe_pkg.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:3516
Call stack traceback locations:
0x1767bb6 0x47adbb 0x4e83f0 0x4e84de 0x4e8a85 0x4e8ad0 0x4e7a85 0x5f4f18 0x5d428f 0x589256 0x58612c 0x58b536 0x593ca4 0x593d41 0x616c08 0x5b9938 0x60a689 0x5b9991 0x617d82 0x569eef 0x62687e 0x62819d 0x629bb5 0x4026cf 0x4013b2 0x401509 0x7ff930417bd2 0x7ff930b2cecf
******************************************************************
D:\ghdl\ghdl-0.37-llvmx64\bin\ghdl.exe: compilation error
```

- OS: Windows 10

tgingold, 2020-04-27:
Which version of osvvm are you using ?

GlenNicholls, 2020-04-27:
I am using VUnit v4.4.0, so whichever revision of OSVVM it checks out. I just checked VUnit and it doesn't look like they rev-tie OSVVM, so it should be https://github.com/OSVVM/OSVVM/commit/185a11d7bbf896e5fb66035cbad7fcb5ab505c79 since I installed v4.4.0 in March.

closed 2020-04-28 by tgingold

GlenNicholls, 2020-04-28:
@tgingold Just out of curiosity, what was the problem?

tgingold, 2020-04-29:
The problem was not related to osvvm nor to protected type, but to the creation of the signature for a subprogram.

commit a006ad6b (2020-04-28): vhdl: fix handling of types name in name attributes.  Fix #1268