ghdl/ghdl #1270: Error when initializing array of records

umarcor, 2020-04-28:
The following package is successfully analyzed:

```vhdl
library ieee;
context ieee.ieee_std_context;

package vga is
  type cfg_t is record
    WrReq : natural;
    RdReq : std_logic;
  end record;

  type cfg_arr_t is array (natural range <>) of cfg_t;
  shared variable pairs_in: cfg_arr_t(0 to 0) := (0 => (640, '0'));
end vga;
```

```sh
# ghdl -a --std=08 -frelaxed bug.vhdl
bug.vhdl:13:19:warning: type of a shared variable must be a protected type [-Wshared]
```

However, if a field is removed, it fails:

```vhdl
library ieee;
context ieee.ieee_std_context;

package vga is
  type cfg_t is record
    WrReq : natural;
  end record;

  type cfg_arr_t is array (natural range <>) of cfg_t;
  shared variable pairs_in: cfg_arr_t(0 to 0) := (0 => (640));
end vga;
```

```sh
# ghdl -a --std=08 -frelaxed bug.vhdl
bug.vhdl:11:57:error: type of element not compatible with the expected type
bug.vhdl:11:19:warning: type of a shared variable must be a protected type [-Wshared]
ghdl:error: compilation error
```

The same applies when `WrReq` is removed, instead of `RdReq`:

```vhdl
library ieee;
context ieee.ieee_std_context;

package vga is
  type cfg_t is record
    RdReq : std_logic;
  end record;

  type cfg_arr_t is array (natural range <>) of cfg_t;
  shared variable pairs_in: cfg_arr_t(0 to 0) := (0 => ('0'));
end vga;
```

```sh
# ghdl -a --std=08 -frelaxed bug.vhdl
bug.vhdl:11:57:error: type of element not compatible with the expected type
bug.vhdl:11:19:warning: type of a shared variable must be a protected type [-Wshared]
ghdl:error: compilation error
```

I tried with a constrained array too, with no success:

```vhdl
library ieee;
context ieee.ieee_std_context;

package vga is
  type cfg_t is record
    WrReq : natural;
  end record;

  type cfg_arr_t is array (0 to 0) of cfg_t;
  shared variable pairs_in: cfg_arr_t := (0 => (640));
end vga;
```

Moreover, I'd like to initialize an unconstrained array as in the following snippet:

```vhdl
library ieee;
context ieee.ieee_std_context;

package vga is
  type cfg_t is record
    WrReq : natural;
  end record;

  type cfg_arr_t is array (natural range <>) of cfg_t;
  shared variable pairs_in: cfg_arr_t := ((640),(800));
end vga;
```

which produces the same error.

tgingold, 2020-04-28:
>   
> shared variable pairs_in: cfg_arr_t(0 to 0) := (0 => (640
> ));

Here, `(640)` is not an aggregate but an expression in parenthesis.

You should write `(0 => (WrReq => 640))`

umarcor, 2020-04-28:
> Here, `(640)` is not an aggregate but an expression in parenthesis. You should write `(0 => (WrReq => 640))`

Thanks for explaining. In my real use case, the record contains multiple items. Hence, I guess this is a corner case I found when trying to write the MWE.

FTR, from LRM 2008 section 9.3.3.1: *"Aggregates containing a single element association
shall always be specified using named association in order to distinguish them from parenthesized
expressions"*.

What about the last example? Is it possible to have an unconstrained array constrained by the number of items that are declared as the initial content?

```vhdl
  type cfg_arr_t is array (natural range <>) of cfg_t;
  shared variable pairs_in: cfg_arr_t := (
    (WrReq => 640),
    (WrReq => 800)
  );
```

According to section *14.4.2.5 Object declarations*, *"the subtype indication is first elaborated"* and *"initialization (...) involves a check that the initial value belongs to the subtype of the object*". Hence, I think it not possible, but I'm unsure.

closed 2020-04-28 by umarcor

umarcor, 2020-05-08:
FTR, it is possible in VHDL 2008 if a `constant` is used, instead of a `shared variable`. In VHDL 2019, the latter is supported too.