ghdl/ghdl #1273: synth: asserts statements on procedure body are not included in generated code

anhangaba, 2020-04-29:
Asserts statements inside procedure body are not included in the generated code.

```vhd file:issue.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity issue is
    port (foo : in  std_logic);
end entity issue;

architecture beh of issue is
    procedure check (arg : in std_logic) is
    begin
        assert (arg xor '1') = (arg and '0');
    end procedure;
begin
    check (foo);
    --assert (foo xor '1') = (foo and '0');
end architecture;
```

```
ghdl --synth -fpsl issue.vhdl -e issue
```

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of issue is
  signal wrap_foo: std_logic;
begin
  wrap_foo <= foo;
end rtl;
```
When the equivalent assert statements is in the architecture body, they are presented in the generated code:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of issue is
  signal wrap_foo: std_logic;
  signal n3_o : std_logic;
  signal n5_o : std_logic;
  signal n6_o : std_logic;
begin
  wrap_foo <= foo;
  -- issue.vhdl:15:17
  n3_o <= wrap_foo xor '1';
  -- issue.vhdl:15:33
  n5_o <= wrap_foo and '0';
  -- issue.vhdl:15:26
  n6_o <= '1' when n3_o = n5_o else '0';
  -- issue.vhdl:15:5
  n7: assert n6_o = '1' severity error;
end rtl;
```

tgingold, 2020-04-29:
Yes, sequential assertions are currently ignored.
I have a plan to handle them.

closed 2020-05-06 by tgingold

tgingold, 2020-05-06:
Now implemented.

commit 3b70e630 (2020-05-06): synth: add support for sequential assertions.  Fix #1273