ghdl/ghdl #1274: constant literal assignment to an aggregate

anhangaba, 2020-05-01:
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    port (foo    : out unsigned (1 downto 0);
          bar    : out unsigned (2 downto 0));
end entity issue;

architecture beh of issue is
begin
    assert 5ub"11_000" = unsigned'(b"11_000");

    --(foo, bar) <= unsigned'(b"11_000"); -- works

    (foo, bar) <= 5ub"11_000";
end architecture;
```
```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 -fpsl issue.vhdl -e issue
```
The assertion holds true.

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 issue.vhdl
```

Both ghdl commands produces an error:
```
issue.vhdl:16:5:error: cannot resolve type of waveform
```

Is this the expected behavior?

anhangaba, 2020-05-01:
Also. Trying to assign expression with with bit string literal to aggregate causes a crash.

```vhdl file:issue2.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    port (foo    : out unsigned (1 downto 0);
          bar    : out unsigned (2 downto 0);
          foobar : out unsigned (4 downto 0));
end entity issue;

architecture beh of issue is
begin
    assert 5ub"11_000" = unsigned'(b"11_000");
    foobar <= 5ub"11_000" + 5ub"00_000"; -- works
    (foo, bar) <= 5ub"11_000" + 5ub"00_000"; -- crashes
end architecture;
```

Report:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-451-ge3eb603c) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-pc-linux-gnu
***
Command line:
ghdl -a --std=08 issue.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : no field Base_Type
Call stack traceback locations:
0x55ff69d4fcd2 0x55ff69aee30f 0x55ff69bdcdc6 0x55ff69bb3275 0x55ff69bb3c75 0x55ff69bb77e2 0x55ff69bb7b36 0x55ff69bb7c9c 0x55ff69bf22e5 0x55ff69bf9440 0x55ff69bae0b7 0x55ff69c6dbba 0x55ff69ba5707 0x55ff69d35629 0x55ff69961e1e 0x7f65eefda021 0x55ff6996059c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-05-01:
> issue.vhdl:16:5:error: cannot resolve type of waveform
> 
> Is this the expected behavior?

Yes, this is the expected behaviour.  You indeed need to qualify the expression.

tgingold, 2020-05-02:
Thank for the issue.  Interesting testcase!

closed 2020-05-02 by tgingold

anhangaba, 2020-05-03:
Thank you for the fix :)

commit 6fe280aa (2020-05-02): vhdl-sem_expr: fix handling of overload lists in merge_wildcard_type. Fix #1274