ghdl/ghdl #1275: synth: PSL assertions without always are active at each cycle

tmeissner, 2020-05-02:
**Description**
PSL assertions without always are active at each cycle.

Instead they are active at each cycle:

```
sby --yosys "yosys -m ghdl" -f -d work/psl_always psl_always.sby prove
SBY 20:42:38 [work/psl_always] Removing directory 'work/psl_always'.
SBY 20:42:38 [work/psl_always] Copy '../src/sequencer.vhd' to 'work/psl_always/src/sequencer.vhd'.
SBY 20:42:38 [work/psl_always] Copy '../src/psl_always.vhd' to 'work/psl_always/src/psl_always.vhd'.
SBY 20:42:38 [work/psl_always] engine_0: smtbmc z3
SBY 20:42:38 [work/psl_always] base: starting process "cd work/psl_always/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 20:42:38 [work/psl_always] base: finished (returncode=0)
SBY 20:42:38 [work/psl_always] smt2: starting process "cd work/psl_always/model; yosys -m ghdl -ql design_smt2.log design_smt2.ys"
SBY 20:42:38 [work/psl_always] smt2: finished (returncode=0)
SBY 20:42:38 [work/psl_always] engine_0: starting process "cd work/psl_always; yosys-smtbmc -s z3 --presat --noprogress -t 25 --append 0 --dump-vcd engine_0/trace.vcd --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Solver: z3
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assumptions in step 0..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assertions in step 0..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assumptions in step 1..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assertions in step 1..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assumptions in step 2..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Checking assertions in step 2..
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  BMC failed!
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Assert failed in psl_always: without_always_a
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace.vcd
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace_tb.v
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace.smtc
SBY 20:42:38 [work/psl_always] engine_0: ##   0:00:00  Status: failed
SBY 20:42:38 [work/psl_always] engine_0: finished (returncode=1)
SBY 20:42:38 [work/psl_always] engine_0: Status returned by engine: FAIL
SBY 20:42:38 [work/psl_always] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:42:38 [work/psl_always] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:42:38 [work/psl_always] summary: engine_0 (smtbmc z3) returned FAIL
SBY 20:42:38 [work/psl_always] summary: counterexample trace: work/psl_always/engine_0/trace.vcd
SBY 20:42:38 [work/psl_always] DONE (FAIL, rc=2)
SBY 20:42:38 One or more tasks produced a non-zero return code.
```

**Expected behaviour**
PSL assertions without always should only be active at cycle 0. Instead they are active at each cycle, which is a bug. An assertion without always which holds on cycle 0 should hold "forever".

**How to reproduce?**

```vhd :file: sequencer.vhd
library ieee;
  use ieee.std_logic_1164.all;

entity sequencer is
  generic (
    seq : string
  );
  port (
    clk  : in  std_logic;
    data : out std_logic
  );
end entity sequencer;

architecture rtl of sequencer is

  signal index : natural := seq'low;
  signal ch    : character;

  function to_bit (a : in character) return std_logic is
    variable ret : std_logic;
  begin
    case a is
      when '0' | '_' => ret := '0';
      when '1' | '-' => ret := '1';
      when others    => ret := 'X';
    end case;
    return ret;
  end function to_bit;

begin

  process (clk) is
  begin
    if rising_edge(clk) then
      if (index < seq'high) then
        index <= index + 1;
      end if;
    end if;
  end process;

  ch <= seq(index);

  data <= to_bit(ch);

end architecture rtl;
```

```vhd :file: psl_always.vhd
library ieee;
  use ieee.std_logic_1164.all;

entity psl_always is
  port (
    clk : in std_logic
  );
end entity psl_always;

architecture psl of psl_always is

  component sequencer is
    generic (
      seq : string
    );
    port (
      clk  : in  std_logic;
      data : out std_logic
    );
  end component sequencer;

  signal a : std_logic;

begin

  --                            012345
  SEQ : sequencer generic map ("--____") port map (clk, a);

  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);

  -- This should hold, but fails at cycle 2
  WITHOUT_ALWAYS_a : assert a;

end architecture psl;

```

```file: psl_always.sby
[tasks]
prove

[options]
depth 25
prove: mode bmc

[engines]
prove: smtbmc z3

[script]
prove: ghdl --std=08 sequencer.vhd psl_always.vhd -e psl_always
prep -top psl_always

[files]
../src/pkg.vhd
../src/sequencer.vhd
../src/psl_always.vhd
```

```sh :image: ghdl/synth:formal
sby --yosys "yosys -m ghdl" -f -d work/psl_always psl_always.sby prove
```

**Context**
Please, provide the following information:

- OS: linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Docker (Image: ghdl/synth:formal)

anhangaba, 2020-05-03:
I think the psl assertion syntax is:

```vhdl
--psl assert a;
```

The code generated with `--psl assert` is different from the one with `assert`. Haven't tried to verify the code thought.

tgingold, 2020-05-03:
Yes, `assert a;` is considered as a non-PSL assertion, so as a standard concurrent assertion.

I agree that this could be considered as a pitfall!

tmeissner, 2020-05-03:
Ups, you‘re right. I remember the section in The Designers Guide Of VHDL08. Would be interesting what’s the behavior when it the assert is located in a PSL vunit. There can also be VHDL code in it.

closed 2020-05-03 by tmeissner

anhangaba, 2020-05-03:
I have very limited experience with verification. But if this a common occurrence, maybe, GHDL could throw a note when a code compiled with `-fpsl` (or some other flag) has an ambiguous assertion.

tmeissner, 2020-05-03:
You don't have to compile with extra flags when using inline PSL with VHDL-Standard >= 08 as PSL was integrated in VHDL with this revision. If you would emit a warning, you would get it for every VHDL assertion outside processes, the only difference is the option to set a severity which don't exist for PSL asserts.

tgingold, 2020-05-04:
Would it be useful to add a warning if a default clock is present ?