ghdl/ghdl #1278: The path of VPI modules is not valid on MINGW

umarcor, 2020-05-03:
The following MWE, extracted from ghdl/ghdl-cosim#21, works on Linux, but not on Windows (MSYS2, v0.37). It seems to be an issue with the format of the path/location passed through `--vpi`.

```c :file: vpi.c
#include <vpi_user.h>
#include <stdio.h>

PLI_INT32 cb_hello(){
  printf("Hello!\n");
  return 0;
}

void entry_point_cb() {
  s_cb_data cb;

  cb.reason = cbStartOfSimulation;
  cb.cb_rtn = &cb_hello;
  cb.user_data = NULL;

  if (vpi_register_cb(&cb) == NULL) {
    vpi_printf ("cannot register cbStartOfSimulation call back\n");
  }
}

// List of entry points called when the plugin is loaded
void (*vlog_startup_routines[]) () = {entry_point_cb, 0};
```

```vhdl :file: tb.vhd
entity tb is
end entity;

architecture arch of tb is
begin
  process
  begin
    report "Hello!" severity note;
    wait;
  end process;
end architecture;
```

```sh :file: run.sh
#!/usr/bin/env sh

set -e

cd $(dirname "$0")

ghdl -a tb.vhd
ghdl -e -o tb tb
ghdl --vpi-compile gcc -c vpi.c -o vpi.o
ghdl --vpi-link gcc vpi.o -o vpi.vpi

set +e

export PATH=$PATH:$(pwd)

echo "> Execute tb"

./tb --vpi=vpi.vpi
./tb --vpi=./vpi.vpi
./tb --vpi="$(pwd)/vpi.vpi"
./tb --vpi="/$(pwd)/vpi.vpi"
./tb --vpi="$(echo $(pwd)/vpi.vpi | sed 's#/#\\#g')"
./tb --vpi="$(echo /$(pwd)/vpi.vpi | sed 's#/#\\#g')"
```

```
loading VPI module 'vpi.vpi'
The specified module could not be found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module  
loading VPI module './vpi.vpi'
The specified module could not be found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module  
loading VPI module 'D:/ghdl-cosim/vpi/quickstart/debug/vpi.vpi'
The specified module could not be found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module  
loading VPI module '//d/ghdl-cosim/vpi/quickstart/debug/vpi.vpi'
The network path was not found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module
loading VPI module '\d\ghdl-cosim\vpi\quickstart\debug\vpi.vpi'
The specified module could not be found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module  
loading VPI module '\\d\ghdl-cosim\vpi\quickstart\debug\vpi.vpi'
The network path was not found.

D:\ghdl-cosim\vpi\quickstart\debug\tb.exe:error: cannot load VPI module
```

umarcor, 2020-05-03:
As seen in https://github.com/ghdl/ghdl/blob/master/testsuite/gna/issue98/testsuite.sh and https://github.com/ghdl/ghdl/blob/master/testsuite/testenv.sh, the issue is that `libghdlvpi.dll` needs to be available in the PATH. Adding `PATH="$PATH:`ghdl --vpi-library-dir | sed -e 's!\\\\!/!g' -e 's!^C:!/C!g'`"` before executing the simulation fixed it.

closed 2020-05-03 by umarcor

tgingold, 2020-05-03:
I am not a Windows or MSYS2 expert but AFAIK:
* either the file must be in the path
* or the file must be an absolute windows file (with backslashes !)

In your tests, you never pass an absolute windows file. Well, you tried, but...
MSYS2 tries to be nice and transforms windows paths to unix-like paths.

You should either try with PowerShell or cmd.
Or you should try to to set:
export MSYS_ARG_CONV_EXCL="--vpi"
before using an absolute path, so that the shell won't change the path after '--vpi='

umarcor, 2020-05-03:
Actually, it's a little more tricky than that...

- `PATH="$PATH:$(ghdl --vpi-library-dir | sed -e 's!\\\\!/!g' -e 's!^C:!/C!g')"` does solve the issue on my host. https://github.com/ghdl/ghdl-cosim/pull/21/files#diff-b4717ee3b922c0c25649639de0c91524 works ok, as is.
- The same sources, fail in CI: https://github.com/ghdl/ghdl-cosim/pull/21/checks?check_run_id=639740762#step:3:36

The same happens with gna/issue98:

- It is executed in CI in this repo, as part of gna, and it passes.
- Locally, I can `cd testsuite/gna/issue98 && GHDL=ghdl ./testsuite.sh`, and it passes.
- However, the same `cd testsuite/gna/issue98 && GHDL=ghdl ./testsuite.sh` fails in CI: https://github.com/umarcor/ghdl-cosim/runs/639908489?check_suite_focus=true#step:5:24

umarcor, 2020-05-03:
So, it did not work because the `sed` command above assumes that GHDL is installed in `C:`, which is not the case in CI. Using `PATH=$PATH:"$(cd $(dirname $(which ghdl))/../lib; pwd)"` instead solves the issue. For completeness:

```sh
# ghdl --vpi-library-dir
C:\msys64\mingw64\lib

# ghdl --vpi-library-dir-unix
/C//msys64/mingw64/lib

# echo "`ghdl --vpi-library-dir | sed -e 's!\\\\!/!g' -e 's!^C:!/C!g'`"
/C/msys64/mingw64/lib

# echo "$(cd $(dirname $(which ghdl))/../lib; pwd)"
/mingw64/lib
```

I'd suggest to use the second or the third.

tgingold, 2020-05-03:
Good catch.  The second solution looks the simplest one!

umarcor, 2020-05-03:
Yes! That's the one I used in ghdl/ghdl-cosim#21. I also added an admonition to the docs.