ghdl/ghdl #1281: Document --time-resolution

umarcor, 2020-05-04:
`--time-resolution` was added in #613.

While writing [Reading waveforms from HDL simulators with PulseView](https://github.com/umarcor/pulseview/tree/ghdl/ghdl) it was not easy to conclude that:

- `--time-resolution` is supported by backend mcode only.
- `--time-resolution` is supported by `-a` and `-r` only. `ghdl -e --time-resolution` produces a failure, even with mcode backend.
- mcode backend, by default, sets the time scale automatically depending on the smallest time unit in the design.
- Units below an overriden time resolution do produce an expected failure.

Before writing this in the docs, I'd like to know:

- Is it possible to have the auto-scaling feature available with LLVM backend by default?
- Can `--time-resolution` be allowed for `-e` (with mcode backend), so that the same `GHDL_ARGS` can be used for `-a`, `-e` and `-r`?

Ref #403 #672

tgingold, 2020-05-04:
`--time-resolution` needs to apply globally, so that's not possible with LLVM or gcc backends.

I think it is allowed by `-e`, but it must appear before `-e`.

Otherwise, the description is correct.

umarcor, 2020-05-04:
> `--time-resolution` needs to apply globally, so that's not possible with LLVM or gcc backends.

:sad:

> I think it is allowed by `-e`, but it must appear before `-e`. Otherwise, the description is correct.

It produces a different error, but none is supported:

```sh
root@d43464f68a53:/src# ./run.sh 
run_sim: ns auto
> Analyze tb_ns.vhd (--time-resolution=auto)
ghdl:error: unknown command '--time-resolution=auto', try --help
root@d43464f68a53:/src# ./run.sh 
run_sim: ns auto
> Analyze tb_ns.vhd (--time-resolution=auto)
ghdl:error: unknown option '--time-resolution=auto'
```

umarcor, 2020-05-04:
Would it make sense for LLVM/GCC backends to use `--time-resolution` for downsampling the waveforms in-memory, even if the simulation resolution is fixed to `fs`?

tgingold, 2020-05-04:
You mean, while dumping waveforms ?

Humm, that's very pulseview specific, isn't it ?

umarcor, 2020-05-04:
In fact... PulseView has a downsampling option for VCD imports. Hence, it is not required for GHDL to do it. I was proposing it for consistency of the waveform generation options across backends.

tgingold, 2020-05-06:
But the behaviour won't be the same.  With the current --time-resolution, time literals are modified.
If we try to downsample, we could easily create a test that shows the effect of this feature: create a waveform with delays that are rounded, and check the result using 'last_event.

PR #1290 (2020-05-08): Update documentation
Fix #1281

This PR includes several minor fixes and enhancements:

- A minor fix in the readme.
- Removing `development/Documentation.rst`, which was unreferenced and is now in the [wiki](https://github.com/ghdl/ghdl/wiki/Ideas:-Documentation).
- Moving `doc/examples/quick_start` to `doc/quick_start`, because there are no other examples.
- Document option `-o`.
- Document option `--time-resolution`.
- Update the year in the copyright notice to 2002-2020.
- Update my contributor role.
- Add a reference to the wiki in section 'Contributing'.
closed 2020-05-08 by tgingold