ghdl/ghdl #1284: wait for 0 ns does not wait for 1 delta cycle

GlenNicholls, 2020-05-04:
During simulation, `wait for 0 ns` does not wait for a single delta, but instead, waits for 0 simulation time.

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process 
    variable v_now : time;
  begin
    v_now := now;
    wait for 0 ns;
    v_now := now;
    assert v_now = 1 fs
      report "Simulator did not wait 1 delta, got now=" & time'image(v_now)
      severity failure;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

yields

```
ent.vhd:12:5:@0ms:(assertion failure): Simulator did not wait 1 delta, got now=0 fs
.\ent:error: assertion failed
in process .ent(a).P0
.\ent:error: simulation failed
```

- OS: windows 10
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `e3eb603c`

diogratia, 2020-05-04:
> I thought that any wait for was always a minimum of a single delta cycle, so I had a procedure to wait for a delta cycle that recently broke some tests that I ran in GHDL. GHDL appeared to hang, but when I turned reporting on, it never actually hung but instead, the simulation time just never changed.  

>I know I can simply write wait for 1 fs, but not all simulators default to min time res of 1 fs. Because of this, I have to instead call a function that recursively searches each time unit to see which is 0.

There seems to be a misconception here. Simulation time doesn't advance.

IEEE Std 1076-2008 10.2 Wait statement:

>The timeout clause specifies the maximum amount of time the process will remain suspended at this wait statement. If no timeout clause appears, the timeout clause for (STD.STANDARD.TIME'HIGH – STD.STANDARD.NOW) is assumed. It is an error if the time expression in the timeout clause evaluates to a negative value.  
>  
>The execution of a wait statement causes the time expression to be evaluated to determine the timeout interval. It also causes the execution of the corresponding process statement to be suspended, where the corresponding process statement is the one that either contains the wait statement or is the parent (see 4.3) of the procedure that contains the wait statement. The suspended process will resume, at the latest, immediately after the timeout interval has expired.

The maximum amount of time the process suspended in the wait statement was 0 ns (which is also 0 fs). Such wait statements are used to insure a delta cycle occurs.

14.7.5 Model Execution, 14.7.5.1 General:

>At certain stages during the initialization phase and each simulation cycle, the current time, Tc, and the time of the next simulation cycle, Tn, are calculated. Tn is calculated by setting it to the earliest of  
>>a) TIME'HIGH,  
>>b) The next time at which a driver or signal becomes active,  
>>c) The next time at which a process resumes, or  
>>d) The next time at which a registered and enabled vhpiCbAfterDelay, vhpiCbRepAfterDe- lay, vhpiCbTimeOut, or vhpiCbRepTimeOut callback is to occur.  
>  
>If Tn = Tc, then the next simulation cycle (if any) will be a *delta cycle*.

A VHDL simulation cycle has distinct ordered steps which do not require a delta cycle consume a minimum interval simulation time. See 14.7.5.3 Simulation cycle (-1993 12.6.4 Simulation cycle  doesn't include PSL and VHPI steps and may be easier to interpret).

Annex I (Informative) Glossary:
>**delta cycle:** A simulation cycle in which the simulation time at the beginning of the cycle is the same as at the end of the cycle. That is, simulation time is not advanced in a delta cycle. Only nonpostponed processes can be executed during a delta cycle. (14.7.5.1)

Here you could remove he second assignment to `v_now` and use `now = v_now` as the assertion expression.

10.3 Assertion statement

>Execution of an assertion statement consists of evaluation of the Boolean expression specifying the condition. If the expression results in the value FALSE, then an *assertion violation* is said to occur. When an assertion violation occurs, the **report** and **severity** clause expressions of the corresponding assertion, if present, are evaluated. The specified message string and severity level (or the corresponding default values, if not specified) are then used to construct an error message.

A simulator which such an *assertion violation* would not be compliant with any revision of the VHDL standard.

tgingold, 2020-05-05:
So no, ghdl is correct here.

I suppose you are using a time resolution less than fs on other simulators.

GlenNicholls, 2020-05-06:
Alright, I will close out this issue while I investigate further. If I come up with something else, I'll open a new issue. Thank you for the thorough explanation @diogratia 

On a separate note, is there anywhere that the VHDL LRM(s) are openly available?

closed 2020-05-06 by GlenNicholls

eine, 2020-05-06:
> On a separate note, is there anywhere that the VHDL LRM(s) are openly available?

I think it might be posible in a future, but not today. Not officially, at least: https://standards.ieee.org/content/ieee-standards/en/standard/1076-2019.html. 5 months ago, libraries were open sourced. I'd expect the standard itself to be open sourced in the next revision. However, this is my desire only.

tmeissner, 2020-05-06:
Some time ago IEEE gave away the standards of Verilog and VHDL for "free" during some days. You could download them from the website. But this ended. I don't know if it is possible (legal) to give away an document which someone has downloaded that time.

The PDF from the working group is a document with marked changes over the standard before. It's also useful, but a little confusing to read, maybe.