ghdl/ghdl #1286: good first issue label

m-kru, 2020-05-06:
@tgingold do you think you could spent some time adding `good first issue`  labels to the issues?

tgingold, 2020-05-06:
I have added the label, and adding it to a few issues.

If you are interested in contributing, do not hesitate to join the gitter channel (if not already done) and to explain what you'd like to do or/and what is your know-how.

Even if you only know C++ and don't want to work with Ada, there are some tasks to do (like a write_vhdl command for yosys, or improving the LLVM backend).

I know, I am not very good at enrolling people!

tmeissner, 2020-05-06:
@tgingold Another hint: You can pin issues, so that they appear on top of the issue page. It looks like this: https://github.com/tmeissner/psl_with_ghdl/issues (Please ignore the fact that my repo only has one issue which is assigned to myself ;))

Maybe you could pin the most interesting issues or that ones where others could help as they aren't Ada specific.

m-kru, 2020-05-07:
@tgingold First I would to like to see if I am even capable of contribution. Is there any good starting point to read about the structure of the project? What tools do you use? Do you use any IDE (GNAT Studio) or terminal emulator is your swiss knife?

tgingold, 2020-05-07:
There is a small section in the doc about the internals of ghdl.

Most of the time I use emacs but build within a terminal emulator.  That's not advanced tooling!

tgingold, 2020-05-07:
Thanks, done!

(What about your PSL issues? :-)

tgingold, 2020-05-07:
Other possible contributions area:

* Integration with vendor libraries.  Paebbels contributed a lot on that and the effort should be continued.

* Installers

* The ghdl language server could be interesting if you prefer to work with python or TypeScript.

* A logo ?

eine, 2020-05-07:
For the record, there is https://ghdl.readthedocs.io/en/latest/development/Roadmap.html. Precisely https://ghdl.readthedocs.io/en/latest/development/Documentation.html and https://ghdl.readthedocs.io/en/latest/development/GSOC.html. The purpose of those sections is not only to list what can be done, but to gather related issues/refs, so that whoever willing to contribute does not start from scratch. There are also pinned issues in [https://github.com/ghdl/ghdl-cosim](ghdl/ghdl-cosim).

I think we'd better homogeneize where we write this kind of info. I'm ok with:

- Adding the missing ideas from this issue to the "Roadmap" or "GSOC" sections of the docs.
- Renaming the "Roadmap" and "GSOC" section names of the docs, to better reflect that those are "areas where contributions are welcome".
- Using the wiki of this repo instead.
- Using one issue for each area/topic instead.
- Leaving everything as is, but renaming this issue and making it the first pinned one.

> Most of the time I use emacs but build within a terminal emulator. That's not advanced tooling!

I think that section https://ghdl.readthedocs.io/en/latest/development/Debugging.html could benefit from a description of either the scripts or the workflow you use. It is not a requirement to work as you do, but I believe there are some "tricks", such as loading symbols in GDB, which are not documented. Those are useful for contributing to GHDL itself, to VPI, VHPIDIRECT...

> Even if you only know C++ and don't want to work with Ada, there are some tasks to do 

I think we need to copy this sentence with large capital red letters and some neon lights in a very visible place. Well, replacing "C++" with "C/C++, Python, JavaScript/TypeScript, etc.".

tgingold, 2020-05-08:
I have created a wiki page and moved the roadmap and GSOCs to it.

As the 'good first issue' label has been created, I am closing this issue.

closed 2020-05-08 by tgingold

cmarqu, 2020-05-11:
BTW, GitHub has permission level "Triage" since a while (https://github.blog/changelog/2019-05-23-triage-and-maintain-roles-beta/) which allows selected users that are not (yet?) "Collaborators" to set issue labels and milestones

eine, 2020-05-11:
@cmarqu, actually, I was a "triager" in this repo for several months. I.e., I could label and close issues, but I could not e.g. create/edit labels. Now, I have some more permissions, but my role is still limited to being a triager and handling some PRs which are very specifically about docs and/or CI. Precisely, yesterday I replicated a subset of the labels from this repo in https://github.com/ghdl/ghdl-cosim/labels.

I think it would be desirable if other users could help with triaging because I am sometimes unsure about how to label an issue until it is solved and I can see how Tristan handled it. However, we don't have any communication channel which is exclusive for maintainers; hence I guess than any user willing to contribute in this regard should get in touch with Tristan, either privately or in the chat room.