ghdl/ghdl #1287: Declaration of array of record subtype triggers bug.

benreynwar, 2020-05-07:
**Description**
Ghdl fails when an array of a record subtype is created.
This occurs whether the bounds on the record are defined in a subtype declaration or in the array type declaration itself.

**Expected behaviour**
Not failing.  An error message that says a feature is not supported would be fine.

**How to reproduce?**
``` vhdl
entity myentity is
end myentity;

architecture arch of myentity is
  type myrecord is record
    a : bit_vector;
  end record;
  subtype myboundedrecord is myrecord(a(1 downto 0));
  type myarray is array (natural range <>) of myboundedrecord;
begin
end arch;
```

```
ben@bookbinder:~/Code/ghdl/testsuite/vpi/vpi009$ ghdl -a --std=08 mydesign.vhdl
ben@bookbinder:~/Code/ghdl/testsuite/vpi/vpi009$ ghdl -r --std=08  myentity

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-501-g9ebdd46d) [Dunoon edition]
Compiled with GNAT Version: 8.4.0
Target: x86_64-linux-gnu
/home/ben/Code/ghdl/testsuite/vpi/vpi009/
Command line:
ghdl -r --std=08 myentity
Exception ORTHO_CODE.SYNTAX_ERROR raised
Exception information:
raised ORTHO_CODE.SYNTAX_ERROR : ortho_code-exprs.adb:464
```
- OS: Ubuntu
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x ] Built from sources: `commit SHA` (see in output above)

eine, 2020-05-07:
I think this is due to partial support of unconstrained arrays. Hence, I believe that "fixing" all these cases to produce a nice error is not much easier than actually fixing them. IIRC, Tristan said that he wanted to focus on this after he is happy with synthesis.

Ref #756, #821, #852, #899, #961, #1201, #780, #1055, #1224... Maybe we should have a specific label about this. However, TBH, I don't feel myself confident enough to actually apply that label to new issues.

benreynwar, 2020-05-07:
That makes sense.  I don't need the feature at all, I just came across it because I wanted to understand how I should deal with this for the vpi interface.

eine, 2020-08-05:
I think that this issue is fixed now. I cannot reproduce the crash with `v0.37.0-924-g36f25166` (either LLVM or mcode). Please, reopen if that is not the case.

closed 2020-08-05 by eine