ghdl/ghdl #1288: PSL: psl.sem_property: cannot handle N_NEXT_E

tmeissner, 2020-05-07:
First real issue resulting from my work at the [psl_with_ghdl](https://github.com/tmeissner/psl_with_ghdl) project :)

**Description**
The PSL next_e operator isn't supported yet. If you use next_e in your code, you get an error like this:

```
psl.sem_property: cannot handle N_NEXT_E
```

**Expected behaviour**
The next_e PSL operator should be supported and work as defined in the IEEE standard for PSL. Taken from the 2010 version, it is mentioned in section 6.2.1.4.2:

> The next_e family of operators, shown in Syntax 6-24, specify that an FL Property holds at least once within some range of future cycles.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: psl_next_e.vhd
library ieee;
  use ieee.std_logic_1164.all;

entity psl_next_e is
  port (
    clk : in std_logic
  );
end entity psl_next_e;

architecture psl of psl_next_e is

  signal a, b : std_logic;
  signal c, d : std_logic;
  signal e, f : std_logic;

begin

  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);

  -- This assertion holds
  NEXT_0_a : assert always (a -> next_e[3 to 5] (b));

  -- This assertion doesn't hold at cycle 6
  NEXT_1_a : assert always (c -> next_e[3 to 5] (d));

  -- This assertion holds
  NEXT_2_a : assert always (e -> next_e[3 to 5] (f));

end architecture psl;

```

```sh :image:ghdl/synth:formal
ghdl -a --std=08 psl_next_e.vhd 
```

The same error also occurs when using GHDL synthesis:
```
ghdl --synth --std=08 psl_next_e.vhd -e psl_next_e
```

**Context**

- OS: Debian Buster
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [ ] Docker (Image: ghdl/synth:formal)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
psl.sem_property: cannot handle N_NEXT_E

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/build/formal/
Command line:
/opt/ghdl/bin/ghdl1-llvm --std=08 -P/opt/ghdl/lib/ghdl/ieee/v08/ -P/opt/ghdl/lib/ghdl/ -c -fpic -o psl_next_e.o ../src/psl_next_e.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-errors.adb:39
Call stack traceback locations:
0x44286c 0x534d75 0x534aaf 0x53485d 0x534923 0x535715 0x53c195 0x53c29c 0x53c3f3 0x57433b 0x57b030 0x532b80 0x64ccc0 0x64e8c2 0x650171 0x409fdb 0x7f339e7eb099 0x409178 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error

```

tmeissner, 2020-05-07:
I've browsed a little through the GHDL code for PSL and found code for next_e at many places. But I don't find what's left to make it working ...

tmeissner, 2020-05-08:
I've created a branch in my GHDL fork with the small change described above. I've ran my [example](https://github.com/tmeissner/psl_with_ghdl/blob/master/src/psl_next_e.vhd) for the PSL `next_e` operator with GHDL built from that fork and it seems to work with correct behaviour.

I can create a PR with the change, but before I would like to hear the opinion of @tgingold to that change. And if there is something left to do regarding the `FIXME: range` comment.

tgingold, 2020-05-08:
Your fix looks ok.  As long as the ranges consist of numbers, they don't need to be analyzed.

So feel free to create a PR!

closed 2020-05-08 by tmeissner