ghdl/ghdl #1292: PSL: psl.sem_property: cannot handle N_NEXT_EVENT_E

tmeissner, 2020-05-08:
Next issue resulting from my work at the [psl_with_ghdl](https://github.com/tmeissner/psl_with_ghdl) project :)

**Description**
The PSL `next_event_e` operator isn't supported yet. If you use `next_event_e ` in your code, you get an error like this:

```
psl.sem_property: cannot handle N_NEXT_EVENT_E
```

**Expected behaviour**
The `next_event_e ` PSL operator should be supported and work as defined in the IEEE standard for PSL. Taken from the 2010 version, it is mentioned in section 6.2.1.4.5:

> The next_event_e family of operators, shown in Syntax 6-27, specify that an FL Property holds at least once during a range of next occurrences of a Boolean expression. The next occurrences of the Boolean expression include an occurrence at the current cycle.

**How to reproduce?**

```vhd :file: psl_next_e.vhd
library ieee;
  use ieee.std_logic_1164.all;

entity sequencer is
  generic (
    seq : string
  );
  port (
    clk  : in  std_logic;
    data : out std_logic
  );
end entity sequencer;

architecture rtl of sequencer is

  signal index : natural := seq'low;
  signal ch    : character;

  function to_bit (a : in character) return std_logic is
    variable ret : std_logic;
  begin
    case a is
      when '0' | '_' => ret := '0';
      when '1' | '-' => ret := '1';
      when others    => ret := 'X';
    end case;
    return ret;
  end function to_bit;

begin

  process (clk) is
  begin
    if rising_edge(clk) then
      if (index < seq'high) then
        index <= index + 1;
      end if;
    end if;
  end process;

  ch <= seq(index);

  data <= to_bit(ch);

end architecture rtl;

library ieee;
  use ieee.std_logic_1164.all;

entity psl_next_e is
  port (
    clk : in std_logic
  );
end entity psl_next_e;

architecture psl of psl_next_e is

  component sequencer is
    generic (
      seq : string
    );
    port (
      clk  : in  std_logic;
      data : out std_logic
    );
  end component sequencer;

  signal a, b, c : std_logic;

begin

  --                              012345678901234
  SEQ_A : sequencer generic map ("_-______-______") port map (clk, a);
  SEQ_B : sequencer generic map ("___-__-___-__-_") port map (clk, b);
  SEQ_C : sequencer generic map ("______-___-____") port map (clk, c);

  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);

  -- This assertion holds
  NEXT_EVENT_a : assert always (a -> next_event_e(b)[1 to 2](c));

end architecture psl;

```

```sh :image:ghdl/synth:formal
ghdl -a --std=08 psl_next_event_e.vhd 
```

The same error also occurs when using GHDL synthesis:
```
ghdl --synth --std=08 psl_next_event_e.vhd -e psl_next_event_e
```

**Context**

- OS: Debian Buster
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Docker (Image: ghdl/synth:formal)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
psl.sem_property: cannot handle N_NEXT_EVENT_E

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-507-ge9ec6bb7-dirty) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/Users/torsten/Projects/git.goodcleanfun.de/psl_with_ghdl/src/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o psl_next_event_e.o psl_next_event_e.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : psl-errors.adb:39
Load address: 0x100cc6000
Call stack traceback locations:
0x100cf9aa4 0x100e01e17 0x100e01b51 0x100e01867 0x100e019c5 0x100e02853 0x100e3e432 0x100e3e53c 0x100e3e691 0x100df9741 0x100e00a8c 0x100de5905 0x100f03eab 0x100f056b9 0x100f06b56 0x100cc9208
******************************************************************
ghdl:error: compilation error
```

tmeissner, 2020-05-08:
I tried a fix with additions to `vhdl-sem_psl.adb` similar to the changes for next_e, but they don't work. In function

```vhdl
Sem_Property (Prop : PSL_Node; Top : Boolean := False)
                     return PSL_Node
```

Following additions in the case:

```vhdl
         when N_Next_Event_A | N_Next_Event_E =>
            --  FIXME: range.
            -- Sem_Number (Prop);
            Sem_Boolean (Prop);
            Sem_Property (Prop);
            return Prop;
```

This don't works. The error stays the same. Maybe I'm too naive to fix it with that small change. However, as for `next_e`, a lot of code handling `next_event_e` & `next_event_a` already exists in other source files.

tgingold, 2020-05-09:
> This don't works. The error stays the same. Maybe I'm too naive to fix it with that small change. However, as for next_e, a lot of code handling next_event_e & next_event_a already exists in other source files.

Are you sure you recompiled ghdl ?  Because it cannot be the same exact error: you cannot get
```
psl.sem_property: cannot handle N_NEXT_EVENT_E
```
as that case is handled.

I have merge your change and I can pass the test!

closed 2020-05-09 by tgingold

tmeissner, 2020-05-09:
Yes, I‘ve tried several times: make clean && make && sudo make install. It didn’t worked. Strange. It was not possible because the case was handled. I have to check my local GHDL source tree, maybe better check out master new.

Thanks for the merge 🙂

tmeissner, 2020-05-09:
It seems that I have two versions of GHDL on my Mac, one in `/usr/local/bin`, another in `/opt/ghdl`. The first one is in the path and is used. But the compile result was installed in the second path. I have to find out, why this happened, maybe I have to re-run configure to set the PREFIX for the install step. It worked correctly during fixing the `next_e `operator before ... strange - again 🙂

commit a3edc868 (2020-05-09): vhdl-sem_psl: handle next_event_a and next_event_e.  Fix #1292

From T. Meissner.