ghdl/ghdl #1298: The specification of unsupported kind of generics produce 'GHDL Bug occurred'

rodrigomelo9, 2020-05-10:
Hi. I am adding GHDL support (the synthesis feature and the ghdl-yosys-plugin) into my project related to support from synthesis to bitstream generation in a vendor-independent way ([PyFPGA](https://gitlab.com/rodrigomelo9/pyfpga)). One of the supported features is to specify generics/parameters, and I was playing with this feature for synthesis.

* Firstly, take into account that the doc says (https://ghdl.readthedocs.io/en/latest/using/Simulation.html) that it is a simulation feature, but it works for synthesis :-D
* I succeeded in to specify booleans, integers, std_logic and strings.
* In the cases of std_logic_vectors and reals, it failed (the second one specifying `unhandled override`).

It is not a request for support, I reported it because it finishes with *Please report this bug...*.

The following files are part of my project ([here](https://gitlab.com/rodrigomelo9/pyfpga/-/tree/ghdl/poc/params)).

`generics.vhdl`:
```
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity Params is
   generic (
      BOO : boolean:=FALSE;
      INT : integer:=0;
      LOG : std_logic:='0';
      VEC : std_logic_vector(7 downto 0):="00000000";
      STR : string:="ABCD";
      REA : real:=0.0
   );
   port (
      boo_o : out std_logic;
      int_o : out std_logic_vector(7 downto 0);
      log_o : out std_logic;
      vec_o : out std_logic_vector(7 downto 0);
      str_o : out std_logic;
      rea_o : out std_logic
   );
end entity Params;

architecture RTL of Params is
begin

   assert BOO=True       report "The boolean is not True" severity failure;
   assert INT=255        report "The integer is not 255" severity failure;
   assert LOG='1'        report "The std_logic is not '1'" severity failure;
   assert VEC="11111111" report "The std_logic_vector is not 11111111" severity failure;
   assert STR="WXYZ"     report "The string is not WXYZ" severity failure;
   assert REA=1.1        report "The real is not 1.1" severity failure;

   boo_o <= '1' when BOO else '0';
   int_o <= std_logic_vector(to_unsigned(INT, 8));
   log_o <= LOG;
   vec_o <= VEC;
   str_o <= '1' when STR="WXYZ" else '0';
   rea_o <= '1' when REA=1.1 else '0';

end architecture RTL;
```

`ghdl.sh`:
```
#!/bin/bash

FLAGS="-fsynopsys -fexplicit -frelaxed --std=08"

ghdl -a $FLAGS ../../hdl/fakes/generics.vhdl
ghdl --synth $FLAGS -gBOO=true -gINT=255 -gLOG=\'1\' -gSTR="WXYZ" Params

ghdl --synth $FLAGS -gBOO=true -gINT=255 -gLOG=\'1\' -gSTR="WXYZ" -gVEC="11111111" Params
ghdl --synth $FLAGS -gBOO=true -gINT=255 -gLOG=\'1\' -gSTR="WXYZ" -gVEC=\"11111111\" Params
ghdl --synth $FLAGS -gBOO=true -gINT=255 -gLOG=\'1\' -gSTR="WXYZ" -gREA=1.1 Params
```

The output:
```
$ make ghdl 
bash ghdl.sh
../../hdl/fakes/generics.vhdl:30:4:error: assertion failure: The std_logic_vector is not 11111111
../../hdl/fakes/generics.vhdl:32:4:error: assertion failure: The real is not 1.1
../../hdl/fakes/generics.vhdl:30:4:error: assertion failure: The std_logic_vector is not 11111111
../../hdl/fakes/generics.vhdl:32:4:error: assertion failure: The real is not 1.1

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-484-gfccf50dc) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
/home/ram/repos/pyfpga/poc/params/
Command line:
ghdl --synth -fsynopsys -fexplicit -frelaxed --std=08 -gBOO=true -gINT=255 -gLOG='1' -gSTR=WXYZ -gVEC=11111111 Params
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-expr.adb:214
Call stack traceback locations:
0x7fbd90e818d1 0x55c394ae7571 0x55c394ae7f75 0x55c394ae862e 0x55c394afd23b 0x55c394afdb4c 0x55c394afdbc7 0x55c394a0d0aa 0x55c394a0d721 0x55c394acd345 0x55c394acd5a3 0x55c394ae6613 0x55c394ae677a 0x55c394a1de68 0x55c394bdc489 0x55c3949e426d 0x55c3949ebca3 0x55c39487a47d 0x7fbd902862df 0x55c394878d58 0xfffffffffffffffe
******************************************************************
../../hdl/fakes/generics.vhdl:30:4:error: assertion failure: The std_logic_vector is not 11111111
../../hdl/fakes/generics.vhdl:32:4:error: assertion failure: The real is not 1.1

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-484-gfccf50dc) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
/home/ram/repos/pyfpga/poc/params/
Command line:
ghdl --synth -fsynopsys -fexplicit -frelaxed --std=08 -gBOO=true -gINT=255 -gLOG='1' -gSTR=WXYZ -gVEC="11111111" Params
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-expr.adb:214
Call stack traceback locations:
0x7f9db249d8d1 0x55f11cc06571 0x55f11cc06f75 0x55f11cc0762e 0x55f11cc1c23b 0x55f11cc1cb4c 0x55f11cc1cbc7 0x55f11cb2c0aa 0x55f11cb2c721 0x55f11cbec345 0x55f11cbec5a3 0x55f11cc05613 0x55f11cc0577a 0x55f11cb3ce68 0x55f11ccfb489 0x55f11cb0326d 0x55f11cb0aca3 0x55f11c99947d 0x7f9db18a22df 0x55f11c997d58 0xfffffffffffffffe
******************************************************************
ghdl:error: unhandled override for generic "rea"

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-484-gfccf50dc) [Dunoon edition]
Compiled with GNAT Version: 6.3.0
Target: x86_64-linux-gnu
/home/ram/repos/pyfpga/poc/params/
Command line:
ghdl --synth -fsynopsys -fexplicit -frelaxed --std=08 -gBOO=true -gINT=255 -gLOG='1' -gSTR=WXYZ -gREA=1.1 Params
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : vhdl-nodes.adb:420 index check failed
Call stack traceback locations:
0x5594b864d00a 0x5594b871de7f 0x5594b871e04a 0x5594b87ebb12 0x5594b87ec417 0x5594b85f426d 0x5594b85fbca3 0x5594b848a47d 0x7f39477b62df 0x5594b8488d58 0xfffffffffffffffe
******************************************************************
Makefile:6: fallo en las instrucciones para el objetivo 'ghdl'
make: *** [ghdl] Error 2
```

eine, 2020-05-10:
> * Firstly, take into account that the doc says (https://ghdl.readthedocs.io/en/latest/using/Simulation.html) that it is a simulation feature, but it works for synthesis :-D

For clarification, the docs do not say that it is a simulation feature *only*. However, I acknowledge that it might not be easy to reach the conclusion. Please, bear with me:

> https://ghdl.readthedocs.io/en/latest/using/Synthesis.html#synthesis
> Note
> Due to GHDL’s modular architecture (see [Overview](https://ghdl.readthedocs.io/en/latest/internals/Overview.html#int-overview)), the synthesis kernel shares the VHDL parsing front-end with the simulation back-ends. Hence, available options for synthesis are the same as for analysis and/or simulation elaboration (see [Options](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#ghdl-options)).

> https://ghdl.readthedocs.io/en/latest/using/Simulation.html#cmdoption-ghdl-ggeneric
> Note
> This is currently a run option; but in the (not near) future it might be deprecated to become an elaboration option only. As a result, now you can generate a single binary and execute it multiple times with different arguments. That might not be possible in the future.
> As explained in [-e](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-e), performing a complete elaboration in terms of the LRM requires to get rid of the compile and link model. This is mostly because delaying certain elaboration steps to the runtime prevents elaboration-time optimisions.

> https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#cmdoption-ghdl-e
> Warning
> This elaboration command is not a complete elaboration in terms of the VHDL standard. The actual elaboration is performed at runtime. Therefore, in order to get a complete VHDL elaboration without running the simulation, ghdl --elab-run --no-run is required. See --no-run.

So, since both analysis/elaboration options and runtime options are required for a complete elaboration in terms of the VHDL standard, it is coherent that "runtime options that make sense during an elaboration for synthesis, are supported OOTB".

What do you think about the following enhancements?

- Rename section https://ghdl.readthedocs.io/en/latest/using/Simulation.html#simulation-runtime from `Simulation (runtime)` to `Runtime`.
  - In https://ghdl.readthedocs.io/en/latest/using/Synthesis.html#synthesis, add a reference to it: *"(see [Options](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#ghdl-options) and [Runtime](https://ghdl.readthedocs.io/en/latest/using/Simulation.html#simulation-runtime))"*
- Move `-gGENERIC=VALUE` out from subsection `Simulation options`. Should be create a `Common runtime options` subsection or should we omit the first subheading?

@tgingold, is the following grouping correct?

- Common: `-gGENERIC`, `--assert-level`, `--ieee-asserts`, `--unbuffered` and `--help`.
- Simulation only: `--stop-time`, `--stop-delta`, `--disp-time`, `--max-stack-alloc`, `--sdf`, `--vpi`, `--vpi-trace` and `--no-run`.

> * I succeeded in to specify booleans, integers, std_logic and strings.
> 
> * In the cases of std_logic_vectors and reals, it failed (the second one specifying `unhandled override`).

You can use any type for generics, and you can use them for top-level generics, as long as you provide a default value in VHDL. However, the types of top-level generics that can be overriden in the CLI is limited. Note the hint:

> https://ghdl.readthedocs.io/en/latest/using/Simulation.html#cmdoption-ghdl-ggeneric
> Currently, GHDL has limited support for generic types in the CLI. It is suggested to use strings or integers. Nonetheless, project [JSON-for-VHDL](https://github.com/Paebbels/JSON-for-VHDL) allows to encode a set of parameters as stringified JSON, and it provides VHDL functions to read specific values from it. It is valid for synthesis.

Hence, you might want to try JSON-for-VHDL if you need to pass non-trivial generics trhough CLI.

See also https://github.com/VUnit/vunit/blob/master/tests/acceptance/artificial/vhdl/run.py#L63-L66 and VUnit/vunit#588.

rodrigomelo9, 2020-05-10:
Hi @eine. I understand your point, but I expected to found this under [Options](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#options) when invoking GHDL.

I think that the rename from `Simulation (runtime)` to `Runtime` is not needed.

I will take a look to JSON-for-VHDL, thanks (I read the hint also, so it is well located in the doc).

eine, 2020-05-10:
> Hi @eine. I understand your point, but I expected to found this under [Options](https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#options) when invoking GHDL.

We need to tell "command options" and "runtime options" apart, because the position in the CLI commands is different. See:

> https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#run-r
> -r <[options...] primary_unit [secondary_unit] [simulation_options...]>

> https://ghdl.readthedocs.io/en/latest/using/InvokingGHDL.html#elaborate-and-run-elab-run
>  --elab-run <[options...] primary_unit [secondary_unit] [simulation_options...]>

All the options in `Options` are allowed before `primary_unit [secondary_unit]` but not after them. Conversely, `Simulation (runtime)` options are allowed after them only.

Hence, if we moved `-gGENERIC` to `Options`, users would try to incorrectly use it for simulation. This happened before and lead to multiple misunderstandings. However, the fact that `-gGENERIC` is expected before the primary unit in `--synth` is certainly not coherent. @tgingold, any thought?

commit 53914c09 (2020-05-13): vhdl-configuration: handle overrides of slv.  Fix #1298
closed 2020-05-13 by tgingold