ghdl/ghdl #1300: Unconstrained array in a record specialization bug

ozbenh, 2020-05-10:
**Description**
In microwatt, I try to define the wishbone types records containing unconstrained vectors for address/data/sel along with specialized variants using subtypes with explicit ranges.

This is causing a STORAGE_ERROR in ghdl when compiling the icache which is a user of those types.
The error is:
```
ghdl -a --std=08 -Psim-unisim --workdir=. icache.vhdl

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-533-g06202188) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/ANT.AMAZON.COM/benh/hackplace/microwatt/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 --workdir=. -Psim-unisim/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o ./icache.o icache.vhdl
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7fe0f6fddf1e 0x7fe0f8b83b22 0x7fe0f8ac75f0 0x41802b 0x6214ea 0x5b9540 0x5c6f08 0x5c7006 0x6475cf 0x58b7ad 0x64e742 0x65001e 0x651917 0x4086a9 0x7fe0f6fc0b95 0x407998 0xfffffffffffffffe
******************************************************************
```
**How to reproduce?**
I haven't yet managed to reproduce a small case but it's easy to reproduce if you can download microwatt source and just do "make" after applying this patch:
```
--- a/wishbone_types.vhdl
+++ b/wishbone_types.vhdl
@@ -2,15 +2,7 @@ library ieee;
 use ieee.std_logic_1164.all;
 
 package wishbone_types is
-    constant wishbone_addr_bits : integer := 32;
-    constant wishbone_data_bits : integer := 64;
-    constant wishbone_sel_bits : integer := wishbone_data_bits/8;
-
-    subtype wishbone_addr_type is std_ulogic_vector(wishbone_addr_bits-1 downto 0);
-    subtype wishbone_data_type is std_ulogic_vector(wishbone_data_bits-1 downto 0);
-    subtype wishbone_sel_type  is std_ulogic_vector(wishbone_sel_bits-1  downto 0);
-
-    type wb_master_out is record
+    type wb_master_out_t is record
         adr : std_ulogic_vector;
         dat : std_ulogic_vector;
         cyc : std_ulogic;
@@ -19,15 +11,33 @@ package wishbone_types is
         we  : std_ulogic;
     end record;
 
-    subtype wishbone_master_out is wb_master_out(adr(31 downto 0), dat(63 downto 0), sel(7 downto 0));
-    constant wishbone_master_out_init : wishbone_master_out := (cyc => '0', stb => '0', we => '0', others => (others => '0'));
-
-    type wishbone_slave_out is record
-        dat   : wishbone_data_type;
+    type wb_slave_out_t is record
+        dat   : std_ulogic_vector;
         ack   : std_ulogic;
         stall : std_ulogic;
     end record;
-    constant wishbone_slave_out_init : wishbone_slave_out := (ack => '0', stall => '0', others => (others => '0'));
+
+    -- Common subtypes
+    constant wb_cpu_addr_bits : integer := 32;
+    constant wb_cpu_data_bits : integer := 64;
+    subtype wb_cpu_addr_t is std_ulogic_vector(wb_cpu_addr_bits-1 downto 0);
+    subtype wb_cpu_data_t is std_ulogic_vector(wb_cpu_data_bits-1 downto 0);
+    subtype wb_cpu_out_t is wb_master_out_t(adr(wb_cpu_addr_bits-1 downto 0),
+                                            dat(wb_cpu_data_bits-1 downto 0),
+                                            sel(wb_cpu_data_bits/8-1 downto 0));
+    subtype wb_cpu_in_t  is wb_slave_out_t(dat(wb_cpu_data_bits-1 downto 0));
+
+    -- "Old" compat types
+    constant wishbone_addr_bits : integer := wb_cpu_addr_bits;
+    constant wishbone_data_bits : integer := wb_cpu_data_bits;
+    constant wishbone_sel_bits : integer := wishbone_data_bits/8;
+
+    subtype wishbone_addr_type is wb_cpu_addr_t;
+    subtype wishbone_data_type is wb_cpu_data_t;
+    subtype wishbone_sel_type  is std_ulogic_vector(wishbone_sel_bits-1  downto 0);
+
+    subtype wishbone_master_out is wb_cpu_out_t;
+    subtype wishbone_slave_out  is wb_cpu_in_t;
 
     type wishbone_master_out_vector is array (natural range <>) of wishbone_master_out;
     type wishbone_slave_out_vector is array (natural range <>) of wishbone_slave_out;
```
(I'll 

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

ozbenh, 2020-05-10:
Sorry, I accidentally submitted before I finished typing. GHDL is from today's git:
```GHDL 1.0-dev (v0.37.0-533-g06202188) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
```

ozbenh, 2020-05-10:
Ah ! If I change it so that the "old" types wishbone_* aren't defined as subtypes of wb_cpu_* but directly as subtypes of the unconstrainted ones, it works ! as follow:
```
    subtype wishbone_master_out is wb_master_out_t(adr(wb_cpu_addr_bits-1 downto 0),
                                            dat(wb_cpu_data_bits-1 downto 0),
                                            sel(wb_cpu_data_bits/8-1 downto 0));
    subtype wishbone_slave_out  is wb_slave_out_t(dat(wb_cpu_data_bits-1 downto 0));
--    subtype wishbone_master_out is wb_cpu_out_t;
--    subtype wishbone_slave_out  is wb_cpu_in_t;
```

tgingold, 2020-05-10:
The patch does not apply (hint: wishbone_master_out element adr and dat were constrained).
I am trying to manually fix.

ozbenh, 2020-05-10:
My apologies, I sent the wrong patch :-( clearly I was tired. I'll upload the right one asap

ozbenh, 2020-05-11:
```
--- a/wishbone_types.vhdl
+++ b/wishbone_types.vhdl
@@ -2,30 +2,49 @@ library ieee;
 use ieee.std_logic_1164.all;
 
 package wishbone_types is
-    constant wishbone_addr_bits : integer := 32;
-    constant wishbone_data_bits : integer := 64;
-    constant wishbone_sel_bits : integer := wishbone_data_bits/8;
-
-    subtype wishbone_addr_type is std_ulogic_vector(wishbone_addr_bits-1 downto 0);
-    subtype wishbone_data_type is std_ulogic_vector(wishbone_data_bits-1 downto 0);
-    subtype wishbone_sel_type  is std_ulogic_vector(wishbone_sel_bits-1  downto 0);
-
-    type wishbone_master_out is record
-        adr : wishbone_addr_type;
-        dat : wishbone_data_type;
+    type wb_master_out_t is record
+        adr : std_ulogic_vector;
+        dat : std_ulogic_vector;
         cyc : std_ulogic;
         stb : std_ulogic;
-        sel : wishbone_sel_type;
+        sel : std_ulogic_vector;
         we  : std_ulogic;
     end record;
-    constant wishbone_master_out_init : wishbone_master_out := (cyc => '0', stb => '0', we => '0', others => (others => '0'));
 
-    type wishbone_slave_out is record
-        dat   : wishbone_data_type;
+    type wb_slave_out_t is record
+        dat   : std_ulogic_vector;
         ack   : std_ulogic;
         stall : std_ulogic;
     end record;
-    constant wishbone_slave_out_init : wishbone_slave_out := (ack => '0', stall => '0', others => (others => '0'));
+
+    -- Common subtypes
+    constant wb_cpu_addr_bits : integer := 32;
+    constant wb_cpu_data_bits : integer := 64;
+    subtype wb_cpu_addr_t is std_ulogic_vector(wb_cpu_addr_bits-1 downto 0);
+    subtype wb_cpu_data_t is std_ulogic_vector(wb_cpu_data_bits-1 downto 0);
+    subtype wb_cpu_out_t is wb_master_out_t(adr(wb_cpu_addr_bits-1 downto 0),
+                                            dat(wb_cpu_data_bits-1 downto 0),
+                                            sel(wb_cpu_data_bits/8-1 downto 0));
+    subtype wb_cpu_in_t  is wb_slave_out_t(dat(wb_cpu_data_bits-1 downto 0));
+
+    -- "Old" compat types
+    constant wishbone_addr_bits : integer := wb_cpu_addr_bits;
+    constant wishbone_data_bits : integer := wb_cpu_data_bits;
+    constant wishbone_sel_bits : integer := wishbone_data_bits/8;
+
+    subtype wishbone_addr_type is wb_cpu_addr_t;
+    subtype wishbone_data_type is wb_cpu_data_t;
+    subtype wishbone_sel_type  is std_ulogic_vector(wishbone_sel_bits-1  downto 0);
+
+    -- GHDL: Works
+--    subtype wishbone_master_out is wb_master_out_t(adr(wb_cpu_addr_bits-1 downto 0),
+--                                            dat(wb_cpu_data_bits-1 downto 0),
+--                                            sel(wb_cpu_data_bits/8-1 downto 0));
+--    subtype wishbone_slave_out  is wb_slave_out_t(dat(wb_cpu_data_bits-1 downto 0));
+
+    -- GHDL: Breaks
+    subtype wishbone_master_out is wb_cpu_out_t;
+    subtype wishbone_slave_out  is wb_cpu_in_t;
 
     type wishbone_master_out_vector is array (natural range <>) of wishbone_master_out;
     type wishbone_slave_out_vector is array (natural range <>) of wishbone_slave_out;
```

tgingold, 2020-05-11:
Ok, I can reproduce the crash.

tgingold, 2020-05-13:
I think this is now fixed.  Thank you for the report.

closed 2020-05-13 by tgingold