ghdl/ghdl #1303: Error when running Altera CycloneII Post-Synthesis Simulation Model

ttobsen, 2020-05-11:
**Description**
When running a post-synthesis simulation model (created using eda_writa with Altera Quartus 13.0sp1) I got the error:

```
/opt/altera/13.0sp1/quartus/eda/sim_lib/cycloneii_atoms.vhd:8020:11: VITAL port type must be Std_Logic_Vector or Std_Ulogic
/opt/altera/13.0sp1/quartus/eda/sim_lib/cycloneii_atoms.vhd:8021:11: VITAL port type must be Std_Logic_Vector or Std_Ulogic
/opt/altera/13.0sp1/quartus/eda/sim_lib/cycloneii_atoms.vhd:1674:8: first declaration must be the VITAL attribute specification
```
The elab-run is called using following settings

```
ghdl --elab-run --std=08 -frelaxed-rules --ieee=synopsys
```

**Expected behaviour**

The current design is under NDA, but I can create some new one if really needed.

**Context**

- OS: Arch Linux
- Origin:
  - [x] Package manager: `0.37`
  - [ ] Released binaries: `?`
  - [ ] Built from sources: `?`

ttobsen, 2020-05-11:
Ok, I've found the `--no-vital-checks` flags and the simulation is running now. Sorry for spamming the issue system. :-(

closed 2020-05-11 by ttobsen

tgingold, 2020-05-11:
No problems.  That's indeed to easiest way to deal with VITAL errors.