ghdl/ghdl #1307: Multiple Driver Nets error in ghdl+yosys+Vivado build

wtatarski, 2020-05-12:
**Description**
I want to build this design https://github.com/wtatarski/Artix-7-HDMI-processing using ghdl + ghdl-yosys-plugin + yosys + Vivado (I use 2017.2 version). I managed to build design using only Vivado, but for flow with ghdl I'm getting about 80 errors related to ``ERROR: [DRC MDRV-1] Multiple Driver Nets``. I suspect that this is a bug in one of the tools 

**Expected behaviour**
I want to discover which tool causes these errors and fix it. Probably this bug is in ghdl, ghdl-yosys-plugin or yosys.

**How to reproduce?**
I provide example as simple as I could. I removed parts of the code to have error related to only one signal. [code.zip](https://github.com/ghdl/ghdl/files/4617178/code.zip)

```
yosys -m ghdl -p 'ghdl -fsynopsys -fexplicit -i hdmi_design.vhd hdmi_io.vhd conversion_to_RGB.vhd  -e hdmi_design; synth_xilinx -edif hdmi_design.edif'
```
Output edif should used for creating Vivado project together with `NexysVideo.xdc` for `xc7a200tfbg484-1` part.
You should get errors like this:
```
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_hdmi_io/i_conversion_to_rgb/out_r[0] has multiple drivers: i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$853/Q, and i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$865/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_hdmi_io/i_conversion_to_rgb/out_r[1] has multiple drivers: i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$854/Q, and i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$866/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_hdmi_io/i_conversion_to_rgb/out_r[2] has multiple drivers: i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$855/Q, and i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$867/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_hdmi_io/i_conversion_to_rgb/out_r[3] has multiple drivers: i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$856/Q, and i_hdmi_io/i_conversion_to_rgb/$auto$simplemap.cc:420:simplemap_dff$868/Q.
```

If you would like to build the whole design, download code from [here](https://github.com/wtatarski/Artix-7-HDMI-processing)
and building it with:
```
yosys -m ghdl -p 'ghdl -Punisim -fsynopsys -fexplicit -i src/symbol_dump.vhd ./src/line_delay.vhd ./src/pixel_processing.vhd ./src/tmds_decoder.vhd ./src/expand_422_to_444.vhd ./src/edid_rom.vhd ./src/hdmi_input.vhd ./src/extract_video_infopacket_data.vhd ./src/audio_to_db.vhd  src/hdmi_design.vhd ./src/guidelines.vhd ./src/alignment_detect.vhd ./src/serialiser_10_to_1.vhd ./src/deserialiser_1_to_10.vhd ./src/detect_interlace.vhd ./src/tmds_encoder.vhd ./src/dvid_output.vhd ./src/hdmi_io.vhd ./src/conversion_to_RGB.vhd ./src/extract_audio_samples.vhd ./src/edge_enhance.vhd ./src/audio_meters.vhd  ./src/input_channel.vhd -e hdmi_design; synth_xilinx -edif hdmi_design.edif'
```
You need precompiled unisim primitives.
Afterwards you need to create Vivado project with `hdmi_design.edif` and `NexysVideo.xdc`

umarcor, 2020-05-12:
Just as alternatives to find the issue, you might try:

- Using `ghdl --synth` instead of Yosys. That is ghdl + Vivado only.
- Simulating the design with GHDL first. Maybe you need `-frelaxed` too.

tgingold, 2020-05-12:
Could you also file the generated edif file ?
The one I get doesn't have multiple driver (by reading it, I have to use vivado on it).

tgingold, 2020-05-12:
If you use the `-flatten` option for `synth_xilinx`, vivado accepts the result.

The multiple driver nets were those that aren't connected.  I have the feeling that something is wrong either in the edif output or in the edif import of Vivado.

You should `write_verilog` after ghdl import and report the issue to Yosys.

daveshah1, 2020-05-12:
Just to note that a similar issue was reported previously as YosysHQ/yosys#1733, it was fixed but it is quite possible there are more EDIF related problems.

wtatarski, 2020-05-13:
@tgingold  
Thanks for your advice!
It looks like `-flatten` solved my testcase, but for the whole design still I'm getting the same kind of errors (multiple driver nets). The amount of errors decreased to 35.
So you think this is a bug in Yosys?

tgingold, 2020-05-13:
I think that vivado doesn't correctly reads the edif file, so maybe yosys has to adapt.

wtatarski, 2020-05-14:
I've reported an issue to Yosys, but I created one more testcase in vhdl. In that case `-flatten` doesn't help. I'm getting the following errors:
```
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_20] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1513/Q, and $auto$simplemap.cc:420:simplemap_dff$1505/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_21] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1514/Q, and $auto$simplemap.cc:420:simplemap_dff$1506/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_22] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1507/Q, and $auto$simplemap.cc:420:simplemap_dff$1515/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_23] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1516/Q, and $auto$simplemap.cc:420:simplemap_dff$1508/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_24] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1509/Q, and $auto$simplemap.cc:420:simplemap_dff$1517/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_25] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1510/Q, and $auto$simplemap.cc:420:simplemap_dff$1518/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_26] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1519/Q, and $auto$simplemap.cc:420:simplemap_dff$1511/Q.
ERROR: [DRC MDRV-1] Multiple Driver Nets: Net i_line_delay_27] has multiple drivers: $auto$simplemap.cc:420:simplemap_dff$1520/Q, and $auto$simplemap.cc:420:simplemap_dff$1512/Q.
report_drc (run_mandatory_drcs) completed successfully
INFO: [Project 1-461] DRC finished with 8 Errors
```
You may want to take a look at this.
[testcase2.zip](https://github.com/ghdl/ghdl/files/4629619/testcase2.zip)

tgingold, 2020-05-14:
I fear that as long as you have unconnected ports you might have issue.
Check the last lines of `synth_xilinx -edif xxx` report.

wtatarski, 2020-05-15:
Your statement ins not fully true. You can try to comment one of assignment in edge_enhance.vhd file and the error disappears.  But warnings in yosys are still present.
Also I spotted that removing unused signals in my testcase solves errors in Vivado. That's really strange.

tgingold, 2020-05-27:
I am closing this issue as the error is not related to vhdl: the netlist generated by ghdl doesn't contain multiple drivers.

closed 2020-05-27 by tgingold