ghdl/ghdl #1308: synth-objtypes: alignment issues in Memory_Array

umarcor, 2020-05-12:
Trying to build latest ghdl on an arm32v7 docker container or on an ARTY-Z7 board with Pynq (Ubuntu 18.04), using https://github.com/dbhi/dbhi/blob/master/scripts/setup_pynq.sh#L9-L37, produces the following warnings:

```
gnata -gnatwe -I- /ghdl/src/synth/synth-objtypes.adb
synth-objtypes.adb:621:04: warning: alignment of "Ghdl_I32" (4) is stricter than alignment of subtype of "Memory_Array" (1)
synth-objtypes.adb:621:04: warning: resulting access value may have invalid alignment
synth-objtypes.adb:626:19: warning: source alignment (1) < alignment of "Ghdl_I32" (4)
synth-objtypes.adb:631:26: warning: source alignment (1) < alignment of "Ghdl_I32" (4)
synth-objtypes.adb:635:04: warning: alignment of "Ghdl_U32" (4) is stricter than alignment of subtype of "Memory_Array" (1)
synth-objtypes.adb:635:04: warning: resulting access value may have invalid alignment
synth-objtypes.adb:640:19: warning: source alignment (1) < alignment of "Ghdl_U32" (4)
synth-objtypes.adb:645:26: warning: source alignment (1) < alignment of "Ghdl_U32" (4)
synth-objtypes.adb:649:04: warning: alignment of "Ghdl_I64" (8) is stricter than alignment of subtype of "Memory_Array" (1)
synth-objtypes.adb:649:04: warning: resulting access value may have invalid alignment
synth-objtypes.adb:654:19: warning: source alignment (1) < alignment of "Ghdl_I64" (8)
synth-objtypes.adb:659:26: warning: source alignment (1) < alignment of "Ghdl_I64" (8)
synth-objtypes.adb:663:04: warning: alignment of "Fp64" (8) is stricter than alignment of subtype of "Memory_Array" (1)
synth-objtypes.adb:663:04: warning: resulting access value may have invalid alignment
synth-objtypes.adb:668:20: warning: source alignment (1) < alignment of "Fp64" (8)
synth-objtypes.adb:673:27: warning: source alignment (1) < alignment of "Fp64" (8)
gnatmake: "/ghdl/src/synth/synth-objtypes.adb" compilation error
Makefile:347: recipe for target 'ghdl_llvm' failed
make: *** [ghdl_llvm] Error 4
```

Note that `--disable-werror` is used in the script in order to allow the build to finish. Simulation worksas expected. I didn't try synthesis.

/cc @adriano-navarro

tgingold, 2020-05-13:
I tried to fix it, but please confirm.

umarcor, 2020-05-13:
Thanks!

It's a partial fix. Some of the warnings were removed, but others are still present:

```
arm-linux-gnueabihf-gcc-8 -c -I./ -I../src -I../src/vhdl -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/o
rtho -I../src/ortho/llvm4-nodebug -I../src/synth -I../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -I- /ghdl/src/synth/
synth-objtypes.adb
synth-objtypes.adb:630:31: warning: source alignment (1) < alignment of "Ghdl_I32" (4)
synth-objtypes.adb:635:38: warning: source alignment (1) < alignment of "Ghdl_I32" (4)
synth-objtypes.adb:644:31: warning: source alignment (1) < alignment of "Ghdl_U32" (4)
synth-objtypes.adb:649:38: warning: source alignment (1) < alignment of "Ghdl_U32" (4)
synth-objtypes.adb:658:31: warning: source alignment (1) < alignment of "Ghdl_I64" (8)
synth-objtypes.adb:663:38: warning: source alignment (1) < alignment of "Ghdl_I64" (8)
synth-objtypes.adb:672:32: warning: source alignment (1) < alignment of "Fp64" (8)
synth-objtypes.adb:677:39: warning: source alignment (1) < alignment of "Fp64" (8)
gnatmake: "/ghdl/src/synth/synth-objtypes.adb" compilation error
Makefile:347: recipe for target 'ghdl_llvm' failed
make: *** [ghdl_llvm] Error 4
```

It was tested on a docker container and on an ARTY-Z7 again. Same result in both environments.

umarcor, 2020-05-13:
Almost there...

```
-I../src/ortho -I../src/ortho/llvm4-nodebug -I../src/synth -I../src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe
-I- /ghdl/src/synth/synth-values.adb
synth-values.adb:301:04: warning: alignment of "Heap_Index" (4) is stricter than alignment of subtype of "Memory_Array" (1)
synth-values.adb:301:04: warning: resulting access value may have invalid alignment
synth-values.adb:306:26: warning: source alignment (1) < alignment of "Heap_Index" (4)
synth-values.adb:311:33: warning: source alignment (1) < alignment of "Heap_Index" (4)
gnatmake: "/ghdl/src/synth/synth-values.adb" compilation error
Makefile:347: recipe for target 'ghdl_llvm' failed
make: *** [ghdl_llvm] Error 4
```

tgingold, 2020-05-14:
Did you try again ?

umarcor, 2020-05-14:
I tried now and it seems to be fixed. Thanks!

closed 2020-05-14 by umarcor