ghdl/ghdl #1309: Incorrect case statement behavior with new "pmux" changes

rlee287, 2020-05-14:
**Description**
Recent changes to case statements introduced in commit c9d2cccb38b985783108f187398a0a0af9f6ab62 (as found by `git bisect`) result in case statements not being handled correctly.

**Expected behaviour**
The `cover` statement in the `tb_formal_top.v` file of the below MWE passes.

**How to reproduce?**
Checkout the given SHA's and and compile the following software from source, in the following order:

 * Yosys: 27b7ffc7
 * SymbiYosys: 87eb475
 * GHDL: c9d2cccb (using the LLVM backend)
 * GHDL-yosys-plugin: 2f7e216

Extract [bugreport.zip](https://github.com/ghdl/ghdl/files/4625940/bugreport.zip) into an empty folder.

(I acknowledge that the name `axis_squarer` is a misnomer given that this module inverts the incoming bits, but this is simplified from the original code where I extracted this from.)

Run the following command on the command line from within the extracted `.zip`: `sby --yosys "yosys -m ghdl" -f axis_squarer.sby cover`.

The `cover` statement will fail for the given GHDL commit c9d2cccb and will pass with the previous commit 1a65ac6b.

**Context**
Please, provide the following information:

- OS: Ubuntu 19.10
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: c9d2cccb38b985783108f187398a0a0af9f6ab62 or later (see additional context below)

**Additional context**

`git bisect` results from f2badf67..553455c5 (SHA's are the ordered tuple (GHDL, GHDL-yosys-plugin) with other relevant SHA's as in the above MWE)
(f2badf67, 386ad81) old
(553455c5, 2f7e216) new
(0d9ed3a1, 2f7e216) new
(659d3579, 386ad81) old
(b13b9e9f, 2f7e216) old
(23c92364, 2f7e216) new
(c9d2cccb, 2f7e216) new
(1a65ac6b, 2f7e216) old

The output of `ghdl --synth axis_squarer.vhd -e axis_squarer > $(FILENAME)` (where "old" corresponds to 1a65ac6b and "new" corresponds to c9d2cccb): [synth_cmd_diff.zip](https://github.com/ghdl/ghdl/files/4625980/synth_cmd_diff.zip)

tmeissner, 2020-05-14:
I also found incorrect behavior in my formal tests. I searched a little bit and the error was introduced with commit https://github.com/ghdl/ghdl/commit/c9d2cccb38b985783108f187398a0a0af9f6ab62. The last working commit I tested is https://github.com/ghdl/ghdl/commit/b13b9e9fa26121a68d74589ef35ea5ed45316cd9

The behavior in my case is really strange, the output of my sequencer ports, which contains a simple case statement, is negated. Do all my assert fail at another cycles than intended.

tmeissner, 2020-05-14:
I wonder how we can test the absence of such regression bugs. The VHDL output of the GHDL synthesis is correct, I‘ve checked that.

tgingold, 2020-05-14:
So maybe the bug is in the yosys plugin.  This part is not so well tested.

tmeissner, 2020-05-14:
No. The Yosys plugin with latest commit works. I have tested it with different Ghdl commits.

tgingold, 2020-05-14:
Well, the 'pmux' handling in the plugin can be wrong.

tmeissner, 2020-05-14:
But maybe it’s a combination of changes intriguing Ghdl and the plugin. I’ve seen that the plugin has exactly one commit in the relevant time (last one from 9. may). I can build with this current head and old versions before the mentioned Ghdl commit that work correctly. These old Ghdl versions don’t generate the pmux, so maybe it’s the plugin.

tmeissner, 2020-05-14:
I've created a simple test case:

```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity sequencer is
  generic (
    seq : string
  );
  port (
    clk  : in  std_logic;
    data : out std_logic
  );
end entity sequencer;

architecture rtl of sequencer is

  signal index : natural := seq'low;
  signal ch    : character;

  function to_bit (a : in character) return std_logic is
    variable ret : std_logic;
  begin
    case a is
      when '0' | '_' => ret := '0';
      when '1' | '-' => ret := '1';
      when others    => ret := 'X';
    end case;
    return ret;
  end function to_bit;

begin

  process (clk) is
  begin
    if rising_edge(clk) then
      if (index < seq'high) then
        index <= index + 1;
      end if;
    end if;
  end process;

  ch <= seq(index);

  data <= to_bit(ch);

end architecture rtl;

library ieee;
  use ieee.std_logic_1164.all;

entity psl_test is
  port (
    clk : in std_logic
  );
end entity psl_test;

architecture psl of psl_test is

    component sequencer is
    generic (
      seq : string
    );
    port (
      clk  : in  std_logic;
      data : out std_logic
    );
  end component sequencer;

  signal a, b : std_logic;

begin

  --                              0123
  SEQ_A : sequencer generic map ("--_") port map (clk, a);
  SEQ_B : sequencer generic map ("__-") port map (clk, b);

  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);

  -- This assertion holds
  SERE_2_a : assert always {a; a} |=> {b};

end architecture psl;
```

Using the following psl_test.sby file:

```yaml
[tasks]
prove

[options]
depth 25
prove: mode bmc

[engines]
prove: smtbmc z3

[script]
prove: ghdl --std=08 psl_test.vhd -e psl_test
prep -top psl_test

[files]
psl_test.vhd
```

```
sby --yosys "yosys -m ghdl" -f -d work/psl_test psl_test.sby prove
```

The assertion should hold.

Xiretza, 2020-05-14:
I'm not sure if this is the same problem that @tmeissner's reproducer shows, but the `$pmux` select input MUST have at most one hot bit at all times. Since `'H'`, `'L'` and friends get mapped down to `'0'`/`'1'` during synthesis, it's possible to construct a case that can match on multiple branches at once, breaking this law.

yosys has a handy pass that inserts assertions for this, `assertpmux`. Here's a tiny reproducer that shows the problem:

```vhdl :file: pmux.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		a : in std_logic;
		b : out std_logic
	);
end entity;

architecture a of ent is
begin
	process(a)
	begin
		case a is
			when '0' =>
				b <= '0';
			when 'L' =>
				b <= '1';
			when others =>
				b <= 'Z';
		end case;
	end process;
end architecture;
```

```
$ yosys -m ghdl -p 'ghdl pmux.vhd -e; assertpmux; sat -prove-asserts'
[...]
 Yosys 0.9+2406 (git sha1 aafaeb66, gcc 9.3.0-1 -march=x86-64 -mtune=generic -O2 -fno-plt -fvar-tracking-assignments -fdebug-prefix-map=/build/yosys-git/src=/usr/src/debug -fPIC -Os)

-- Running command `ghdl pmux.vhd -e; assertpmux; sat -prove-asserts' --

1. Executing GHDL.
note: top entity is "ent"
Importing module ent.

2. Executing ASSERTPMUX pass (add asserts for $pmux cells).
Adding assert for $pmux cell ent.\11.

3. Executing SAT pass (solving SAT problems in the circuit).

Setting up SAT problem:
Final constraint equation: { } = { }
Imported 7 cells to SAT database.
Import proof for assert: $auto$rtlil.cc:1940:Le$9 when 1'1.

Solving problem with 61 variables and 162 clauses..
SAT proof finished - model found: FAIL!

   ______                   ___       ___       _ _            _ _ 
  (_____ \                 / __)     / __)     (_) |          | | |
   _____) )___ ___   ___ _| |__    _| |__ _____ _| | _____  __| | |
  |  ____/ ___) _ \ / _ (_   __)  (_   __|____ | | || ___ |/ _  |_|
  | |   | |  | |_| | |_| || |       | |  / ___ | | || ____( (_| |_ 
  |_|   |_|   \___/ \___/ |_|       |_|  \_____|_|\_)_____)\____|_|

  no model variables selected for display.

End of script. Logfile hash: 447ec8fd84, CPU: user 0.03s system 0.01s, MEM: 26.96 MB peak
Yosys 0.9+2406 (git sha1 aafaeb66, gcc 9.3.0-1 -march=x86-64 -mtune=generic -O2 -fno-plt -fvar-tracking-assignments -fdebug-prefix-map=/build/yosys-git/src=/usr/src/debug -fPIC -Os)
Time spent: 91% 1x ghdl (0 sec), 7% 1x sat (0 sec), ...
```

Xiretza, 2020-05-14:
Alright, the reproducer I posted is for a different issue, @tmeissner's reproducer fails because the inputs for `$pmux` were mapped in reverse order, resulting in the equivalent of something like this:

```vhdl
case a is
    when '0' | '_' => ret := '1';
    when '1' | '-' => ret := '0';
    when others    => ret := 'X';
end case;
```

This should be fixed by ghdl/ghdl-yosys-plugin#118 - @rlee287, can you confirm this also fixes your original issue?

rlee287, 2020-05-14:
The `axis_squarer` testcase generates the cover trace successfully with GHDL 553455c and ghdl-yosys-plugin 1bbc923 (from the PR ghdl/ghdl-yosys-plugin#118). I believe this issue can be closed after that PR is merged.

rlee287, 2020-05-14:
As the issue is now fixed by the merged PR, I am closing this issue.

closed 2020-05-14 by rlee287

tgingold, 2020-05-14:
First thanks for the issue and the small reproducer.  I have added it to the plugin testsuite as the regression was unnoticed.

There are a few other testcase in this issue I will also add in the testsuite.

tmeissner, 2020-05-14:
I can confirm that https://github.com/ghdl/ghdl-yosys-plugin/pull/118 also fixed my problem. Good work :smile:

tgingold, 2020-05-14:
@Xiretza I think your case statement test with 'L' shouldn't synthesize.  Or at least it is worth a warning.