ghdl/ghdl #1310: synth: signed type ordering operators

anhangaba, 2020-05-14:
```vhdl file:issue.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity issue is
    port (sig_gt, sig_ge, sig_lt, sig_le : out boolean;
          uns_gt, uns_ge, uns_lt, uns_le : out boolean);
end issue;

architecture beh of issue is
begin
    -- all of those works
    uns_gt <= (unsigned'("1111") >  unsigned'("0111"));
    uns_ge <= (unsigned'("1111") >= unsigned'("0111"));
    uns_lt <= (unsigned'("1111") <  unsigned'("0111"));
    uns_le <= (unsigned'("1111") <= unsigned'("0111"));

    sig_gt <= (signed'("1111") >  signed'("0111"));
    sig_ge <= (signed'("1111") >= signed'("0111"));
    sig_lt <= (signed'("1111") <  signed'("0111"));
    sig_le <= (signed'("1111") <= signed'("0111"));
end architecture beh;
```

```
$ ghdl --synth --std=08 issue.vhdl -e issue
```

```
issue.vhdl:14:34:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_GE_UNS_UNS
issue.vhdl:18:32:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_GT_SGN_SGN
issue.vhdl:19:32:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_GE_SGN_SGN
issue.vhdl:20:32:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_LT_SGN_SGN
issue.vhdl:21:32:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_LE_SGN_SGN
issue.vhdl:6:11:warning: no assignment for port "sig_gt"
issue.vhdl:6:19:warning: no assignment for port "sig_ge"
issue.vhdl:6:27:warning: no assignment for port "sig_lt"
issue.vhdl:6:35:warning: no assignment for port "sig_le"
issue.vhdl:7:19:warning: no assignment for port "uns_ge"
```

closed 2020-05-15 by tgingold

anhangaba, 2020-05-15:
It seems you missed one of the cases I added one hour after the original submission.

```vhdl
sgn_eq2 <= (signed'("1111") =  1); -- unhandled
```

tgingold, 2020-05-15:
Yes, I missed it because I simply read the email.

commit 6a57bfe3 (2020-05-15): synth-static_oper: handle more comparisons.  Fix #1310