ghdl/ghdl #1314: synth: name clash when replicating named PSL directives with for generate

tmeissner, 2020-05-15:
**Description**
If you use a for generate construct to replicate named PSL directives all these replicated directives have the same name in the resulting netlist. This leads to an Yosys error like this:

```
ERROR: Assert `count_id(cell->name) == 0' failed in kernel/rtlil.cc:1613.
```

**Expected behaviour**
All the replicated directives should get an name formed of the given name + an prefix (loop index for example).

**How to reproduce?**

Get this file: 

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 issue_1314.vhd -e issue | grep next_event_a
```

Result:

```
issue_1314.vhd:42:21:note: found ROM "n1316", width: 8 bits, depth: 13
issue_.1314vhd:102:21:note: found ROM "n1363", width: 8 bits, depth: 13
issue_1314.vhd:42:21:note: found ROM "n1470", width: 8 bits, depth: 13
issue_1314.vhd:102:21:note: found ROM "n1517", width: 8 bits, depth: 13
  next_event_a: postponed assert n167_o = '1' severity error;
  next_event_a_cover: assert n168_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n242_o = '1' severity error;
  next_event_a_cover: assert n243_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n317_o = '1' severity error;
  next_event_a_cover: assert n318_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n392_o = '1' severity error;
  next_event_a_cover: assert n393_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n467_o = '1' severity error;
  next_event_a_cover: assert n468_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n542_o = '1' severity error;
  next_event_a_cover: assert n543_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n617_o = '1' severity error;
  next_event_a_cover: assert n618_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n692_o = '1' severity error;
  next_event_a_cover: assert n693_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n767_o = '1' severity error;
  next_event_a_cover: assert n768_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n842_o = '1' severity error;
  next_event_a_cover: assert n843_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n917_o = '1' severity error;
  next_event_a_cover: assert n918_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n992_o = '1' severity error;
  next_event_a_cover: assert n993_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n1067_o = '1' severity error;
  next_event_a_cover: assert n1068_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n1142_o = '1' severity error;
  next_event_a_cover: assert n1143_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n1217_o = '1' severity error;
  next_event_a_cover: assert n1218_o = '1' severity note; --  assert_cover
  next_event_a: postponed assert n1292_o = '1' severity error;
  next_event_a_cover: assert n1293_o = '1' severity note; --  assert_cover
```

**Context**
Please, provide the following information:

- OS: Debian Buster
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Docker (Image: ghdl/synth:formal)

tgingold, 2020-05-16:
Thank you for the issue!

closed 2020-05-16 by tgingold

tmeissner, 2020-05-16:
You're welcome. There's more to come regarding PSL 😜

commit 7285f745 (2020-05-16): synth: use expanded name for assert labels.  Fix #1314