ghdl/ghdl #1319: synth: Possible issue in microwatt with SPR decode

ozbenh, 2020-05-17:
**Description**
Hi Tristan !

Sadly, I don't have a good clean repro-case and simple description here... I'm not even
sure the bug is in GHDL, I might be mistaken here. But here is so far what I found out
and I could use your help (I wanted to email that one but I don't have your address :-)

When converting the current microwatt (or at least my intercon branch but the problem is I believe the same with Anton's master) to verilog with ghdl, plugin and yosys, the end result doesn't "work" in LiteX sim.

Now I've spent too much time already eyeballing VCDs ... but I found so far that it *looks* like the output of decode1 is wrong for the MTSPR instruction. More specifically this:

```
		elsif v.decode.insn_type = OP_MFSPR or v.decode.insn_type = OP_MTSPR then
		    v.ispr1 := fast_spr_num(decode_spr_num(f_in.insn));
		    -- Make slow SPRs single issue
		    if is_fast_spr(v.ispr1) = '0' then
			v.decode.sgl_pipe := '1';
		    end if;
```

Seems to end up in sim with all 0's in v.ispr1 instead in my case of the expected "0x21"
value for an mtctr instruction.

Now, I can't yet tell whether the problem is in ghdl-synth or elsewhere for sure, but my digging *seem* to point to ghdl unless I missed something...

fast_spr_num() function looks like this:

```
    function fast_spr_num(spr: spr_num_t) return gspr_index_t is
       variable n : integer range 0 to 31;
    begin
       case spr is
       when SPR_LR =>
           n := 0;
       when SPR_CTR =>
           n:= 1;
       when SPR_SRR0 =>
           n := 2;
       when SPR_SRR1 =>
           n := 3;
       when SPR_HSRR0 =>
           n := 4;
       when SPR_HSRR1 =>
           n := 5;
       when SPR_SPRG0 =>
           n := 6;
       when SPR_SPRG1 =>
           n := 7;
       when SPR_SPRG2 =>
           n := 8;
       when SPR_SPRG3 | SPR_SPRG3U =>
           n := 9;
       when SPR_HSPRG0 =>
           n := 10;
       when SPR_HSPRG1 =>
           n := 11;
       when SPR_XER =>
           n := 12;
       when others =>
           n := 0;
           return "000000";
       end case;
       return "1" & std_ulogic_vector(to_unsigned(n, 5));
    end;
```

In this case it should hit the "SPR_CTR" case.

decode_spr_num is rather trivial:

    function decode_spr_num(insn: std_ulogic_vector(31 downto 0)) return spr_num_t is
    begin
	return to_integer(unsigned(insn(15 downto 11) & insn(20 downto 16)));
    end;

Now, the "simplified VHDL" generated by ghdl looks fairly hard to parse but here's what I think
is the relevant bit:

  assign _30_ = _29_ ? 6'h21 : 6'h00;
  assign _31_ = _25_[7:2] == 6'h08;
  assign _32_ = ~ f_in[76];
  assign _33_ = _32_ ? 6'h20 : 6'h21;
  assign _34_ = _31_ ? _33_ : 6'h00;
  assign _35_ = _25_[7:2] == 6'h27;
  assign _36_ = _25_[7:2] == 6'h2b;
  assign _37_ = _35_ | _36_;
  assign _38_ = ~ 1'h0;
  assign _39_ = _44_ ? 1'h1 : _25_[36];
  assign _40_ = _25_[7:2] == 6'h32;
  assign _41_ = _40_ ? 12'h8e2 : 12'h000;
  assign _42_ = _37_ ? 6'h00 : _41_[5:0];
  assign _43_ = _37_ ? 6'h00 : _41_[11:6];
  assign _44_ = _37_ & _38_;
  assign _45_ = _28_ ? { _34_, _30_ } : { _43_, _42_ };
  assign _46_ = _28_ ? _25_[36] : _39_;
  assign _47_ = flush_in ? 1'h0 : f_in[0];
  assign rin = rst ? 147'h0000000000000000000000000000000000000 : { _46_, _25_[35:0], _45_, f_in[97:1], _47_ };

r_in is the final assignment to the pipe register. It looks like ispr1 and ispr2 were grouped together
as _45_ as far as I can tell. 

_41_ is 0x8e2 or 0. This is 0x23 and 0x22, so SRR1 | SRR0 combined, so that correspond to
the "rfid" instruction decoding. That also means that _42_ is ispr1 and _43_ is ispr2 and are either
srr1/0 or 0.  That branch can't give values decoded from the instruction word.

Thus my case must _34_ and _30_. Since I'm looking for ispr1, I think it's _30_. But that's either 0x21 which is CTR (decoding a branch) or ... 0 !

So unless I've missed something, it looks like there is no path in there to extract the numbers from the instruction and map them like the function should be doing.  But I'm not 100% certain I didn't misread something...

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [*] Built from sources: `f01c93b00c618e994f795285a5eb3debeaf03fcf`

ozbenh, 2020-05-17:
Note: The signals name change a lot depending on which specific branch you build, but the end result appears the same.

tgingold, 2020-05-17:
I fear you are using the hard way.

The easiest way to investigate such a bug is to write a simple program that trigger the bug, test it on a full simulation, and then synthesis just decode1.vhdl (using `ghdl --synth`) and run the simulation with the synthesized unit and the remaining of the sources.

ozbenh, 2020-05-18:
Hey, maybe :-)

Long story short, that microwatt works fine in "pure" ghdl sim or synthetized with vivado, it's the yosys conversion that fails. The reason I was doing this conversion was to be able to sim it alongside litex which is currently only capable of doing verilog. In fact, Florent tried to do that and reported that the core didn't boot in that case (it boots if synthetized in LiteX with Vivado).

So I got LiteX to spit out waves and dug in ... I didn't have a "test case", I first had to find a string to pull as to what was going on. I quickly saw the core fetching instructions for a little while then stopping. It was painful as you can imagine, but I did manage to find that decode1 seemed the culprit when hitting an mtctr instruction.

Now today I proved that as follow:

I wrote a little trivial test bench for decode1:

https://ozlabs.org/~benh/decode1_tb.vhdl

If you build and run this in ghdl along with microwatt's decode1.vhdl

```
ghdl -a --std=08 --workdir=. wishbone_types.vhdl
ghdl -a --std=08 --workdir=. decode_types.vhdl
ghdl -a --std=08 --workdir=. common.vhdl
ghdl -a --std=08 --workdir=. decode1.vhd
ghdl -a --std=08 --workdir=. decode1_tb.vhdl
ghdl -e --std=08 decode1_tb
```

And run the result, you should see in the output:

```
decode1_tb.vhdl:52:8:@35ns:(report note): out1: V='0'
decode1_tb.vhdl:54:8:@45ns:(report note): out2: V='0'
decode1_tb.vhdl:56:8:@55ns:(report note): out3: V='1'
decode1_tb.vhdl:57:8:@55ns:(report note):   ispr1=00
decode1_tb.vhdl:58:8:@55ns:(report note):   ispr2=00
```

Then, I generated a "synth" version of decode1.vhdl:

```
ghdl --synth --std=08 decode1.vhdl >decode1_s.vhdl
```

And re-generated the work library using the above commands but using decode1_s.vhdl instead of decode1.vhdl, and the result was this time:

```
decode1_tb.vhdl:52:8:@35ns:(report note): out1: V='0'
decode1_tb.vhdl:54:8:@45ns:(report note): out2: V='0'
decode1_tb.vhdl:56:8:@55ns:(report note): out3: V='1'
decode1_tb.vhdl:57:8:@55ns:(report note):   ispr1=00
decode1_tb.vhdl:58:8:@55ns:(report note):   ispr2=00
```

This seem to confirm my analysis that something is wrong when function fast_spr_num() is used
with a non-constant input. I will try to do a smaller test case just with that function, but I've run out of time today :-)

tgingold, 2020-05-18:
Maybe this was due to issue #1309 ?

A bug with case statements was fixed last week in the yosys plugin.  So for a few days the ghdl+yosys chain was not correct.

ozbenh, 2020-05-18:
I can reproduce this purely with ghdl --synth without using yosys. I managed to make a smaller test case:

https://ozlabs.org/~benh/test_ent.vhdl
https://ozlabs.org/~benh/test_tb.vhdl

test_ent is the test entity, test_tb is the test bench.

If you elaborate them as-is and run, you'll get this output:

```
test_tb.vhdl:26:8:@1ns:(report note):  ispr1=21
test_tb.vhdl:27:8:@1ns:(report note):  ispr2=2C
```

If you first convert test_ent.vhdl to "simple vhdl" with ghdl --synth, and then elaborate that with test_tb and run it, you get this outout:

```
test_tb.vhdl:26:8:@1ns:(report note):  ispr1=00
test_tb.vhdl:27:8:@1ns:(report note):  ispr2=2C
```

So it looks like if the argument to the function is constant, it works, but if it's a signal, it doesn't

tgingold, 2020-05-18:
Ok, I will have a look.  Thanks for the testcase.

tgingold, 2020-05-18:
Now fixed.

Thank you for the reduced test, that was really useful.
And sorry for the bug, I fear you spent more time on it than me.

closed 2020-05-18 by tgingold

ozbenh, 2020-05-18:
Thanks ! No need to be sorry :-)

ozbenh, 2020-05-18:
Tested, the microwatt.v conversion now boots in LiteX :-)

tgingold, 2020-05-18:
Great!!!

commit 4291773d (2020-05-18): synth-environment: fix handling of static values in case statements. Fix #1319