ghdl/ghdl #1331: Procedures to start/stop wave file generation

ozbenh, 2020-05-21:
When simulating a fairly large design, generating wave files slows it down to a crawl.

In some cases, for example when debugging the DDR RAM interface of Microwatt in sim, I would like to "go fast" while the "firmware" runs the register init sequence without generating a wave file, and start generation once we start doing actual DRAM accesses.

Any mechanism would do as long as I can control from within the simulation when to start/stop the generation.

eine, 2020-05-21:
Ref #1281

ozbenh, 2020-05-22:
A partial resolution would be https://github.com/ghdl/ghdl/pull/782

Now even in that case, there is probably still value in being able to start/stop wave generation
as to avoid creating multi-gigabytes wave files that will crash gtkwave for the entire "warm up" time before the interesting data is present.

eine, 2020-05-22:
@ozbenh, ATM #782 works with mcode backend only. But you can try it.

Anyway, I agree that being able to define "generation time windows" would be useful.

There is already the option to filter which signals are dumped (https://ghdl.readthedocs.io/en/latest/using/Simulation.html#cmdoption-ghdl-read-wave-opt). I assume that your issue is even with a reduced set of signals, isn't it?

ozbenh, 2020-05-22:
Yes. I've been using reduced signal sets so far. It helps a lot in some cases. Thanks