ghdl/ghdl #1333: synth: crash when using ascending attribute

marph91, 2020-05-21:
**How to reproduce?**

```vhd
library ieee;
use ieee.std_logic_1164.all;

entity ascending_attribute is
  port (
    a : in std_logic_vector(7 downto 0);
    b : out boolean
  );
end ascending_attribute;

architecture rtl of ascending_attribute is
  function is_ascending(i : std_logic_vector) return boolean is
  begin
    return i'ASCENDING;
  end function;
begin
  b <= is_ascending(a);
end rtl;
```

**Context**

```
synth_expression_with_type: cannot handle IIR_KIND_ASCENDING_ARRAY_ATTRIBUTE (../ascending_attribute.vhdl:14:14)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-617-g032f2940) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-suse-linux
/home/workspace/vhdl/test_ghdlsynth/ascending_attribute/build/
Command line:
ghdl --synth ascending_attribute
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x4c6543 0x5e4e5f 0x5c970f 0x5cabb9 0x5c606d 0x5c6d7f 0x5c6ea2 0x5cb4bf 0x5e47d5 0x5bc0be 0x5bf41e 0x5cccb9 0x5ccf18 0x5bb5b1 0x5bb718 0x5b4b84 0x631f82 0x54bcf3 0x63b2ff 0x403b79 0x7f0dc5be5348 0x402b28 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-05-21:
I am curious about the real use of 'ascending in a design..

closed 2020-05-21 by tgingold

marph91, 2020-05-22:
In the meanwhile, it was replaced by something more elegant. However, I thought it's better to report the bug, since 'ascending is part of the synthesizable subset of VHDL.

Thanks for the effort!

commit dbd8dc2d (2020-05-21): synth: add support for ascending attribute.  Fix #1333