ghdl/ghdl #1336: missing target libghdlsynth.a

kost, 2020-05-22:
**Description**
libghdlsynth.a is not possible target any more. It is needed by static version of ghdl-yosys-plugin available here https://github.com/ghdl/ghdl-yosys-plugin/blob/master/yosys.diff

**Expected behaviour**

That target libghdlsynth.a is possible to build in order to have fully static build of yosys/ghdl.

**How to reproduce?**
```
/opt/ghdl-git # make /opt/ghdl-git # make libghdlsynth.a
make: *** No rule to make target 'libghdlsynth.a'.  Stop.
```

tgingold, 2020-05-22:
The static build hasn't be used for a long time.  I will remove that option.

Unless there is a very good reason for it instead of the plugin.

daveshah1, 2020-05-22:
Yosys plugins aren't supported on Windows, at the moment, whereas I think the static build should work for that.

kost, 2020-05-22:
It is also convient to have static binary for minimal container deployments and distributution.

eine, 2020-05-22:
Hi @kost! I saw that you are building multiple docker images for ghdl, yosys, etc. I don't know if you are aware of [ghdl/docker](https://github.com/ghdl/docker). Although currently stopped due to lack of time, I want to split non-ghdl resources to [eine/symbiflow-containers](https://github.com/eine/symbiflow-containers) (https://hub.docker.com/u/symbiflow). I believe it would be interesting to provide minimal containers with static builds too.

However, last time I tried building GHDL with musl (3 years ago #331), it would fail because of `backtrace()`. That's why Debian Buster (slim) is used as the base image. Did you have any similar issue or are you installing glibc in Alpine?

kost-div, 2020-05-23:
Thanks @eine. Saw it, but it was dynamically linked AFAIK. Regarding musl, you just need to use separate libunwind-dev package and link with it. I'm using it that way and it works (no need for glibc).
You can find my complete patches for Makefile.in available here:
https://github.com/alpin3/ulx3s/blob/master/root/opt/patches/ghdl.patch

And other set of patches for yosys:
https://github.com/alpin3/ulx3s/blob/master/root/opt/patches/yosys.diff

Resulting static binaries are here (ghdl, yosys with ghdl, etc - much smaller than with glibc):
https://github.com/alpin3/ulx3s/releases

If you need a helping hand - let me know! Will be glad to help since I've spent quite some time building all of this statically.

kost, 2020-05-23:
Thanks @eine. Saw it, but it was dynamically linked AFAIK. Regarding musl, you just need to use separate libunwind-dev package and link with it. I'm using it that way and it works (no need for glibc).
You can find my complete patches for Makefile.in available here:
https://github.com/alpin3/ulx3s/blob/master/root/opt/patches/ghdl.patch

And other set of patches for yosys:
https://github.com/alpin3/ulx3s/blob/master/root/opt/patches/yosys.diff

Resulting static binaries are here (ghdl, yosys with ghdl, etc - much smaller than with glibc):
https://github.com/alpin3/ulx3s/releases

If you need a helping hand - let me know! Will be glad to help since I've spent quite some time building all of this statically.

tgingold, 2020-05-23:
So windows and containers are indeed good reasons to have a static version of yosys including ghdl.

The first step is to create a static version of libghdl.a, then that will simply be a small build change for yosys.

kost, 2020-05-23:
Sounds good @tgingold !

BTW if you look closely at this patch:
https://github.com/alpin3/ulx3s/blob/master/root/opt/patches/ghdl.patch

You will see I'm already generating/installing libghdl.a and libghdlsynth.a since it is needed for statically compiled yosys. You just need to modify it to suit your need/vision. I can make it any license you need even without any credit. If you need any help - let me know!

tgingold, 2020-05-23:
So I have added the build of libghdl.a, and re-added the instructions to build yosys with the static libghdl.a

edbordin, 2020-05-24:
I independently set up a patch for this about a week ago so that I could also compile the plugin into yosys . https://github.com/open-tool-forge/fpga-toolchain/blob/b532ee9/scripts/libghdl_static.diff

When I looked at the old target for `libghdlsynth.a` it was also writing out a `.link` file with linker flags so I replicated that and made use of it in my build script https://github.com/open-tool-forge/fpga-toolchain/blob/b532ee9/scripts/compile_yosys.sh#L56
Any chance we can add there here too? Or is it not a good way of doing things?

tgingold, 2020-05-24:
> When I looked at the old target for libghdlsynth.a it was also writing out a .link file with linker flags so I replicated that and made use of it in my build script [...]

Why do you need to overwrite LDLIBS ?  The .link file is already read in the yosys Makefile change for adding ghdl
(see https://github.com/open-tool-forge/fpga-toolchain/blob/b532ee91b0f0a897f7cd03a4dabee71c435fbeeb/scripts/yosys-ghdl.diff#L49)

edbordin, 2020-05-24:
> Why do you need to overwrite LDLIBS ?

We're aiming to produce fully static binaries, so it's mainly to add `-static` (or on OS X, the full path to the static libs). I also found I had to remove `pragma Linker_Options ("-lz");` otherwise OS X always chooses the dynamic lib if it can find it, but I can keep patching that out downstream of you.

tgingold, 2020-05-25:
So I close this ticket, as a Makefile rule for building libghdl.a has been added.

closed 2020-05-25 by tgingold