ghdl/ghdl #1337: Way to add clocks generated with ecppll?

kost, 2020-05-22:
Is there any way to add clocks generated with ecppll to ghdl? Is there any other generator? Is there any examples somewhere?

tgingold, 2020-05-22:
For synthesis, you can mix (with restrictions) verilog and vhdl.

For simulation, no.

ecppll is a simple tool, so an option to generate vhdl shouldn't be a lot of work.

In the vhdl-yosys-plugin, there is an example for ecp5 which use the pll.

eine, 2020-05-22:
@kost, the example mentioned by @tgingold is https://github.com/ghdl/ghdl-yosys-plugin/tree/master/examples/ecp5_versa. It was contributed by @hackfin in February and later adjusted by Tristan: https://github.com/ghdl/ghdl-yosys-plugin/commits/master/examples/ecp5_versa. Martin has been using PLL primitives in his MaSoCist. See "Vendor primitives" in https://section5.ch/index.php/2020/02/17/masocist-support-of-opensource-synthesis-for-ecp5/. Precisely, ghdl/ghdl-yosys-plugin#46 is related to the changes that Tristan did on the example.

Regarding simulation, yesterday @ozbenh reported some tests combining Verilator and GHDL's VHPIDIRECT: #1335. This is the first time I've seen someone try such approach with non-trivial modules. It is very promising, but not ready-to-use off-the-shelf.

ozbenh, 2020-05-23:
It was actually surprisingly easy in the end. However I don't expect the performance to be stellar if the clock is coming from the verilated binary as you'd have to constantly eval() it to extract it (but it's probably worth trying). In my case, I pass the clock into my verilated design and don't care about non-synchronous activites so I only need to "run" the verilated designed on clock  rising edges which works reasonably well.

hackfin, 2020-05-24:
Interesting one and constant nudge for all sorts of different PLL types.
You can use the *farey sequence* generation to get by with less of a brute force error minimizing linear search than ecppll is doing, but for the particular case with boundary conditions of the intermediate frequencies, each PLL type seems to need its own partcular solver for max. precision. But when precision doesn't matter much, a bit of binary searching can help. I was playing with this for the ECP5 side (although Python/MyHDL, where the brute force approach takes too much time), but haven't gotten to the full conclusion yet either.
In particular for yosys I'm generating black boxes, so therefore wrappers for instancing VHDL and Verilog modules would still have to be written.

tgingold, 2020-05-25:
I close this issue as answered.

closed 2020-05-25 by tgingold