ghdl/ghdl #1338: Improve usefulness of IEEE library warnings by adding signal name

ozbenh, 2020-05-22:
When running Microwatt, we get a LOT of warnings such as

```../../src/ieee2008/numeric_std-body.vhdl:3036:7:@206015ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
```
So much that I often run with --ieee-asserts=disable because the output is otherwise unusable.

The problem is that it means I might be missing the few useful ones... For example, I noticed some do happen at runtime after we are past the reset stage (when we are expected to have a bunch).

Now they may be ok, .. or not. But I don't know because the assertion message gives me no way to figure out which signal is bad ! :-)

maehne, 2020-05-22:
I agree that more context for debugging the cause for these warnings would be very helpful. I am currently resorting in these cases to opening the waveform viewer and inspecting all signals at the mentioned time point to get a clue.

ozbenh, 2020-05-22:
Yup. Doing that can be really hard on a fairly large design with many components though.

Xiretza, 2020-05-22:
FYI, there is a possibly slightly less tedious way to find the source of these warnings: if compiled with gcc backend and libbacktrace support, ghdl can print a backtrace for failing assertions. Compile all your VHDL sources with `-g` and run simulation with `--assert-level=warning` to make sure the assertion causes a simulation failure and you will get a backtrace. This will of course kill the simulation and it's not possible to skip certain "acceptable" warnings (except for those at `t=0` using `--ieee-asserts=disable-at-0`).

RocketRoss, 2020-05-22:
>if compiled with gcc backend and libbacktrace support, ghdl can print a backtrace for failing assertions. Compile all your VHDL sources with -g and run simulation with --assert-level=warning to make sure the assertion causes a simulation failure and you will get a backtrace.

I don't know if this is as neatly stated in the docs...

tgingold, 2020-05-22:
So what about adding an option like --assert-backtrace-level=xxx, to display backtraces from a certain level ?

diogratia, 2020-05-23:
here are also some number of numeric_std."=" warnings. Package numeric_std warnings show up for both icache and dcache.

There are 98 distinct to_integer calls in the VHDL code.

You could use the grt flag `--trace-processes`:

```none
 ...
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).itlb_lookup [00007FD084007000]
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).itlb_update [00007FD084007000]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).icache_comb [00007FD084007000]
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).icache_hit [00007FD084007000]
...
```

How far you're willing to peer into your model can be limited with `--stop-time`.

Here arbitrarily set to 200 ns and still finding the same warning:
```none
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).icache_hit [00007FD084007000]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).itlb_update [00007FD084007000]
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@195ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).maybe_plrus.plrus(31).plru@plru(rtl).update_lru [00007FD08505C840]
```
The delta cycle number is also provided. 

We can see there are six uses of the call in `icache.vhdl`:

```vhdl
grep -i -n  to_integer icache.vhdl
210:        return to_integer(unsigned(addr(SET_SIZE_BITS - 1 downto LINE_OFF_BITS)));
216:        return to_integer(unsigned(addr(SET_SIZE_BITS - 1 downto ROW_OFF_BITS)));
261:	return to_integer(unsigned(row_v));
269:        word := to_integer(unsigned(addr(INSN_BITS+2-1 downto 2)));
299:        return to_integer(unsigned(hash));
498:	replace_way <= to_integer(unsigned(plru_victim(req_index)));
```
Only one of which falls in process `ltlb_update`, here on line 498 signal `plru_victim`. Some processes  have multiple `to_integer` calls.

There is general disdain in textbooks for initial values for declarations citing the lack of need for synthesis. Metavalues other than 'Z' are solely intended to tell us something about the model in simulation. IEEE Std 1076.6-2004 tells us that except for signal declarations in packages requiring initial values, initial values for variables and signals are ignored. This tells us it's possible to write simulation friendly models that don't have 'U's (for instance you could initialize cache tags to their invalid binary equivalent value). For those caches without reset or preset (cleared) tags they'd have to be initialized by software action (which may not be happening here, 206.015 us is a long time with a 50 MHz clock).  There's the use of BIT/BIT_VECTOR/NUMERIC_BIT instead, a filter could be written to remove the warnings,...

Enabling `--trace-processes` caused the log to balloon from around 14 MiB to 1.35 GiB with a `stop-time=206015ns`  and caught the issue author's shown assertion:

```none
...
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).decode1_0@decode1(behaviour).decode1_0 [00007FCE94012200]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).fetch2_0@fetch2(behaviour).regs [00007FCE94010A00]
fetch2.vhdl:44:16:@206015ns:(report note): fetch2 rst:'0' S:'1' F:'0' T:'0' V:'1' FF:'0' nia:000000000002B240
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).icache_miss [00007FCE9380C800]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).icache_hit [00007FCE9380C800]
icache.vhdl:536:16:@206015ns:(report note): cache hit nia:000000000002B240 IR:'0' SM:'0' idx:9 tag:000000000056 way:0 RA:0000000002B240
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).itlb_update [00007FCE9380C800]
../../src/ieee2008/numeric_std-body.vhdl:3036:7:@206015ns:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).maybe_plrus.plrus(31).plru@plru(rtl).update_lru [00007FCE950142B0]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).maybe_plrus.plrus(30).plru@plru(rtl).update_lru [00007FCE95013C20]
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).maybe_plrus.plrus(29).plru@plru(rtl).update_lru [00007FCE95013790]
 ...
.core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).rams(1).way@cache_ram(rtl).P0 [00007FCE94BE1770]
 ...
run process .core_tb(behave).soc0@soc(behaviour).processor@core(behave).icache_0@icache(rtl).rams(0).way@cache_ram(rtl).P0 [00007FCE93721620]
 ...
```

ozbenh, 2020-05-23:
Yup. I've fixed all the warnings in microwatt now, I'll send a PR to Anton soon.

Of course, Vivado did strike again...  you would think  that just assigning initial values to block rams (to 0) wouldn't change the design ... bet again. Timing changes. Some times, adding init values to some latches also changes utilization ! (just 0 as init values).

So I had to try quite a few combinations but I managed to get it to a state where it runs without warnings in GHDL and is back to no timing violations in Vivado (though removing a 1 bit mux somewhere causes 8 violations to show up completely elsewhere). How do you HW guys cope with that stuff ? :-)

tgingold, 2020-05-23:
I fear there is no magic bullet.

Usually you shouldn't initialize variables or signals, as that's no synthesizable.  But you could use a reset.
You could also completely ignore those ieee warnings.  After all, the X are propagated to the outputs, so tests will fail if the output depends on an uninitialized value.

eine, 2020-05-23:
> How do you HW guys cope with that stuff ? :-)

Let's say were making our best as an open source and free community not to cope with that never again... 😆

FTR, two months ago there was an interesting discussion about "Provide granular control over numeric_std warnings" in the VASG reflector: http://grouper.ieee.org/groups/1076/email/msg01180.html

aspdigital, 2020-05-23:
@tgingold  -- Xilinx recommends that designers use initializers on signals in preference to an asynchronous reset. There are two related reasons for this.

One, they've said for literally decades that the global set/reset net (GSR), the net on which async reset/preset is driven, is "slow" and they won't guarantee all flip-flops will come out of reset on the same clock tick. The static timing analyzer doesn't account for this. Thus their "rule" is "don't use a global async reset in your designs." The reset/preset inputs on the flip-flops can be configured as synchronous or asynchronous, and at least for Vivado, ISE, Mentor Precision and Synplify, the synthesis will infer the correct reset type from the HDL code and configure the flip-flop properly. 

(Aside: as the flip-flop itself does the reset/preset, in Xilinx devices there is no penalty for using a sync reset. Other devices will build the synchronous reset out of CLB logic. I took some module I wrote for a Xilinx device and ported it to MicroSemi ProASIC-3E and the sync reset added a timing-killing level of logic.) 

For purposes of code portability, Xilinx synthesis offers a feature that will "helpfully" convert all asynchronous resets to synchronous.

Two, and the reason for using the signal initializer, is that in Xilinx FPGAs, configuration is a logic reset. The devices guarantee that once the configuration is complete, the magic signal (and I forget what it is at the moment) indicating such ensures that all flip-flops in a design assume their default state of '0' at the same time. Using the signal initializer allows the designer to set that default state to something other than '0'.

As for adding initializers to block RAMs -- the reset of a block RAM, as such, should affect only the output register, not the memory block itself. In the Xilinx world one can initialize a BRAM from a text file. (One can initialize a ROM with initializers.) My guess is that when @ozbenh wrote the initializer, synthesis said, "you can't use an initializer on a BRAM, so I'm going to give you a large block of flip-flops." That kills both timing and area. Been there, done that.

As for latches -- that depends on how they are described. Async latch inference is greatly discouraged in FPGAs for timing-analysis reasons (you can't analyze the timing), and ISE and Vivado and the others complain when you code one. If Ben H could post a code snippet of his latch, we could see what he's doing.

tgingold, 2020-05-23:
I agree with you, but only if you target an sram fpga (initial values are ignored for e.g. Proasic) and if you don't plan to reset your fpga multiple times.

ozbenh, 2020-05-23:
Actually the initialiser on the block rams didn't seem to turn then into LUTs since it improved timing for me without changing utilisation. Odd. In fact it doesn't make sense at all...

Ask for resets, this discussion is a good illustration of the result of trying to do research on that subject, everybody disagrees :-)

It does look like flop initializers are supposed to be free at least on xilinx (except in a couple of cases I found where adding them changed utilisation and timing, again, go figure what vivado is doing...) and sync resets on the minimum amount of stuff that needs to be reset for a "runtime" reset to work.

tmeissner, 2020-05-24:
We use async resets with synched deassertion for all of our designs, regardless of the target FPGA platform (Lattice, Microsemi, Xilinx). We have to reset our designs in various cases, not only on start-up.

If I remember, for Microsemi FPGA you get a warning during synthesis if you’ve forgotten a Reset on a FF. Something like “Add a Reset to improve timing”.

Furthermore we get the same behavior in verification and synthesis if you start with a clean reset.

umarcor, 2020-05-25:
> I don't know if this is as neatly stated in the docs...

There is https://ghdl.readthedocs.io/en/latest/development/Debugging.html. Up to proposing some enhancement? Say, some note and a ref to @diogratia's comment...

tgingold, 2020-05-27:
Fixed with the addition of --backtrace-severity.
Maybe this will require improvements to the backtrace!

closed 2020-05-27 by tgingold