ghdl/ghdl #1339: GHDL fails to build with gnat-2020 Community Edition

AppMathIsRealMath, 2020-05-23:
**Description**
A plain ./configure builds successfully on gnat-2019 CE but a successful build on gnat-2020 CE requires adding "--disable-werror"

The build error is:

`./src/ortho/mcode -I./src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe -gnatw.A -I- /home/sam/.usr/local/src/ghdl/src/grt/grt-main.adb`
`grt-main.adb:136:17: warning: object "False" is always False at this point`
`grt-main.adb:156:17: warning: object "False" is always False at this point`
`grt-main.adb:160:14: warning: object "True" is always True at this point`
`gnatmake: "/home/sam/.usr/local/src/ghdl/src/grt/grt-main.adb" compilation error`

**Expected behaviour**
A successful build without warnings.

**How to reproduce?**
With current master branch and gnat-2020 CE:

./configure
make

**Context**
Please, provide the following information:

- OS:
Solus Linux with a manually downloaded and installed gnat-2020 CE
- Origin:
  - [x] Built from sources: `bb270862b25e863901008e59e4a5eabcc0b3e007`

tgingold, 2020-05-23:
Humm, better to turn off this warning, as it is clearly noise.  -gnatwC should be the option to add.

AppMathIsRealMath, 2020-05-23:
I added "-gnatwC" to Makefile.in and got a bit further with another warning that terminated compilation:

`gcc -c -I./ -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -I./src/synth -gnaty3befhkmr -gnatwa -gnatf -gnatwC -g -gnata -gnatwe -gnatw.A -I- /home/sam/.usr/local/src/ghdl/src/grt/grt-disp_rti.adb`
`grt-disp_rti.adb:895:40: warning: writable actual for "Obj_Loc" overlaps with actual for "Addr"`
`gnatmake: "/home/sam/.usr/local/src/ghdl/src/grt/grt-disp_rti.adb" compilation error`
`make: *** [Makefile:184: ghdl_mcode] Error 4`

Sorry for the piecemeal issues - I don't know Ada and which warnings are important.

AppMathIsRealMath, 2020-05-23:
By disabling the configure -werror option I found the complete set of Ada warnings:

`grt-main.adb:136:17: warning: object "False" is always False at this point`
`grt-main.adb:156:17: warning: object "False" is always False at this point`
`grt-main.adb:160:14: warning: object "True" is always True at this point`

`grt-disp_rti.adb:895:40: warning: writable actual for "Obj_Loc" overlaps with actual for "Addr"`

`grt-fst.adb:332:34: warning: writable actual for "Ref" overlaps with actual for "Res"`

`grt-vcd.adb:359:35: warning: writable actual for "Obj_Loc" overlaps with actual for "Addr"`

`grt-vital_annotate.adb:176:22: warning: writable actual for "Cur" overlaps with actual for "Res"`
`grt-vital_annotate.adb:488:28: warning: writable actual for "Cur" overlaps with actual for "Res"`

`grt-avhpi_utils.adb:41:43: warning: writable actual for "Ref" overlaps with actual for "Res"`

tgingold, 2020-05-23:
Thanks.  I have just downloaded gnatgpl-2020 and fixing those warnings.
The overlap warning is correct.

tgingold, 2020-05-23:
SHould be fixed.

closed 2020-05-23 by tgingold

AppMathIsRealMath, 2020-05-23:
Fix confirmed working, thanks!