ghdl/ghdl #1340: Assertion failed with (others => signal) on component instanciation

ozbenh, 2020-05-24:
**Description**
I tried but failed to create a small repro-case so far, so I'm afraid you might need microwatt, but if you apply this patch:

```
--- a/cache_ram.vhdl
+++ b/cache_ram.vhdl
@@ -16,7 +16,7 @@ entity cache_ram is
        rd_en   : in  std_logic;
        rd_addr : in  std_logic_vector(ROW_BITS - 1 downto 0);
        rd_data : out std_logic_vector(WIDTH - 1 downto 0);
-       wr_en   : in  std_logic;
+--     wr_en   : in  std_logic;
        wr_sel  : in  std_logic_vector(WIDTH/8 - 1 downto 0);
        wr_addr : in  std_logic_vector(ROW_BITS - 1 downto 0);
        wr_data : in  std_logic_vector(WIDTH - 1 downto 0)
@@ -43,7 +43,7 @@ begin
        variable widx : integer range 0 to SIZE - 1;
     begin
        if rising_edge(clk) then
-           if wr_en = '1' then
+--         if wr_en = '1' then
                if TRACE then
                    report "write a:" & to_hstring(wr_addr) &
                        " sel:" & to_hstring(wr_sel) &
@@ -65,7 +65,7 @@ begin
                        " dat:" & to_hstring(ram(to_integer(unsigned(rd_addr))));
                end if;
            end if;
-       end if;
+--     end if;
     end process;
 
     buf: if ADD_BUF generate
diff --git a/dcache.vhdl b/dcache.vhdl
index cf09afd..a9223b3 100644
--- a/dcache.vhdl
+++ b/dcache.vhdl
@@ -956,7 +956,7 @@ begin
                rd_en   => do_read,
                rd_addr => rd_addr,
                rd_data => dout,
-               wr_en   => do_write,
+--             wr_en   => do_write,
                wr_sel  => wr_sel,
                wr_addr => wr_addr,
                wr_data => wr_data
diff --git a/icache.vhdl b/icache.vhdl
index fd0a4b4..f621f73 100644
--- a/icache.vhdl
+++ b/icache.vhdl
@@ -355,8 +355,8 @@ begin
                rd_en   => do_read,
                rd_addr => rd_addr,
                rd_data => dout,
-               wr_en   => do_write,
-               wr_sel  => (others => '1'),
+--             wr_en   => do_write,
+               wr_sel  => (others => do_write),
                wr_addr => wr_addr,
                wr_data => wishbone_in.dat
                );
```
Then we get the GHDL internal error below. (Yes I know I still need to gate wr_sel with do_write in dcache ... but this is enough to show the error).

**How to reproduce?**
Grab microwatt (I assume you know where from :-) and apply the above patch. Then do "make" to make core_tb and observe it break.

**Context**
Please, provide the following information:

- OS: Fedora 32
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [* ] Built from sources:
```
$ ghdl --version
GHDL 1.0-dev (v0.37.0-652-g4e5ab2e5) [Dunoon edition]
 Compiled with GNAT Version: 10.1.1 20200507 (Red Hat 10.1.1-
 llvm code generator
Written by Tristan Gingold.
```

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-579-g7285f745) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/ANT.AMAZON.COM/benh/hackplace/microwatt/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 --work=unisim -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ --anaelab core_tb --ghdl-source=decode_types.vhdl --ghdl-source=common.vhdl --ghdl-source=wishbone_types.vhdl --ghdl-source=fetch1.vhdl --ghdl-source=fetch2.vhdl --ghdl-source=utils.vhdl --ghdl-source=plru.vhdl --ghdl-source=cache_ram.vhdl --ghdl-source=icache.vhdl --ghdl-source=decode1.vhdl --ghdl-source=helpers.vhdl --ghdl-source=insn_helpers.vhdl --ghdl-source=gpr_hazard.vhdl --ghdl-source=cr_hazard.vhdl --ghdl-source=control.vhdl --ghdl-source=decode2.vhdl --ghdl-source=register_file.vhdl --ghdl-source=cr_file.vhdl --ghdl-source=crhelpers.vhdl --ghdl-source=ppc_fx_insns.vhdl --ghdl-source=rotator.vhdl --ghdl-source=logical.vhdl --ghdl-source=countzero.vhdl --ghdl-source=multiply.vhdl --ghdl-source=divider.vhdl --ghdl-source=execute1.vhdl --ghdl-source=loadstore1.vhdl --ghdl-source=mmu.vhdl --ghdl-source=dcache.vhdl --ghdl-source=writeback.vhdl --ghdl-source=core_debug.vhdl --ghdl-source=core.vhdl --ghdl-source=wishbone_arbiter.vhdl --ghdl-source=wishbone_bram_wrapper.vhdl --ghdl-source=wishbone_debug_master.vhdl --ghdl-source=xics.vhdl --ghdl-source=syscon.vhdl --ghdl-source=soc.vhdl --ghdl-source=sim_console.vhdl --ghdl-source=sim_uart.vhdl --ghdl-source=sim_bram_helpers.vhdl --ghdl-source=sim_bram.vhdl --ghdl-source=sim_jtag_socket.vhdl --ghdl-source=sim_jtag.vhdl --ghdl-source=sim-unisim/BSCANE2.vhdl --ghdl-source=sim-unisim/BUFG.vhdl --ghdl-source=sim-unisim/unisim_vcomponents.vhdl --ghdl-source=dmi_dtm_xilinx.vhdl --ghdl-source=core_tb.vhdl -c -fpic -o e~core_tb.o e~core_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:555
Call stack traceback locations:
0x4a1fee 0x4a2464 0x4a2755 0x4abc17 0x6268d2 0x638940 0x5d2498 0x5f8126 0x5f7ae5 0x5b147b 0x5b1632 0x5b1e98 0x5b2b48 0x5b664c 0x5b677f 0x5bac79 0x5c00f1 0x5bff57 0x5c01de 0x648897 0x58c4b5 0x5892f5 0x5980c6 0x6503f3 0x65208b 0x4086a9 0x7fdae30adb95 0x407998 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
Makefile:71: recipe for target 'core_tb' failed
```

tgingold, 2020-05-25:
I think this is similar to issue #1129.

GHDL needs to declare an anonymous signal using bounds defined later.  Needs some specialized circuitry.

As a workaround, you can declare manually an intermediate signal.

ozbenh, 2020-05-25:
Ok, quite possibly. Yes a workaround with an intermediate signal works :-) I thought you might want to know about the issue nonetheless.

tgingold, 2020-05-25:
Sure, let me know all the issues!

tgingold, 2021-09-07:
Should have been fixed.

closed 2021-09-07 by tgingold