ghdl/ghdl #1341: Unexpected "declaration of a too large object" error

ozbenh, 2020-05-25:
**Description**
I'm trying to declare a memory based on some generics and I get an unexpected runtime error
```
./test:error: declaration of a too large object (384 > --max-stack-alloc=128 KB)
./test:error: error during elaboration
```

The object I'm trying to allocate is less than 64k. If I replace the size calculation with a hard wired 64k it works fine. I added reports to verify the constant is calculated properly

**How to reproduce?**

A small'ish test case is at:

https://ozlabs.org/~benh/test_decl_too_large.tar.gz

Elaborate it with:
```
ghdl -c --std=08 wishbone_types.vhdl litedram-initmem.vhdl test.vhdl -e test
```

Run ./test and see the error. Now if you edit litedram_wrapper.vhdl and locate those two lines:
```
    type ram_t is array(0 to TOTAL_RAM_WORDS - 1) of std_logic_vector(31 downto 0);
--    type ram_t is array(0 to (65536 / 4) - 1) of std_logic_vector(31 downto 0);
```

And flip the comment (comment the first one, uncomment the second one), it works fine and you can observe the report below that shows that the calculation is ok.

- OS: Ubuntu 18.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ *] Built from sources:
```
$ ghdl --version
GHDL 1.0-dev (v0.37.0-579-g7285f745) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
 llvm code generator
```

umarcor, 2020-05-25:
I think the size is `65536 / 4 * 32 / 1024 = 512KB`. Each std_logic is an enum of size byte. Hence, since `32=2**5`, to stay below `2**17`, can you try `(0 to 2**(17-5)-1) of std_logic_vector(31 downto 0)`?

See also comments about `max-stack` and `ulimit` in https://ghdl.github.io/ghdl-cosim/vhpidirect/mistakes.html. Note that limits seem to be different depending on the backend: #822.

tgingold, 2020-05-25:
The report is:

litedram-initmem.vhdl:39:9:@0ms:(report note): Extra payload size:32768 Total:49155

So, the total allocation is 48K * 8 std_logic so 384KB.

The message is displayed only for dynamic allocation.

ozbenh, 2020-05-25:
Ouch, GHDL allocates a byte per-bit for memories ? That's rather ... painful :-) I suppose that's why Anton created a separate C module for the main RAM of Microwatt.

So when I specify 64k explicitly it's considered a static allocation rather than dynamic and not on the stack then ?

umarcor, 2020-05-25:
Not for memories, but for signals/variables of type std_logic, which is a 9-value type. So, you are requesting it explicitly by using type `std_logic_vector`. For simulation, GHDL does not have any concept of "a memory". That's something that belongs to synthesis.

Moreover, I think that, in fact, GHDL uses bytes for bit or boolean too, according to https://ghdl.github.io/ghdl-cosim/vhpidirect/declarations.html#restrictions-on-type-declarations. Overall, restrictions on type declarations provide information about GHDL's internal representation of data types; it's not exclusive to VHPIDIRECT.

ozbenh, 2020-05-25:
Thanks. Makes sense.

closed 2020-05-25 by ozbenh