ghdl/ghdl #1342: Support for VHDL-2008 unary operators

rlee287, 2020-05-26:
**Is your feature request related to a problem? Please describe.**
GHDL `master` branch (4e5ab2e5 as of the time of writing) does not support VHDL-2008 unary operators:

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

entity test_unop is
    Port (vec_in: in STD_LOGIC_VECTOR(31 downto 0);
         vec_out_and: out STD_LOGIC_VECTOR(31 downto 0);
         vec_out_or: out STD_LOGIC_VECTOR(31 downto 0));
end entity;

architecture Behavioral of test_unop is
begin
    vec_out_and <= and vec_in;
    vec_out_or <= or vec_in;
end Behavioral;
```

```
$ ghdl -a --std=08 test.vhd
test.vhd:12:20:error: no function declarations for operator "and"
test.vhd:13:19:error: no function declarations for operator "or"
ghdl:error: compilation error
$ ghdl --synth --std=08 test.vhd -e test_unop
test.vhd:12:20:error: no function declarations for operator "and"
test.vhd:13:19:error: no function declarations for operator "or"
```

**Describe the solution you'd like**
Implementation of the VHDL-2008 unary operators.

**Describe alternatives you've considered**
There are workarounds but implementing this feature would improve support for the VHDL-2008 standard.

tgingold, 2020-05-26:
The result of reduction is std_logic, not a vector!

May you'd like a better error message.

rlee287, 2020-05-26:
I think better error messages would be nice, but I would file a new issue for that to consolidate feedback of that type. For now I have refocused this issue on implementing the rest of the unary operators other than `and` and `or`. Please let me know if you think this would work better as a new issue.

tgingold, 2020-05-27:
Yes, please create a new issue if you think the error message could be improved.

commit 67f926fc (2020-05-27): synth: handle reduction operators.  Fix #1342
closed 2020-05-27 by tgingold