ghdl/ghdl #1344: Wrong compilation order in compile-uvvm.sh script

mariuselv, 2020-05-27:
**Description**
Calling compile-uvvm.sh -a will fail due to wrong compilation order of UVVM VIPs

**Expected behaviour**
UVVM Utility Library and UVVM VVC Framework will compile, but compiling Bitvis VIP SBI will fail because the SBI VIP has Scoreboard VIP dependency. So, the bitvis_vip_scoreboard has to be moved to the beginning of the uvvm_vips list on line 60 in compile-uvvm.sh script.

**How to reproduce?**
Run: compile-uvvm.sh -a

mariuselv, 2020-05-28:
There is a `script/component_list.txt` with the correct compile order which would better to utilize rather than hard coding the compile order in the compile-uvvm.sh script.

closed 2020-05-31 by tgingold

tpcorrea, 2021-01-17:
GHDL has updated both shell and PowerShell script, so this issue on GHDL is fixed. Nevertheless, some VIPs fail to compile with GHDL.

commit 84a5ac49 (2020-05-31): compile-uvvm.sh: fix order.  Fix #1344