ghdl/ghdl #1345: synth/PSL: Crash when using PSL eventually! operator

tmeissner, 2020-05-28:
**Description**
The PSL `eventually!` operator isn't supported in synthesis yet. If you use that operator, it leads to an GHDL crash. For simulation it is supported (I only tried analysis & elaboration, not a running simulation). 

**Expected behaviour**
The  PSL `eventually!` operator should be supported and work for both simulation & synthesis.
This would make formal verification of liveness properties possible.

**How to reproduce?**

Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_1345.vhd

The error occurs in synthesis. Analysis & Elaboration for Simulation works:

```
ghdl -a --std=08 issue_1345.vhd
ghdl -e --std=08 issue
```

but synthesis not:

```sh :image: ghdl/synth:formal
ghdl --synth --std=08 issue_1345.vhd -e issue
```

**Context**

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl/synth:formal)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/build/issues/
Command line:
ghdl --synth --std=08 issue_1345.vhd -e issue
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-stmts.adb:3141
Call stack traceback locations:
0x7f68713104f1 0x563c1322ef4e 0x563c1322f45f 0x563c1322f950 0x563c1322fc78 0x563c13230597 0x563c13230619 0x563c1321ee55 0x563c1321efbc 0x563c13218182 0x563c13299837 0x563c131abbcf 0x563c132a3127 0x563c13058fc5 0x7f68
```

tmeissner, 2020-05-28:
It seems that it has something to do with missing Location info in function `Synth_PSL_Expression()` in `synth-stmts.adb`:

```ada
   function Synth_PSL_Expression
     (Syn_Inst : Synth_Instance_Acc; Expr : PSL.Types.PSL_Node) return Net
   is
      use PSL.Types;
      use PSL.Nodes;

      Ctxt : constant Context_Acc := Get_Build (Syn_Inst);
      Loc : constant Location_Type := Get_Location (Expr);
      Res : Net;
   begin
      case Get_Kind (Expr) is
         when N_HDL_Bool =>
           ...
         when N_And_Bool =>
            pragma Assert (Loc /= No_Location);
         ...
```

If I looked correctly, the Location info is created in function `Parse_FL_Property_1()` in `vhdl-parse_psl.adb`:

```ada
   function Parse_FL_Property_1 return Node
   is
      Res : Node;
      Tmp : Node;
   begin
      case Current_Token is
         when Tok_Always =>
            Res := Create_Node_Loc (N_Always);
            Scan;
            Set_Property (Res, Parse_FL_Property (Prio_FL_Invariance));
         ...
         when Tok_Eventually_Em =>
            Res := Create_Node_Loc (N_Eventually);
            Scan;
            Set_Property (Res, Parse_FL_Property (Prio_FL_Occurence));
         when Tok_Next =>
            Res := Create_Node_Loc (N_Next);
```

Sadly, this bug seems not to be a fruit which is hanging low enough that I could fix it :neutral_face:

tgingold, 2020-05-28:
Yes, the location is missing on the N_EOS node (end of simulation).  I have a fix for it.

closed 2020-05-29 by tgingold

tmeissner, 2020-05-29:
@tgingold thanks for the fix.

My tests raised a question: If I have a code snippet like this:

```vhdl
  SEQ_A : sequencer generic map ("__-__-____-____") port map (clk, a);
  SEQ_B : sequencer generic map ("_______-_______") port map (clk, b);

  -- All is sensitive to rising edge of clk
  default clock is rising_edge(clk);

  -- This assertion holds
  -- This assertion leads to a GHDL synthesis crash with bug report
  EVENTUALLY_a : assert always (a -> eventually! b);
```

Then the assertion should fail. It does fail if I simulate with `--stop-time` option. It does not fail when I use the `stop()` procedure of `std.env` package. I assume that it has something to do with the exact meaning of end of simulation?

It also does not fail in SymbiYosys, but I don't know, if SymbiYosys is really supporting liveness properties. In my understanding the assert should fail in unbounded proof, but maybe not in bounded model check.  The bounded model check only checks the correctness for a given number of cycles. What is after these cycles doesn't matter, even if `b` holds some cycle after the cycles the bmc checks.

tgingold, 2020-05-29:
> Then the assertion should fail. It does fail if I simulate with --stop-time option. It does not fail when I use the stop() procedure of std.env package. I assume that it has something to do with the exact meaning of end of simulation?

Ah, that could be changed.

> It also does not fail in SymbiYosys, but I don't know, if SymbiYosys is really supporting liveness properties. In my understanding the assert should fail in unbounded proof, but maybe not in bounded [...]

As far as I understand, only boolector supports liveness.  But I am not sure that the definition of liveness for SymbiYosys is really what we need.  That's something I investigated a little bit but not enough.

tmeissner, 2020-05-29:
> Ah, that could be changed.

That would be nice 😃 

> As far as I understand, only boolector supports liveness.  But I am not sure that the definition of liveness for SymbiYosys is really what we need.  That's something I investigated a little bit but not enough.

I will try to use boolector ... from GHDL side, it is enough to support it as it is implemented now. With the small change with above ...

tgingold, 2020-05-30:
> Ah, that could be changed.
> 
> That would be nice 😃

Please, open an issue with a testcase!

tmeissner, 2020-05-30:
Okay, I will create an issue.

I also tried the liveness mode in SymbiYosys. Wow, what a mess.

It is supported only by `avy` & `super_prove` solvers which both work with the Aiger engine. I didn't get `avy` to build (some serious compile erros), `super_prove` can be build if you patch the source before. Then it compiles with a lot of warning messages. The developers of these solvers seem to be better mathematicians than programmers 😄 

But using the built `super_prove` with SymbiYosys liveness mode don't works, I get a Python exception from SymbiYosys. It seems, that the aiger-model is nearly empty. `super_prove` processes these aiger model and outputs the result: two empty files. Maybe the empty models are caused by optimizations Yosys does, I don't know.

tgingold, 2020-05-31:
Yes, I have some doubts about the current support of liveness and fairness in Yosys.  I am not sure it is widely used.  I am even not sure it is working correctly.

commit 26a94c92 (2020-05-29): psl-nodes: set a location on false/true/one/EOS nodes.  Fix #1345