ghdl/ghdl #1347: eventually! behaviour when simulation ends using std.env stop() or finish()

tmeissner, 2020-05-30:
**Description**
This is a follow-up to #1345 

When simulation ends using the stop() or finish() procedures, an assert using eventually! never fails. If the simulation is ended using the GHDL stop-time option, then eventually! asserts which could not fullfilled are reported as failed at the end of simulation.

**Expected behaviour**
Mhm. Not that clear. The two procedures stop() and finish() have a different meaning as defined in the VHDL-08 standard:

From section 16.5:
> Execution of the STOP procedures causes the same action by the host simulator as that caused by the vhpi_control function called with the argument vhpiStop (see 23.5). Execution of the FINISH procedures causes the same action by the host simulator as that caused by the vhpi_control function called with the argument vhpiFinish (see 23.5). Execution shall not return to the VHDL description after a call to the FINISH procedure.

From section 23.5:

> — If command is vhpiStop, after control returns to the tool from the callback function from which the vhpi_control function was invoked, the tool stops simulation then accepts further directives from an interactive user or a command source. Additional arguments to vhpi_control beyond the command argument may be interpreted by the tool in an implementation-defined manner. The tool shall provide an implementation-defined default action if no additional arguments are provided.
> — If command is vhpiFinish, after control returns to the tool from the callback function from which the vhpi_control function was invoked, the tool enters the termination phase (see 20.10). Additional arguments to vhpi_control beyond the command argument may be interpreted by the tool in an implementation-defined manner. The tool shall provide an implementation-defined default action if no additional arguments are provided. 

Does that mean, that after executing stop() the simulation could be continued from that point, if the simulator would support that? In that case, a PSL eventually! assert could be full-filled in that run. Or is the simulation ended in both cases without the possibility of continue it. Then, the assert should fail in both cases, I think.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_1345.vhd

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 issue_13xx.vhd
ghdl -e --std=08 test_issue
```

If you run it with stop-time option:

```sh :image: ghdl/ghdl:buster-mcode
./test_issue --psl-report=psl_coverage.json --stop-time=20ns
./tb_issue:info: simulation stopped by --stop-time @20ns
issue_13__.vhd:90:3:@20ns:(psl assertion error): Assertion violation
```

if not using the stop-time option:

```sh :image: ghdl/ghdl:buster-mcode
./test_issue --psl-report=psl_coverage.json --stop-time=20ns
simulation stopped @29ns with status 0
```

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl:buster-mcode)

tgingold, 2020-05-31:
Ok, but how to handle the exit status ?
If you call `finish(0)` and a PSL assertion fails, should the exit status still be 0 ?

tmeissner, 2020-05-31:
At the moment, at PSL assert errors, the exit status is 0. We could let that as it is. I look in the PSL report file after the simulation anyway.

tgingold, 2020-06-01:
Ok, so I close this issue.

closed 2020-06-01 by tgingold