ghdl/ghdl #1348: Slightly confusing synthesis of memories

Xiretza, 2020-06-01:
**Description**
First of all, thanks a lot for implementing #1069! From what I can tell, it works prefectly fine. However, I have found a few "cosmetic" issues:

1. Memories declared as `(0 to x)` are converted to `(x downto 0)` internally, with all address inputs wrapped in `unsigned'("111111111111") - unsigned(address)`. Since this applies to all address inputs, the behaviour works out correctly in the end, but it still creates a lot of unnecessary cells.
2. Multi-word reads, such as those on the bigger port of memories with asymmetric ports, are always asynchronous. Instead of using the memory's read-enable pin, the memory is always read asynchronously, and discrete muxes and flip-flops are places on the output for the read-enable. In theory, this is also equivalent, but synthesis tools might have a hard time mapping it to device resources.
3. The ordering of read and write ports is different. This might only be a consequence of (2), and logically, it has absolutely no impact, but it might make for some annoying debugging hurdles.

**Expected behaviour**
The least amount of logic necessary is used.

**How to reproduce?**

```vhd :file: sdp_simple.vhd
library ieee;
use ieee.std_logic_1164.all,
    ieee.numeric_std.all;

entity sdp_simple is
    port (
        clk          : in  std_logic;

        read_a       : in  std_logic;
        write_a      : in  std_logic;
        byteen_a     : in  std_logic_vector(0 downto 0);
        addr_a       : in  std_logic_vector(11 downto 0);
        data_read_a  : out std_logic_vector(7 downto 0);
        data_write_a : in  std_logic_vector(7 downto 0);

        read_b       : in  std_logic;
        write_b      : in  std_logic;
        byteen_b     : in  std_logic_vector(3 downto 0);
        addr_b       : in  std_logic_vector(9 downto 0);
        data_read_b  : out std_logic_vector(31 downto 0);
        data_write_b : in  std_logic_vector(31 downto 0)
    );
end sdp_simple;

architecture behavioral of sdp_simple is
    type ram_t is array(0 to 4095) of std_logic_vector(7 downto 0);
    signal store : ram_t := (others => (others => '0'));
begin
    process(clk)
    begin
        if rising_edge(clk) then
            if write_a = '1' and byteen_a(0) = '1' then
                store(to_integer(unsigned(addr_a))) <= data_write_a;
            end if;

            if read_a = '1' then
                data_read_a <= store(to_integer(unsigned(addr_a)));
            end if;

            for i in 0 to 3 loop
                if write_b = '1' and byteen_b(i) = '1' then
                    store(to_integer(unsigned(addr_b) & to_unsigned(i, 2))) <=
                        data_write_b((i+1) * 8 - 1 downto i * 8);
                end if;

                if read_b = '1' then
                    data_read_b((i+1) * 8 - 1 downto i * 8) <=
                        store(to_integer(unsigned(addr_b) & to_unsigned(i, 2)));
                end if;
            end loop;
        end if;
    end process;
end behavioral;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth sdp_simple.vhd -e
# for visualization:
yosys -m ghdl -p 'ghdl sdp_simple.vhd -e; show'
```

[image]

As can be seen above, there is a lot more logic than there has to be. The bottom-left part are the subtractions to invert the memory direction, the top right are the muxes controlled by `read_b`, which could just be routed to the memory's `RD_EN` input instead, like `read_a` already is. Additionally, for the read ports (`RD_ADDR`, `RD_CLK`, `RD_EN`, `RD_DATA`), `A` occupies the most significant bits, while in the write ports (`WR_ADDR`, `WR_DATA`, `WR_EN`), it's in the leat significant bits. Here's an equivalent circuit with all the above fixed, and using only 5 AND gates of logic:

[image]

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin: Built from sources: 8f6a304b

If a `GHDL Bug occurred` block is shown in the log, please paste it here: N/A

tgingold, 2020-06-02:
Humm, you like big issues!

tgingold, 2020-06-02:
Should be fixed.  I assume I have to wait for the next memory issue...

commit 2edae856 (2020-06-02): netlists-memories: simplify addresses when using a 'to' range.  Fix #1348
closed 2020-06-02 by tgingold