ghdl/ghdl #1349: Analysing memories with LLVM backend very slow

gardners, 2020-06-02:
**Description**
Attempting to analyse a RAM of ~384KB runs for more than an hour.

**Expected behaviour**
It should analyse in a few seconds.

**How to reproduce?**

```
git clone https://github.com/MEGA65/mega65-core.git
cd mega65-core
git checkout ghdl-llvm-slow-analysis
make simulate-llvm
```

**Context**
Please, provide the following information:

- OS: Linux Ubuntu 
```
$ uname -a
Linux F96NG92-L 4.15.0-91-generic #92-Ubuntu SMP Fri Feb 28 11:09:48 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - X Built from sources: `commit eaae9cb33f67aa6cf230e345724511b4b4d5c9cf`

**Additional context**
This problem used to happen with the mcode backend, but was fixed. 
(Incidentally, I'm trying the llvm backend so that I can get backtraces on metavalue warnings, because I am seeing some other weird behaviour there.)

tgingold, 2020-06-02:
When I try `make simulate-llvm`, I get:

`make: *** No rule to make target `src/vhdl/s27kl0641.vhdl', needed by `simulate-llvm'.  Stop.`

But assuming most of the time is spent in files like shadowram_new.vhdl, you can drastically reduce analysis time by using this style:

```
 type ram_t is array (0 to 131071) of unsigned(7 downto 0);
  constant init_ram : ram_t := (
          x"00", -- $00000
          x"00", -- $00001
          x"00", -- $00002
...
          x"00", -- $1fffe
          x"00"); -- $1ffff
  shared variable ram : ram_t := init_ram;
```

So: positional aggregate and a constant for the initial value.

That's a workaround, but a pretty effective one!

gardners, 2020-06-02:
Thanks! I'll give it a punt. I'll tweak the code that auto-generates the shadowram.vhdl file.
Meanwhile, the analysis without the workaround is still running, after 8 hours :)

And, yes, sorry about that missing file.  Its a proprietary file from the hyperram manufacturer, so I don't have it in the repo.

Paul.

gardners, 2020-06-02:
I've made the change, and set it analysing. It's been going for a couple of minutes. I'll see in the morning if it has completed, but either way, its still slow even using the constant initialisation.

tgingold, 2020-06-02:
No, it should take less than 1 minute.

Did you also removed the association in the aggregate ?

gardners, 2020-06-02:
It's still going this morning.  Here is how the file looks now. I've also push the change to the generator (and removed that s27k... file from the dependency list).

ghdl commit I built from is: eaae9cb33f67aa6cf230e345724511b4b4d5c9cf (HEAD, tag: nightly)

```
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use ieee.numeric_std.all;

--
entity shadowram is
  port (ClkA : in std_logic;
        addressa : in integer range 0 to 1048575;
        wea : in std_logic;
        dia : in unsigned(7 downto 0);
        writes : out unsigned(7 downto 0);
        no_writes : out unsigned(7 downto 0);
        doa : out unsigned(7 downto 0);
        ClkB : in std_logic;
        addressb : in unsigned(19 downto 0);
        dob : out unsigned(7 downto 0)
        );
end shadowram;

architecture Behavioral of shadowram is

  signal write_count : unsigned(7 downto 0) := x"00";
  signal no_write_count : unsigned(7 downto 0) := x"00";
  
  type ram_t is array (0 to 393215) of unsigned(7 downto 0);
  constant initram : ram_t := (
          0 => x"00", -- $00000
          1 => x"00", -- $00001
...
          393214 => x"00", -- $5fffe
          393215 => x"00"); -- $5ffff
  shared variable ram : ram_t := initram;
begin

  writes <= write_count;
  no_writes <= no_write_count;
--process for read and write operation.
  PROCESS(ClkA)
  BEGIN
    if(rising_edge(ClkA)) then 
      if wea /= '0' then
        write_count <= write_count + 1;        
          ram(addressa) := dia;
      else
        no_write_count <= no_write_count + 1;        
      end if;
        doa <= ram(addressa);
    end if;
  END PROCESS;
PROCESS(ClkB)
BEGIN
  if(rising_edge(ClkB)) then
      dob <= ram(to_integer(addressb));
  end if;
END PROCESS;

end Behavioral;
```

tmeissner, 2020-06-02:
I think, @tgingold also meant, that you should positional assigmnemt instead of named one in the constant definition:

```vhdl
  constant initram : ram_t := (
          x"00", -- $00000
          x"00", -- $00001
...
          x"00", -- $5fffe
          x"00"); -- $5ffff
```

instead of 

```vhdl
  constant initram : ram_t := (
          0 => x"00", -- $00000
          1 => x"00", -- $00001
...
          393214 => x"00", -- $5fffe
          393215 => x"00"); -- $5ffff
```

gardners, 2020-06-02:
Ah, thanks for spotting what I missed. Confirming it now analyses in a couple of seconds.