ghdl/ghdl #1350: Documentation:  Shared

JimLewis, 2020-06-03:
**Description**
In https://ghdl.readthedocs.io/en/latest/quick_start/README.html

The documentation states:
Use -fexplicit and -frelaxed-rules if needed. For instance when relaxing VHDL 2008’s need for shared variables to be protected types, you can use --std=08 -frelaxed-rules.

It was 1076-2000 that deprecated shared variables of a regular type.    In fact, as was noted in 1076-1993, the ability to use a regular type was only intended to be a temporary thing.

Since GHDL seems to be the most pedantic tool I work with, I thought you would like to know.

eine, 2020-06-03:
Hi @JimLewis! I believe the purpose of the comment is to clarify an error that many users report when trying to use shared variables and VHDL 2008. That is, it is not to explain that the change was done in 2008, and I had not understood it in that sense. However, I'm not a native speaker, so I might be biased because I knew what it wanted to mean, before actually reading it.

What do you think about adding the following?

> For instance when relaxing VHDL 2008’s need for shared variables to be protected types **(required since 1076-200)**, you can use --std=08 -frelaxed-rules.

JimLewis, 2020-06-03:
How about:

For instance, if you would like to use VHDL 2008 and also use shared variables with an ordinary type (deprecated in VHDL 2000), you can use --std=08 -frelaxed-rules

eine, 2020-06-04:
Done. Thanks!

closed 2020-06-04 by eine