ghdl/ghdl #1351: GHDL compilation error encountered

gardners, 2020-06-04:
**Description**
Compiling my project causes ghdl to crash, as documented below

**Expected behaviour**
Compiling the project should succeed without error

**How to reproduce?**

With a LLVM backend GHDL in the path:

```
git clone https://github.com/MEGA65/mega65-core.git
cd mega65-core
git checkout ghdl-crash-20200604
time -p make simulate-llvm
```

**Context**
Please, provide the following information:

- OS: Ubuntu Linux 18.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - X Built from sources: eaae9cb33f67aa6cf230e345724511b4b4d5c9cf

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with GNAT Version: 8.4.0
Target: x86_64-linux-gnu
/home/paul/Projects/mega65/mega65-core/
Command line:
/usr/local/bin/ghdl1-llvm -g -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ --anaelab cpu_test --ghdl-source=src/vhdl/cpu_test.vhdl --ghdl-source=src/vhdl/fake_expansion_port.vhdl --ghdl-source=src/vhdl/fake_sdcard.vhdl --ghdl-source=src/vhdl/fake_reconfig.vhdl --ghdl-source=src/vhdl/gen_utils.vhdl --ghdl-source=src/vhdl/conversions.vhdl --ghdl-source=src/vhdl/dummy_uart_monitor.vhdl --ghdl-source=src/vhdl/gs4510.vhdl --ghdl-source=src/vhdl/multiply32.vhdl --ghdl-source=src/vhdl/divider32.vhdl --ghdl-source=src/vhdl/shifter32.vhdl --ghdl-source=src/vhdl/machine.vhdl --ghdl-source=src/vhdl/ddrwrapper.vhdl --ghdl-source=src/vhdl/framepacker.vhdl --ghdl-source=src/vhdl/hyppo.vhdl --ghdl-source=src/vhdl/mega65r2_i2c.vhdl --ghdl-source=src/vhdl/version.vhdl --ghdl-source=src/vhdl/sid_6581.vhdl --ghdl-source=src/vhdl/sid_coeffs.vhdl --ghdl-source=src/vhdl/sid_coeffs_mux.vhdl --ghdl-source=src/vhdl/sid_components.vhdl --ghdl-source=src/vhdl/sid_filters.vhdl --ghdl-source=src/vhdl/sid_voice.vhdl --ghdl-source=src/vhdl/iomapper.vhdl --ghdl-source=src/vhdl/mouse_input.vhdl --ghdl-source=src/vhdl/cia6526.vhdl --ghdl-source=src/vhdl/c65uart.vhdl --ghdl-source=src/vhdl/UART_TX_CTRL.vhdl --ghdl-source=src/vhdl/cputypes.vhdl --ghdl-source=src/vhdl/viciv.vhdl --ghdl-source=src/vhdl/pixel_driver.vhdl --ghdl-source=src/vhdl/pixel_fifo.vhdl --ghdl-source=src/vhdl/frame_generator.vhdl --ghdl-source=src/vhdl/sprite.vhdl --ghdl-source=src/vhdl/vicii_sprites.vhdl --ghdl-source=src/vhdl/bitplane.vhdl --ghdl-source=src/vhdl/bitplanes.vhdl --ghdl-source=src/vhdl/victypes.vhdl --ghdl-source=src/vhdl/pal_simulation.vhdl --ghdl-source=src/vhdl/ghdl_alpha_blend.vhdl --ghdl-source=src/vhdl/rain.vhdl --ghdl-source=src/vhdl/lfsr16.vhdl --ghdl-source=src/vhdl/visual_keyboard.vhdl --ghdl-source=src/vhdl/uart_charrom.vhdl --ghdl-source=src/vhdl/oskmem.vhdl --ghdl-source=src/vhdl/termmem.vhdl --ghdl-source=src/vhdl/sdcardio.vhdl --ghdl-source=src/vhdl/touch.vhdl --ghdl-source=src/vhdl/hyperram.vhdl --ghdl-source=src/vhdl/i2c_master.vhdl --ghdl-source=src/vhdl/i2c_wrapper.vhdl --ghdl-source=src/vhdl/hdmi_i2c.vhdl --ghdl-source=src/vhdl/hdmi_spdif.vhdl --ghdl-source=src/vhdl/spdif_encoder.vhdl --ghdl-source=src/vhdl/buffereduart.vhdl --ghdl-source=src/vhdl/mfm_bits_to_bytes.vhdl --ghdl-source=src/vhdl/mfm_decoder.vhdl --ghdl-source=src/vhdl/mfm_gaps_to_bits.vhdl --ghdl-source=src/vhdl/mfm_gaps.vhdl --ghdl-source=src/vhdl/mfm_quantise_gaps.vhdl --ghdl-source=src/vhdl/crc1581.vhdl --ghdl-source=src/vhdl/ethernet.vhdl --ghdl-source=src/vhdl/ethernet_miim.vhdl --ghdl-source=src/vhdl/ghdl_fpgatemp.vhdl --ghdl-source=src/vhdl/expansion_port_controller.vhdl --ghdl-source=src/vhdl/slow_devices.vhdl --ghdl-source=src/vhdl/audio_complex.vhdl --ghdl-source=src/vhdl/audio_mixer.vhdl --ghdl-source=src/vhdl/pdm_to_pcm.vhdl --ghdl-source=src/vhdl/pcm_to_pdm.vhdl --ghdl-source=src/vhdl/i2s_clock.vhdl --ghdl-source=src/vhdl/i2s_transceiver.vhdl --ghdl-source=src/vhdl/pcm_clock.vhdl --ghdl-source=src/vhdl/pcm_transceiver.vhdl --ghdl-source=src/vhdl/keymapper.vhdl --ghdl-source=src/vhdl/keyboard_complex.vhdl --ghdl-source=src/vhdl/kb_matrix_ram.vhdl --ghdl-source=src/vhdl/keyboard_to_matrix.vhdl --ghdl-source=src/vhdl/matrix_to_ascii.vhdl --ghdl-source=src/vhdl/widget_to_matrix.vhdl --ghdl-source=src/vhdl/ps2_to_matrix.vhdl --ghdl-source=src/vhdl/keymapper.vhdl --ghdl-source=src/vhdl/virtual_to_matrix.vhdl --ghdl-source=src/vhdl/internal1541.vhdl --ghdl-source=src/vhdl/driverom.vhdl --ghdl-source=src/vhdl/dpram8x4096.vhdl --ghdl-source=src/vhdl/ps2_to_uart.vhdl --ghdl-source=src/vhdl/dummy_uart_monitor.vhdl --ghdl-source=src/vhdl/uart_rx.vhdl --ghdl-source=src/vhdl/ghdl_chipram8bit.vhdl --ghdl-source=src/vhdl/ghdl_farstack.vhdl --ghdl-source=src/vhdl/shadowram.vhdl --ghdl-source=src/vhdl/colourram.vhdl --ghdl-source=src/vhdl/charrom.vhdl --ghdl-source=src/vhdl/ghdl_ram128x1k.vhdl --ghdl-source=src/vhdl/ghdl_ram32x1024.vhdl --ghdl-source=src/vhdl/ghdl_ram18x2k.vhdl --ghdl-source=src/vhdl/ghdl_ram8x4096.vhdl --ghdl-source=src/vhdl/ghdl_ram8x4096_sync.vhdl --ghdl-source=src/vhdl/ghdl_ram32x1024_sync.vhdl --ghdl-source=src/vhdl/ghdl_ram8x512.vhdl --ghdl-source=src/vhdl/ghdl_ram9x4k.vhdl --ghdl-source=src/vhdl/ghdl_screen_ram_buffer.vhdl --ghdl-source=src/vhdl/ghdl_videobuffer.vhdl --ghdl-source=src/vhdl/ghdl_ram36x1k.vhdl --ghdl-source=src/vhdl/asym_ram.vhdl --ghdl-source=src/vhdl/debugtools.vhdl --ghdl-source=src/vhdl/crc.vhdl --ghdl-source=src/vhdl/cputypes.vhdl --ghdl-source=src/vhdl/debugtools.vhdl -c -fpic -o e~cpu_test.o e~cpu_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:1604
Call stack traceback locations:
0x7fd386a6f701 0x47cda6 0x587519 0x5966ac 0x64daff 0x64f74d 0x408ccd 0x7fd385929b95 0x407e58 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
```

**Additional context**
Built using:

```
./configure "--with-backtrace-lib=../../libbacktrace/.libs/libbacktrace.a" --with-llvm-config=`which llvm-config-8`
make clean
make
sudo make install
```

I've also tried with LLVM 7, and LLVM 6, and similar error result.

tgingold, 2020-06-04:
Can you commit generated files ?  I get:

$ make simulate-llvm
java -jar KickAss/KickAss.jar src/hyppo/main.asm -afo
No Java runtime present, requesting install.

So I prefer not to have to install java!

gardners, 2020-06-04:
Will do now.

gardners, 2020-06-04:
HICKUP.M65 now in the repo.

tgingold, 2020-06-04:
Ok, but then I have:

$ make simulate-llvm
make: *** No rule to make target `src/vhdl/s27kl0641.vhd', needed by `simulate-llvm'.  Stop.

And if I remove this dependency, the makefile tries to build pngprepare (for with I need to install libpng), then the makerom utility needs python2 (I need to change it because I have either python3 or python2.7)...

The simplest would be to package all the source files, and maybe prune the source set a little bit!

gardners, 2020-06-04:
Throw me an email at paul@m-e-g-a.org and I'll send you a tarball (the s27
file and some others may have copyright problems for posting publicly).

Paul.

tgingold, 2020-06-05:
I have sent you an email yesterday, did you receive it.

Otherwise, you could directly write to ghdl at free dot fr

gardners, 2020-06-05:
I didn't see it, but I think I have found the cause: There were .o files or some other build artefacts from a previous ghdl compilatioon of the target which seemed to be confusing things.  In any case, I can't reproduce the problem here now after I cleaned up my build directory. So let's mark it closed.

closed 2020-06-05 by gardners