ghdl/ghdl #1354: predefined floating to string fails

GlenNicholls, 2020-06-05:
`to_string(<pre defined floating>)` fails with some raw values like so:

```vhd :file: mwe.vhd
library ieee;
context ieee.ieee_std_context;
use ieee.math_real.all;
use ieee.numeric_std_unsigned.all;

entity mwe is
end entity;

architecture a of mwe is
begin
  process
  begin
    report "some=" & to_string(25.0e6);
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 mwe.vhd
```

yields

```
internal error: eval_monadic_operator: IIR_PREDEFINED_FLOATING_TO_STRING

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-451-ge3eb603c) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-w64-mingw32
C:\ghdl\mwe\prev_real_in_string\
Command line:
C:\ghdl\ghdl-e3eb603-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\ghdl\ghdl-e3eb603-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\ghdl\ghdl-e3eb603-mingw64-llvm\lib/ghdl\ -c -o mwe.o mwe.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:190
Call stack traceback locations:
0x4aba2a 0x4e55ef 0x4ee235 0x4ee3c5 0x4ee8e2 0x53a7c2 0x550702 0x546a41 0x546ad0 0x5470ab 0x54fbce 0x550748 0x551756 0x52bf0f 0x52bf53 0x52d865 0x52d9e9 0x52e552 0x52ecb9 0x52eef7 0x52f039 0x5650db 0x56b7e5 0x5263af 0x637c74 0x6398c5 0x63b2e5 0x402701 0x4013c2 0x401519 0x7ff930417bd2 0x7ff930b2cecf
******************************************************************
C:\ghdl\ghdl-e3eb603-mingw64-llvm\bin\ghdl.exe: compilation error
```

- OS: windows 10
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `e3eb603c`

eine, 2020-06-05:
I think it needs to be `to_string(25000000.0)` and `to_string(to_real(v_real))`. Not sure about `25.0e6`.

GlenNicholls, 2020-06-05:
For `report "some=" & to_string(25000000);`, that was just to show that it works for integers. The second two were to show it fails for real types. GHDL didn't seem to like `report "some=" & to_string(real(v_real));` even though it is completely valid. It was a typo on my part when I found this bug, but took me a while to figure out what was failing since ModelSim didn't seem to mind.

eine, 2020-06-05:
I updated with some further tests. The reference to `to_real` was my mistake, as I thought that the variable was sfixed/ufixed. It does not apply here.

It seems that `report "some=" & to_string(real(v_real));` has been fixed between your version of GHDL and mine. I agree that others should work.

commit d3c280b0 (2020-06-13): vhdl-evaluation: handle to_string for floating point types.  Fix #1354
closed 2020-06-13 by tgingold