ghdl/ghdl #1361: Segfault in SPI flash sim model

ozbenh, 2020-06-09:
**Description**
When trying to simulate an SPI flash with my SPI flash controller test bench,
the generated binary segfaults.

**How to reproduce?**
```
git clone git@github.com:ozbenh/microspi.git
make
./tb_flash
```
The resulting backtrace in gdb is:

```
Program received signal SIGSEGV, Segmentation fault.
0x0000555555625b04 in model__s25fl128s__ARCH__vhdl_behavioral__behavior__mempreload( const &&&*) ()
(gdb) backtrace 
#0  0x0000555555625b04 in model__s25fl128s__ARCH__vhdl_behavioral__behavior__mempreload( const &&&*) ()
#1  0x000055555563b836 in grt.processes.run_processes (postponed=false) at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-processes.adb:808
#2  0x000055555563bb0b in grt.processes.initialization_phase () at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-processes.adb:877
#3  0x000055555563c105 in <__ghdl_simulation_init> () at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-processes.adb:1130
#4  0x000055555563c455 in grt.processes.simulation () at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-processes.adb:1255
#5  0x00005555556474c0 in __ghdl_run_through_longjump (func=0x55555563c448 <grt.processes.simulation>)
    at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/./src/grt/config/jumps.c:232
#6  0x00005555556470f8 in grt.main.run_simul () at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-main.adb:169
#7  0x00005555556471cf in grt.main.run () at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/grt-main.adb:204
#8  0x00005555556603e4 in ghdl_main (argc=1, argv=0x7fffffffde68) at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/src/grt/ghdl_main.adb:60
#9  0x000055555564773c in main (argc=1, argv=0x7fffffffde68) at /home/ANT.AMAZON.COM/benh/grabbag/ghdl/./src/grt/main.adb:31
```

It dies in the MemPreload process. Commenting it out works.

Since "UserPreload" is FALSE however, it shouldn't have done anything other than
opening the files. (BTW. Failing to open files shouldn't be fatal should it ? I've had to include dummy files to avoid that other problem since it will try to open them even when UserPreload is FALSE).

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [*] Built from sources: 
```
GHDL 1.0-dev (v0.37.0-707-g1ec534e9) [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
 llvm code generator
```

tgingold, 2020-06-10:
Same issue as #1030 ?

tgingold, 2020-06-10:
The workaround is to modify `model/s25fl128s.vhd` around line 6705 like this:

``
        -- memory preload
        IF (mem_file_name /= "none" AND UserPreload) THEN
          ind := 0;
          for j in Mem'range loop
            mem (j) := MaxData;
          end loop;
--            Mem := (OTHERS => MaxData);
            WHILE (not ENDFILE (mem_file)) LOOP
                READLINE (mem_file, buf);
                IF buf(1) = '/' THEN
```

(ie not using an aggregate).

ozbenh, 2020-06-10:
Yes, same issue. Thanks. Closing.

tgingold, 2020-06-11:
I keep it open, as there is a full testcase.

ozbenh, 2020-06-11:
Ah let me make a branch for that then as master's been changing

eine, 2020-06-11:
@ozbenh, you can also drag & drop a zipfile/tarball in a comment here, should you prefer.

commit 71a81bae (2020-06-13): vhdl/translate: simplify some variable assignments.  Fix #1361

When the expression is an aggregate of the form (others => X), do not use
intermediate variables.
closed 2020-06-13 by tgingold